bipropellant/bipropellant-hoverboard-firmware #3: Flash access not working

p-h-a-i-l, 2019-04-29:
as noted in https://github.com/bipropellant/hoverboard-firmware/pull/2 and https://github.com/bipropellant/hbprotocol/pull/7

btsimonh, 2019-04-30:
Reason I found it was that I'd swapped use of #ifdef FLASH_CONTENT (which is a structure) for FLASH_STORAGE (which is the actual controlling define?) in main, but not both entries, and initflash was not called.
This lead to the flash not being intialised, and so the data being empty every start - and that lead me to investigate flash.  I found the non-inialisation fault, but then found that the flash values did not survive reboot.

I enabled 'DEBUG_SOFTWARE_SERIAL' which enables console logs after 'Ec\n' on ascii....
The flash writer was running, and incrementing the pointer, and the address was in the 0x8000xxxx range...  

but this warning could be the key:
'Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!'
(but; is this just a generic warning - we specify linker script in command line arguments in platformio.ini?)

the flash code relies on the data begin in a special segment ('.myFlashSection') which must be 2048 aligned, (and beyond the end of code?).

It works like this:
An area the size of one erase block is created outside of the code area, filled with 0xff in the initial flash image. (is there some way this is made to be not present in the image?)
If the flash magic is not at the start, it's erased and initialised.
When reading, the flash read looks for the END of used data, and so reads the most recently written.  Only if a new block will not fit is an erase operation done, and the data starts to be written from the start again.
Flash values are kept in RAM after read.  When writing a flash variable, it is written to RAM.
only by using 'Fm1234\n' in ascii can the RAM be written to flash..

I'll try to load the old firmware and see if that worked..... If i get a chance.

btsimonh, 2019-04-30:
I just did a few commits; nothing I think i changed should have changed anything, but flash IS working.  So not quite understanding what was wrong; probably just me.  Anyway, new commits add memory dump and flash dump to ascii

closed 2019-04-30 by btsimonh