bipropellant/bipropellant-hoverboard-firmware #5: Debug configuration

adverst, 2019-05-01:
How do you all have your debug configuration setup? I'm using the PIO setup in windows with ST-Link V2. Are you using black magic probe or similar?

btsimonh, 2019-05-01:
For me - I've not run a debugger on it for quite some time.  My dev env is Windows 10, VSCode+platofrm.io, but since I mounted the RPiZeroW+STLinkV2 inside, I have programmed from there exclusively with OCD (before that, I may have tried some things, don't recall - maybe simple GDB, but my experience of gdb is limited so it just slowed me down?).
Maybe I should look at an IDE direct on the pi?  Or another option is some form of OCD server for LAN based debug.
For the moment, it's just sprintf and consoleLog() for me.

p-h-a-i-l, 2019-05-01:
I have been using openocd with gdb to debug. I found PIO also to be convenient, but I wasn't convinced enough to pay for a premium Membership. Since I was only using a stlink v2, I also had to do some workarounds to get useful information.

Be careful when setting breakpoints, you don't want commutation to stop.

Do you have experience with black magic probes?

adverst, 2019-05-03:
I don't have any experience with black magic probes, I'm a bit of a beginner with hardware tbh. I've got some experience doing basic screwing around with raspberry pi/arduino, but not a ton. 

I'm more of a software guy, lack of ability to easily debug makes me feel like there's a magical black box that controls everything. I may get a BMP just so I can try it out, just figured I'd check here to see what sort of setup you all were using.

btsimonh, 2019-05-25:
now an old issue...  closing

closed 2019-05-25 by btsimonh