bipropellant/bipropellant-hoverboard-firmware #6: Default config

p-h-a-i-l, 2019-05-04:
Check out how I set up the config in my fork: https://github.com/p-h-a-i-l/hoverboard-firmware-hack

Especially this line: https://github.com/p-h-a-i-l/hoverboard-firmware-hack/blob/7def7804e4fb3f2ca095c9f34b3edcdd963ca029/inc/config.h#L3

The whole config file is deactivated, as soon as `IGNORE_GLOBAL_CONFIG` is defined.

I use this to tack my config through platformio.ini, but still avoid annoying everybody by changing the default behaviour all the time. (https://github.com/p-h-a-i-l/hoverboard-firmware-hack/blob/bewegungsappar.at/platformio.ini)

Another nice thing: if no environment https://github.com/p-h-a-i-l/hoverboard-firmware-hack/blob/7def7804e4fb3f2ca095c9f34b3edcdd963ca029/platformio.ini#L4-L8 is set as default, all variants are build. This is nice to check if I break other customizations without setting all possible defines by hand.

p-h-a-i-l, 2019-05-06:
Also, how what do you think about using travis as CI? Would be nice if travis could automatically build the same targets as described above.

btsimonh, 2019-05-06:
if we can find a repo which uses travis & platformio, it should work?

p-h-a-i-l, 2019-05-06:
Ah, platformio and travis are already compatible. Nice.
I just [played around with travis and default make](https://github.com/p-h-a-i-l/hoverboard-firmware-hack/tree/travis).

p-h-a-i-l, 2019-05-06:
@EmerickH please allow me to add travis, owner permissions are neessary for that.

p-h-a-i-l, 2019-05-09:
@EmerickH can you please add @btsimonh and myself to the owners of bipropellant?

I've seen that you approved my request to grant travis access to this organization, but other things are still necessary which require owner permissions. There should be another pending request from two days ago to install github apps.

EmerickH, 2019-05-10:
Sorry I was in exam this weel I had not a lot of time.
I'm sorry but I can't give you more than admin permissions:
[image]
Same thing for teams...

I installed Travis on both repos
https://travis-ci.com/organizations/bipropellant/repositories
But it is failling on both:
 * For hb-firmware on gcc-arm-none-eabi-5 only (not pio and gcc-arm-none-eabi-7) because of this:
https://github.com/bipropellant/hoverboard-firmware/blob/master/src/setup.c#L41
(unknown file?)
 * For hb-protocol beacause there is no makefile/it cannot recognize it as C

p-h-a-i-l, 2019-05-10:
Thanks!

I'll look into the build problems with gcc-arm-none-eabi 5.
In order to use travis with hb-protocol, would be great, but for me this is not a priority.

btsimonh, 2019-05-10:
if travis can be used with the main repo, that's the only thing that matters?

p-h-a-i-l, 2019-05-10:
Yeah, but it would also be nice to test only the submodule. If you push to hbprotocol, the main repo does not get a automatic submodule update. So travis is only triggered, if somebody updates the main repo manually.

p-h-a-i-l, 2019-05-10:
travis running nicely now: https://github.com/bipropellant/hoverboard-firmware/runs/124546527
@btsimonh: I changed from memory.h to string.h as a path to find memset, memcpy.. Is that ok for you? 
seems to be better with gcc-arm-none-eabi-5

p-h-a-i-l, 2019-05-10:
also environments created in platformio. Set the environment you need in platformio.ini or specifiy via command line
for example  `platformio run -e hoverboard`

btsimonh, 2019-05-11:
> @btsimonh: I changed from memory.h to string.h as a path to find memset, memcpy.. Is that ok for you?
> seems to be better with gcc-arm-none-eabi-5

whatever works :).

closed 2019-05-22 by p-h-a-i-l