bipropellant/bipropellant-hoverboard-firmware #8: Repository Name

p-h-a-i-l, 2019-05-08:
I noticed today, that our repository name is probably not the best name..

First working Firmware for Hoverboard: https://github.com/salviador/Hoverboard-Board-Hack
Niklas refactoring: https://github.com/NiklasFauth/hoverboard-firmware-hack
Isballes hard fork: https://github.com/isabellesimova/HoverboardFirmware
Fork for split motor controller https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2

Ours https://github.com/bipropellant/hoverboard-firmware

I can see, how this will lead to a lot of confusion..

EmerickH, 2019-05-10:
I totally agree, what about renaming to **bipropellant-firmware**?

p-h-a-i-l, 2019-05-10:
Ilike it, only thing would be that it's harder to find. I think most people search for hoverboard.

btsimonh, 2019-05-10:
the differentiation here is
1/ ability to use it as a hoverboard (not important to ether of you :) )
2/ the reliable machine protocol.
BOTH are important :)

p-h-a-i-l, 2019-05-10:
Let's not focus the name too much on our current main features, who knows what will come?

lalalandrus, 2019-05-19:
Balancing hoverboard firmware

btsimonh, 2019-05-20:
there are a couple or arduino based projects using normal motors with fairly good explanations and code;  one of my next tasks is to tap a few holes in the top of the HB to facilitate mounting of different 'bodies'.  For balancing, you need some decent mass some distance above the HB.

p-h-a-i-l, 2019-05-20:
What about simply "bipropellant-hoverboard-firmware"?

p-h-a-i-l, 2019-05-23:
@btsimonh and @EmerickH: Let's try to make a decision soon, this repo is gaining popularity and it's GPN Next week :)

btsimonh, 2019-05-23:
"bipropellant-hoverboard-firmware" +1

btsimonh, 2019-05-25:
renamed :)

closed 2019-05-25 by btsimonh