bipropellant/bipropellant-hoverboard-firmware #10: Top speed

lalalandrus, 2019-05-11:
Is it possible to include the field weakening code from https://github.com/larsmm/hoverboard-firmware-hack-bbcar to increase the top speed of the hoverboard. 

Because this projects re-integrates the sensors, it can be used in conjunction with the hoverboard carts and achieve a high top speed.

btsimonh, 2019-05-11:
looks like weakl and weakr are already there; we'd just need the code to set them....
Then I guess you could use a sensor angle as an accelerator pedal!
Testing would be an issue; the three of us are 'low speed' applications.  With a suitable build, it could be 'road tested' unmanned and wifi controlled, but would exceed the speed limit on my road :).
PRs welcome.

p-h-a-i-l, 2019-05-11:
Pull requests are very welcome 😋

lalalandrus, 2019-05-11:
What if I told you I can help convert all your wheels into 10inch. I can write a tutorial and you can do balance at high speed :)

lalalandrus, 2019-05-11:
What if I told you I can help convert all your wheels into 10inch. I can write a tutorial and you can do balance at high speed :)
https://imgur.com/a/8RkTix0

btsimonh, 2019-05-11:
haha!  I did think about if we could put something around the wheels - or fit 'snow chains' for operation on grass :).

lalalandrus, 2019-05-11:
This widens the base and the stability increases substantially

btsimonh, 2019-05-11:
I can see why you want to go faster that your standard hoverboard; with standard hoverboard control :).  May need to upgrade the way the wheels are attached, and check the bearings occasionally.  But also I guess that your weight is not necessarily directly over the wheels, so the bearings may be under a slightly lighter load.

lalalandrus, 2019-05-18:
I tried compiling it through platformIO and programming firmware.bin but the board is now unresponsive. Is there anything special I need to do?

[image]

btsimonh, 2019-05-18:
you are in hoverboard mode?
Is it a real STM32 or a GD32?
For my GD32 based board, the serial rates to the sensor boards did not match that reported by others...  
```
  #define USART2_BAUD     52177    // control via usart from GD32 based sensor boards @52177 baud
  #define USART3_BAUD     52177    // control via usart from GD32 based sensor boards @52177 baud
//#define USART2_BAUD     26300    // reported baudrate for other sensor boards?
//#define USART3_BAUD     26300    // reported baudrate for other sensor boards?
```
To my knowledge, I am the only person who has run this with sensor boards (so they have only been tested with my GD32 based YST board setup).
Did you get a serial connection to it?  - that's probably the place to start....
And get a 'scope on the serial out of the sensor boards and estimate the baud rate - i think they pump data out constantly from power on.

good luck!

lalalandrus, 2019-05-18:
My main board is a real ST32 and the sensors are GD32. I will scope it and read the serial port. Right now it is just stuck always on doing nothing.

btsimonh, 2019-05-18:
first time i programmed mine from the new repo, my flash played up; never got to the bottom of it, but now seems ok.  if Hoverboard mode is disabled in flash, then it will not do hoverboard functions.
You have no lights, and no response to sensor foot pads?  try 'double tap' on the footpads (two activations in one second).

lalalandrus, 2019-05-18:
Abusing the power off pin I tracked it down to init flash. I haven't had a chance to test but I am suspecting that you are not really saving a flash area. It is not defined in the linker or the flash area.c. Also I don't understand how you got 0x8000000+62*2048 in flash access but you want 0x800c000 (but not declared) in flash area.

btsimonh, 2019-05-18:
the flash was my imaginative from scratch implementation, just stole some code from elsewhere. :).
I think (a little time ago now) I intended it to beyond the end of any code, and if flash magic not matched, to re-write.  Using the serial, you have to write the correct value to the flash magic to save to the flash....  but it seemed to work as i diagnosed it a few weeks ago.
The flash area should be a segment in the linker file: (but see below... this is irrelevant)...
https://github.com/bipropellant/hoverboard-firmware/blob/master/STM32F103RCTx_FLASH.ld
```
  /* create a section in flash which is aligned on a page boundary, don't really care where. */
  .myFlashSection :
  {
     . = ALIGN(2048);
     *(.myFlashSection)
     *(.myFlashSection*)
     . = ALIGN(2048);
  } >FLASH
```
the referenced:
`const uint8_t __attribute__((section (".myFlashSection"))) flash_data_inimage[FLASH_LEN]  __attribute__ ((aligned (2048))) = {
`

but.... you are right, in flash_access.c:
`volatile uint8_t *flash_data = (uint8_t *) 0x08000000 + 62*2048;`

... so we don't use the 'inimage' variable.... and we rely on 0x08000000 + 62*2048 being beyond the end of our code.
best double check you have this amount of flash in STM32?

s

lalalandrus, 2019-05-19:
I found the issues and cleaned up the section defines + rearranged inits until I got the code working. I am now stuck with both "power" led's being red but at least it is not crashing. The hoverboard is turning off via button but not via the 60s timeout, this might be my next clue. Ill send the pull requests for the other stuff.\

btsimonh, 2019-05-19:
power leds mean you have coms to sensor boards (i believe).
if HB mode is active, you should get dull green for one foot, bright green for two, and motors should try to make HB level in red.
If not, double tap should enable HB mode.
did you get serial to it yet?

lalalandrus, 2019-05-19:
Serial is the correct baud. But I suspect there is a bug there sensor board uart died because I can't see any serial from the main board to the sensors when it should be constantly sending light info

btsimonh, 2019-05-19:
you have software serial for diags?

lalalandrus, 2019-05-19:
After the power on message nothing else pops up

btsimonh, 2019-05-20:
did you send unlockASCII? - but it should tell you to do this on the serial.....
To get it to output debug you may need to enable Ascii and 'EC<cr>'.
then to dump flash 'Fa'<cr>.
To commit any changes to flash 'Fm1234<cr>'.

lalalandrus, 2019-05-20:
while i have found a few quirks and optimizations i think the issue is that the type of sensor boards i have are sending the data immediately one after another and the mainboard code that i have is unable to find the SOF and the bit alignment is wrong. the 9 bit serial i am sending to the led is working

btsimonh, 2019-05-22:
check the new 'combined' branches...  I believe the 0x1xx is the first thing received in each message.
For lights, I had to move 'color' to the front.
For receive, I now wait for 0x1xx, and then put that and up to  bytes into a buffer.  on the 10th byte, I coy the buffer to the used structure, and process it.

lalalandrus, 2019-05-22:
i already moved colour to the front in sensorcomms.h
i think your yst boards have more info, the standard sensor boards only have 6 bytes including the header mem copy would be dangerous in this instance and direct variable definition would yield better cross compatibility should your strut packing change

lalalandrus, 2019-05-23:
i finally got my hoverboard working but it seems to go into some emergency stop mode once i hit 3km/h

any thoughts?

btsimonh, 2019-05-23:
yes, you are right; I have 10 byte frames from the sensor board.  Better not implement anything important based on Roll or acceleration!

A thought on baud rate:
1/ they want to have a constant FRAME RATE.  
2/ My frames are 10 words (11 bits each?) at 52177 -> frame interval 2.1ms
3/ Your frames are 6 words (11 bits each?) at 26315 -> frame interval 2.5ms
4/ @EmanuelFeru is at ~32100  -> maybe he has 7 words?

ref stop mode.  Does it bleep?
It could be batt, temp, or timeout.  Power off timer should not activate if you have motion.
It could be voltage detect calibration, or temp calibration?
Let us know after a charge :).

TomTinkering, 2019-05-23:
Hey guys, I saw this comment on the "NiklasFauth" thread for Sinusoidal control of PMSM;

"@lalalandrusVector control like FOC, indeed would be nice, however I think we will reach the STM32F1 Series computational limitations, since we need to run the algorithm at 100 kHz and plus some additional code for balancing controller or other calculations. STM32F4 series are much more capable but they are not in the hoverboard."

I just ordered [this](https://www.ebay.nl/itm/Circuit-Board-Main-Motherboard-For-6-5-10inch-Balance-Scooter-Replacement-Part/143217214985) hoverboard control board, and it comes with an F4 by GigaDevices. It also includes Mosfet drivers rather than a custom discrete implementation with BJT's for the motor drives. 

Looking on Ebay I see a lot more of these boards around, so could it be worth developing for this board instead? I could help out with the base firmware (ADC's, PWM outputs etc.), but I don't know much about FOC. I do have an updated [schematic](http://vocke.tv/lib/exe/fetch.php?media=20190518_hoverboard_sch.pdf) of the board.

btsimonh, 2019-05-23:
it's probably the same as my YST board....  GD32 based, double the CPU power (not sure what clock I'm running with the firmware though).
there is some info on my board herE:
https://github.com/btsimonh/hoverboard-firmware-hack/tree/newprotocol/otherboards/YST-6.5-10-PCBA-10
and some pics in folder above.
If it is the same, it's got some nice spare pins, not too difficult to access....
Confirm?

TomTinkering, 2019-05-23:
Yes, it is an YST board, but I think the next generation, as it features a GD32F403 instead of 103

EmanuelFeru, 2019-05-23:
> 
> 
> Yes, it is an YST board, but I think the next generation, as it features a GD32F403 instead of 103.
> 
> I will be using this board in the coming weeks for a project, and was planning to port the NiklasFauth firmware for it. Of course it would be even better if FOC would also be possible.

Cortex®-M4 Core @ 168 MHz​ ... nice! The board is cheap actually. I hope the interfaces are the same. Let us know if you the firmware runs on it. So, no issues with the configuration of port, timers, etc.

EmanuelFeru, 2019-05-23:
By the way, is someone of you using the [STM-STUDIO-STM32](https://www.st.com/en/development-tools/stm-studio-stm32.html) for vizualization and debugging? 
I tryied it and it worked nicely first times but after few usages it cannot read the variables any more from the "firmware.elf" file. I wonder if you had the same issue. 
I would like to use it because if very nice for Visualization and online tuning.

TomTinkering, 2019-05-23:
> > Yes, it is an YST board, but I think the next generation, as it features a GD32F403 instead of 103.
> > I will be using this board in the coming weeks for a project, and was planning to port the NiklasFauth firmware for it. Of course it would be even better if FOC would also be possible.
> 
> Cortex®-M4 Core @ 168 MHz​ ... nice! The board is cheap actually. I hope the interfaces are the same. Let us know if the firmware runs on it. So, no issues with the configuration of ports, timers, etc.

The interfaces are largely the same, but not all. Also the DC current measurement is removed (so only phase current/voltage). The firmware will likely not run, but I will work on that the coming weeks. I will let you know when I get Niklas's firmware running, then maybe we can continue from there

lalalandrus, 2019-05-23:
we have a LOT of flash, with look up tables and fixed point math we might be able to achieve pretty good motor control. 

if full vector control cant be achieved just lead / lag angle control might be close enough to be versatile to both the high speed and low speed applications

EmanuelFeru, 2019-05-23:
Just to tease you a bit :D check this out
https://drive.google.com/file/d/1vC_kEkp2LE2lAaMCJcmK4z2m3jrPUoBD/view

1st half of the video = Commutation method
2nd half of the video = Sinusoidal 3rd armonic

btsimonh, 2019-05-23:
ok, when can we implement? :)  is this yours?  or what git?  !!!!  can it do slowwwwwww?

EmanuelFeru, 2019-05-23:
My pure implementation... fresh new :)

EmanuelFeru, 2019-05-23:
I use Commutation method to start the motor.  I estimate the rotor position based on Hall sensors, so I need a spinning motor.
It spins slowly as well. How slow do you need?

btsimonh, 2019-05-23:
Really looking forward to it, because sinusoidal had me really flumuxed for quite a few days...  with no viable result...

btsimonh, 2019-05-23:
want fine control of position.....  for very slow - inspiration is here:
https://hackaday.com/2015/04/20/driving-a-brushless-dc-motor-sloooooooowly/

think robot rather than moving vehicle....

EmanuelFeru, 2019-05-23:
You can here a small wobble sound in the video... that is because the wheel is not perfect straight. It is also visible in the video. My hoverboard is a used one... I said if I am going to break it at least I should use a used one :)

EmanuelFeru, 2019-05-23:
> want fine control of position..... for very slow - inspiration is here:
> https://hackaday.com/2015/04/20/driving-a-brushless-dc-motor-sloooooooowly/
> 
> think robot rather than moving vehicle....

When I will publish you can try to see if it is sufficient for what you need. If not, we see what can we do.
One thing, also the original firmware starts the first spin with Commutation method, you can hear the tick tick noise, then it switches I guess to also a sinusoidal method. Did notice it?

btsimonh, 2019-05-23:
hmm...  I do get ticks.  But not really related to starting.  I'll try to understand your code when I see it :).

lalalandrus, 2019-05-23:
that sent chills down my spine...
i use my hoverboard for this
[image]

but with big tires https://imgur.com/a/8RkTix0 

i ride the hoverboard everyday at work and i was testing it on a ridge to think about how they do the commutation before your comment and yes there is a tick.

btsimonh, 2019-05-23:
My clicks are definitely bad sensor reads:
```
"colour 0 c -> 4 (51)"
5/23/2019, 11:28:55 PMnode: 9bab8706.92f058
"SensorOK -> 0 in unprocessed poll"
5/23/2019, 11:28:56 PMnode: 9bab8706.92f058
"colour 0 4 -> 8 (52)"
5/23/2019, 11:28:56 PMnode: 9bab8706.92f058
"colour 1 c -> 8 (53)"
5/23/2019, 11:28:58 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:28:58 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:08 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:11 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:14 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:35 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:35 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:46 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:47 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:47 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:49 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:50 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:53 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:55 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
5/23/2019, 11:29:56 PMnode: 9bab8706.92f058
"sensor data not 55 or AA"
```

p-h-a-i-l, 2019-05-23:
> Just to tease you a bit :D check this out
> https://drive.google.com/file/d/1vC_kEkp2LE2lAaMCJcmK4z2m3jrPUoBD/view
> 
> 1st half of the video = Commutation method
> 2nd half of the video = Sinusoidal 3rd armonic
> 
> * Pay attention on the noise level

What a teaser!!! We NEED this 😀

EmanuelFeru, 2019-05-24:
I am happy that you like it :-) I will do a bit more testing then I will share it with you.

@TomTinkering I saw in the presentation of Niklas that you built a GUI for tuning parameters [link](https://github.com/TomTinkering/CfgBus-GUI). What do I need to do to use it for hoverboard parameters tuning?

TomTinkering, 2019-05-24:
> @TomTinkering I saw in the presentation of Niklas that you built a GUI for tuning parameters [link](https://github.com/TomTinkering/CfgBus-GUI). What do I need to do to use it for hoverboard parameters tuning?

Yes, it is a bit involved unfortunately. There link you shared is a Qt5 gui that talks to the hoverboard using a serial port. On the hoverboard side you need to implement the client. It is based on modbus, and works with any modbus RTU tool. The tunable parameters need to be globals in the hoverboard code. You can dig through the code [here](https://github.com/TomTinkering/hoverboard-firmware-hack). It is some work unfortunately. If you let me know which parameters should be tunable, I might be able to port it to your project this weekend if you want to try it.

EmanuelFeru, 2019-05-24:
> If you let me know which parameters should be tunable, I might be able to port it to your project this weekend if you want to try it.

Thanks for the info. I would like to access the parameters for the new motor control that I made to do some calibration. These are new parameters compared to the Niklas repo.
Ideally, I was looking for a tool that can access the memory locations by loading the description file "firmware.elf". This file is created when building the code (I am using VS code).

lalalandrus, 2019-05-24:
There is a serial debug that will allow you to read and write flash that is implemented already.

btsimonh, 2019-05-24:
@EmanuelFeru - take a look at the hbprotocol wiki.
it's a little out of date - the parameter table has changed a bit (for the better).  (changes welcome!)

but basically, add your new params to 
https://github.com/bipropellant/hoverboard-firmware/blob/master/inc/flashcontent.h
with suitable defaults.

then add lines in 
https://github.com/bipropellant/hbprotocol/blob/master/protocol.c
near
`    { 0xA0, "hoverboard enable",       "he",  UI_SHORT, &FlashContent.HoverboardEnable,       sizeof(short),             PARAM_RW, fn_preWriteClear } // e.g. 20
`
to reference your new variables:
give them a human name, chars used to access them.  Only UI_SHORT is supported at the moment.
Choose a new number (maybe start at 0xB0?) - the number is used in the machine protocol to read.write.

then, e.g. working with 'hoverboard enable' as an example:
to read, use:
Fhe<\n>
to set to 1
Fhe1\n

this will SET the variable in MEMORY.
To commit the memory to flash, use 
Fm1234<\n>

First firmware update with new variables, use
Fi<\n>
to initialise.

(positive decimals only at the moment, pull requests welcome...).
br,
s

EmanuelFeru, 2019-05-28:
Hello everyone @btsimonh @lalalandrus @p-h-a-i-l @TomTinkering,

First release of the new BLDC control is committed here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack

Feel free to give it a try and let me know what you think about it.
@btsimonh What do you say of integrating this new control into your FW to restore the hoverboard functionality? It should now be very close to the original FW, if not better.

@lalalandrus  By the way... the Phase advance or Field weakening is crazy! it can reach really high speeds.

lalalandrus, 2019-05-28:
@EmanuelFeru i am excited, i will start trying to integrate the baseline into my fork now since i am bored at work. @btsimonh can make it pretty

EmanuelFeru, 2019-05-28:
Nice to do something fun like this at work, I agree :) I am exited too, for your feedback. 

> wow, it looks like you went through the trouble of modeling the motor parameters too!
I modeled the parameters of the motors roughly. The Simulink model is more for functional development. So, it is not exact with reality, but I fined tuned the controller on the real motor.

lalalandrus, 2019-05-28:
@EmanuelFeru you should test out the newest commits, the calibration makes the hoverboard functions really slick now, there is a 'pwm = anlge/3' which should be changed as it is a bit sluggish right now maybe a /2 is suffice.

EmanuelFeru, 2019-05-28:
Do you mean this repo https://github.com/bipropellant/bipropellant-hoverboard-firmware?
Because I really want to try this repo to "feel" the hoverboard functionality. Is it now working with STM32 boards?

lalalandrus, 2019-05-28:
yes this current repo, it is working with my stm board reliably but you might need to fiddle with your baud rate settings, your "should" be a 6word sensor as per the config.h file

EmanuelFeru, 2019-05-28:
I am not sure if mine is 6 word sensor. When I did the hack on the serial I was sending the data below. The motors were listening to my commands.

`		// Send data
		dataL[0] = 256;
		dataL[1] = uL & 0xFF;
		dataL[2] = (uL >> 8) & 0xFF;
		dataL[3] = uL & 0xFF;
		dataL[4] = (uL >> 8) & 0xFF;
		dataL[5] = 85;
		dataL[6] = 82;
		dataL[7] = 82;	
		
		dataR[0] = 256;
		dataR[1] = uR & 0xFF;
		dataR[2] = (uR >> 8) & 0xFF;
		dataR[3] = uR & 0xFF;
		dataR[4] = (uR >> 8) & 0xFF;
		dataR[5] = 85;
		dataR[6] = 82;
		dataR[7] = 82;
		
		if (stop)
		{
			dataL[5] = 170;
			dataR[5] = 170;
		}		
		
		HAL_UART_Transmit_IT(&huart1, (uint8_t *)dataL, n);
		
		HAL_UART_Transmit_IT(&huart3, (uint8_t *)dataR, n);

`

lalalandrus, 2019-05-28:
i believe the way the firmware is written now, if you select 6 word with the right baud rate, it will still work even if it has too much data as it will just ignore it until the next new frame. 

i have integrated your code into my branch, will need to do some testing before a PR but @EmanuelFeru and @btsimonh can test from my branch the new motor control

https://github.com/lalalandrus/hoverboard-firmware

p-h-a-i-l, 2019-05-28:
Got it running :+1: 
very nice :)

(had to add the new .c files to the makefile, but its working)

Some strange things: Melody on first boot was very slow, motors start after 5 minutes waiting, sometimes only one wheel.

p-h-a-i-l, 2019-05-28:
https://wolke.hacknology.de/index.php/s/4KyrzZtR6YeAbTF

lalalandrus, 2019-05-28:
@p-h-a-i-l that is quite the setup you have there

lalalandrus, 2019-05-28:
@p-h-a-i-l i think @EmanuelFeru has changed the irq behaviour in his new code

[93 more replies]

closed 2019-05-30 by lalalandrus