bipropellant/bipropellant-hoverboard-firmware #16: Calibration

lalalandrus, 2019-05-23:
Implement calibration. 

I have implemented calibration when holding the button when the HB is on. Holding the on button from startup is still reserved for future use. 

I have also implemented on the hoverboard updating of the center position to the calibrated center at the earliest possible. This mimics the behaviour of the original HB. 

I will wait until the current outstanding PR's are completed since the existing PR will make the code rather annoying to merge. (no pressure or anything)

p-h-a-i-l, 2019-05-23:
Calibration for mems sensors in general or also for electrical measurements?

btsimonh, 2019-05-23:
I hold the power button every flash :(.
What it currently does:
on power up, when HB mode is enabled with double tap, it will drive the motors to angle 0, but with a low clamp.  This 'levels' the board.  If you stand on one side, 'Center' is taken from angle at moment of step on (for both sides at first side stepped), and motors allowed full power.  This effectively keeps the second foot place level until the 2nd foot is on.
for calibration, would suggest '0' is replaced with 'flash center L'/'flash center R'?

Advanced power button commands  (press-release, no other press-> power off, long press -> x, two presses -> y, three ->z) would be a good generic addition to the firmware.  I'd prefer press and hold to  not do anything except on release.  I'd prefer for 'button held at startup' to not do anything - btoh of these are what I do daily when developing :).

lalalandrus, 2019-05-23:
Not the way the defines are working also the way it implemented now it will wipe out the flash

btsimonh, 2019-05-23:
Are your sensor boards far off level by default?  - I've only ever seen my boards in operation.
What angle does the HB settle at after double tap?
But yes, we could do with the default 'Center angle' being stored in flash, and a way to calibrate it.

lalalandrus, 2019-05-23:
Right now if you stand on the hoverboard on an angle it will stay at an angle forever.  Venter is declared in multiple places and it doesn't recenter.  To implement an updating center  without doing crazy stuff like averaging you just store the calibrated center and trust that.

Also once you double tap the board is active forever which I fixed too.

btsimonh, 2019-05-23:
yep, so the first foot causes center to be taken from both sides at step-on time.
Agreed, the center value should probably be zeroed at step off.
It will get a new center at step on regardless, but the board may be trying to move if you got a 'bad center' in there :).  Beware - center in the sensor_data is not used, There is also a variable in main.  This should be changed, and it set in one place.

lalalandrus, 2019-05-23:
Yes the new code fixes this but we need to get over that darn merge.

btsimonh, 2019-05-25:
implemented...

closed 2019-05-25 by btsimonh