bipropellant/bipropellant-hoverboard-firmware #17: Main loop interval

btsimonh, 2019-05-23:
Currently the main loop timing is determined by a delay at the start of the loop, and so the loop interval could vary.

We should change this to be 'delay until 5ms after the start of the previous loop', or 'delay until the next 5ms tick'.
When we do this, we should probably instrument the time in the main loop, and always delay by at least 1ms.

e.g. 
```
unsigned long now = HAL_GetTick();
unsigned long start_processing = now + DELAY_IN_MAIN_LOOP;
unsigned long main_too_long = 0;

// main loop
while (1) {
  unsigned long time_in = now;
  if (start_processing < now) {
    main_too_long++;
    start_processing = now + 1;  // at least 1ms of delay
  }

  // deliberate delay loop
  while (now < start_processing){
    // do idle stuff - e.g. comms.
    now = HAL_GetTick();
  }

  unsigned long processing_in = now;

  //////////////////////////////////
  // do 5ms main loop stuff
  //////////////////////////////////

  ////////////////////////////////
  // take stats
  now = HAL_GetTick();
  unsigned long main_interval = now - time_in ;
  unsigned long main_delay = processing_in - time_in ;
  unsigned long main_processing = now - processing_in ;
  // maybe average main_dur as a stat?

  // select next loop start point
  start_processing += DELAY_IN_MAIN_LOOP;
}
```
/////////////////

thoughts?

p-h-a-i-l, 2019-05-23:
Good Idea, I don't like periods which are based on config or state. 
What do you think of cleaning up main() in general? we can improve readability greatly by grouping into functions.

lalalandrus, 2019-05-23:
Ha I was in the process of doing that in my code too. Both readability and a set pause. I got hung up because you still can get up to 500us of jitter. We should try to implement a us counter which I had found code for but was focusing on the emerge.

btsimonh, 2019-05-23:
Hall effect code is running a high frequency timer value which should extend to 64 bits :).
We can just set a copy of it, and it's rate is 100khz, if the comments are to be believed.
or we can read it direct with
```
    __disable_irq(); // but we want both values at the same time, without interference
    unsigned long time = h_timer_hall.Instance->CNT;
    long long timelong = (timerwraps << 16) | time;
    __enable_irq();
```
(see hallinterrupts.c)

lalalandrus, 2019-05-23:
that looks pretty good, we can use this for better timing control too

btsimonh, 2019-05-23:
Any taker on grabbing this?  Else I'm happy to do the timing stuff on a new branch off master?

lalalandrus, 2019-05-23:
u can, i am waiting for the final merge and re-implement calibration. we might be better served if we tidy up main and split it into helper functions to help facilitate the fast/slow scan move

btsimonh, 2019-05-23:
see maintime branch....  (but on top of 'combined').
result is we spend ~800us out of 5000us processing the main loop.

btsimonh, 2019-05-25:
committed....

closed 2019-05-25 by btsimonh