bipropellant/bipropellant-hoverboard-firmware #20: Two inactivity timeouts

lalalandrus, 2019-05-24:
There are two inactivity timeouts. 

I tried to move one but the one uses a variable in hbprotocol which is in another branch.

If hbprotocol is to stand on its own, it shouldnt depend on variables declared in other files such as main.c.
We should combine the two inactivity timeouts or eliminate one of them

p-h-a-i-l, 2019-05-24:
Also timeout it is not functional right now with the machine protocol. Motors keep on running when input is long gone. Need to add a timeout++;

lalalandrus, 2019-05-25:
This issue should be expanded to just general code cleanup, but it is low priority until at least the new motor code is intergrated.

p-h-a-i-l, 2019-05-25:
maybe we shouldn't delay adding timeout++ for too long.

lalalandrus, 2019-05-25:
Because of that hbprotocol dependency I don't want to get involved with that block of code at this time. I don't want to get stretched thin.

LeoDJ, 2019-06-21:
I'll just comment here:
Yeah, a timout function would be very nice. Otherwise I'll see if I'm able to implement it and an emergency stop digital input on my own in the next days/weeks ;)