bipropellant/bipropellant-hoverboard-firmware #31: Slow speed control using sinusoidal waveforms.

btsimonh, 2019-05-29:
@EmanuelFeru -

I think what we need is a function:

`    BLDC_controller_angle_L( theta, amplitude );`

which fills rtY.DC_phaALeft; rtY.DC_phaBLeft; rtY.DC_phaCLeft;

with the values for that phase angle at the amplitude specified.

Also, I think that the 
`   BLDC_controller_step()`
should probably be separated into BLDC_controller_step_L and BLDC_controller_step_R, as it's likely we'd want to change between low speed and higher speed modes separately on each wheel?

no rush, off to sleep :).

AntumArk, 2019-05-30:
You could look to VESC code for help. They may have solved that problem.

btsimonh, 2019-05-30:
@EmanuelFeru - 
pls look at my test branch I just pushed: https://github.com/bipropellant/bipropellant-hoverboard-firmware/tree/separateleftright

I find some interesting results when looking at slow speed positon control....

This implements STATIC use of your sine tables.
On serial, use 'If\n' to select 'fine control'.
Then keys W/S will increment through the 36 positions in rtConstP.pooled12 (etc.).
keys A/D decrease/increase current.  
Take the current up to ~80, then increment the position.
q to exit/disable.
only controls one wheel.

The movement is 
1/ more than I expected?
2/ not even? (4 bigger steps, 2 smaller steps).

The non-even could mean we need 'non-linear' sine tables for smooth running?  Could this be one of our issues?

br,
s

EmanuelFeru, 2019-05-31:
Aa yes, Now I put the LEFT and RIGHT wheel code together and let the Embedded Simulink coder do the re-usability of functions optimization. So, it automatically reuses functions and pieces of code where it can.

I can generate code for each wheel separate: BLDC_controllerLeft_step(), BLDC_controllerRight_step(), but then the code will increase not really double but definitely will increase.