bipropellant/bipropellant-hoverboard-firmware #33: Just trying to test this out with arduino and UART3

papaxiclas, 2019-06-02:
Hello,
I am trying to control the board without sensors attached and using one arduino nano.
After loading the firmware with the `#define CONTROL_TYPE USART3_CONTROLLED` on config.h and using the provided arduino example on the same file
`void loop() {

  int sensorValue = analogRead(A2);
 
  speed = sensorValue;
  Serial.write((uint8_t *) &steer, sizeof(steer));
  Serial.write((uint8_t *) &speed, sizeof(speed));
  delay(20);
}` with a serial speed of 115200 but I simple don't get any sort of movement.
When using the ASCII all works has it should...
Is there anything else I should look for? Is this info correct?
Thanks

btsimonh, 2019-06-02:
I'm not aware we had an arduino sample?
In terms of machine control, it uses a more formal protocol -  see the biprolpellant-protocol [wiki](https://github.com/bipropellant/bipropellant-protocol/wiki/Protocol-Defn) for some details, but that page is a little out of date, so also check the code.

papaxiclas, 2019-06-02:
On config.h there is this text:
// for Arduino, use void loop(void){ Serial.write((uint8_t *) &steer, sizeof(steer)); Serial.write((uint8_t *) &speed, sizeof(speed));delay(20); }

Read the wiki page and protocol.h and realized this is way out of my coding abilities.
I will wait for some more development and maybe and arduino example I can mess with.
Thanks for your help :)

btsimonh, 2019-06-02:
you could try the original repo....  
https://github.com/NiklasFauth/hoverboard-firmware-hack
this should function as desired, but the serial control is a little weak.... (what happens if they get in the wrong order?)

papaxiclas, 2019-06-02:
I got the original working nicely. https://youtu.be/RF1e5va7jVo
But I have UART from remote to car, and one arduino doing uart->PPM in the car to drive the NiklasFauth firmware.
Reliable UART and PID drive modes attracted me here:)

p-h-a-i-l, 2019-07-01:
closed, sample code will be added with this PR https://github.com/bipropellant/bipropellant-hoverboard-firmware/pull/56

closed 2019-07-01 by p-h-a-i-l