bipropellant/bipropellant-hoverboard-firmware #34: Pid control issues

AntumArk, 2019-06-05:
Hello, 
I have been testing PID speed control. I have encountered this weird behavior, where I need to push motor in order for it to move. https://photos.app.goo.gl/Qaz6EH3oQySmb5k48  
Hoverboard is controlled b Ascii.
Could it be motor direction issue ? Position is increasing although motors are spinning backwards.
PWM mode works fine. Motors start spinning at around 200 .

btsimonh, 2019-06-05:
if PWM does not start until 200, then your setup needs a lot more PWM than mine :(.  Mine runs from about 40-45 PWM.

The Speed PID control has a PWM limit on it - SpeedData.speed_max_power.
this seems to be set to 600 in hbprotocol/protocol.c
- but in CAN be set via the protocol as part of the speed command.
- but not by ascii....

However, I would say 600 is enough to move the wheels.  so maybe the PWM required is so much higher that the PID constants need to change significantly?

You can change the speed PID values with
Fskp<number>\n
Fski<number>\n
Fskd<number>\n
Fsl<number>\n

If you find better ones, you can commit them to flash with
Fm1234\n

You can check the current with 
Fskp\n
Fski\n
Fskd\n
Fsl\n

Also check flashed current limit (Fcl\n) - although this would apply in PWM mode also.

Silly question; have you got a full 36v?  Low voltage or restricted current would explain the 200?

AntumArk, 2019-06-05:
Yes, full 36v. I will check other

AntumArk, 2019-06-05:
[image]
So current limit is 1500*100mA?

btsimonh, 2019-06-05:
no I read it as 1500/100 A -> 15A.
(see 'FlashContent.MaxCurrLim' in main.c)

AntumArk, 2019-06-05:
Wiring was messed up. Changed hall sensor and phase wiring, and now it starts at 40-80 Pwm. Trying speed control fixing

AntumArk, 2019-06-05:
Speed control works, only PID tuning is required for one wheel.

closed 2019-06-05 by AntumArk