bipropellant/bipropellant-hoverboard-firmware #37: swd pin location on a YST board

sjamthe, 2019-06-08:
Hi All,

Need some help in determining SWD pins. sorry this is not an issue.
I got a replacement board from ebay but it turned out to be a different microcontroller AT32F403. The board is 
[YST board](https://i.ebayimg.com/images/g/wAkAAOSwJcxc015Z/s-l1600.jpg)

Does any one know is the SWD pins are the same? my ST-link doesn't detect in the layout of green board.

Is there a way I can find out the pins of which is which by probing voltage or something? I blue my old board by connecting 3.3 so want to be really safe.

thanks

AntumArk, 2019-06-08:
I had the same board. SWD pins are same as other( from your photo down- SWDIO, GND, SCLK,3.3).
But if you want to be sure, open Stm32f103rc datasheet for 64 pinout and check continuity with multimeter 

Datasheet - https://octopart.com/stm32f103rct6-stmicroelectronics-8119626?gclid=CjwKCAjwlujnBRBlEiwAuWx4LWjH_Spw4WjkUiuQJqzjS-Zx84roV06VOhMfqGbaoVRG148okSlbExoCDhkQAvD_BwE

sjamthe, 2019-06-08:
Thanks a lot

Turns out my 1st pin (photo down) is GND, then SDIO (PA13), SWCLK (CLK).

Is it ok to power by 3.3 on 4th pin to just test out ST-Link read? or I
should never power by the 4th pin?

btsimonh, 2019-06-08:
I have the 3.3. connected, but I only flash with hoverboard powered (as I flash from an rpizero, which is also powered by the HB).  There was some suggestion that you should NOT connect the 3.3, mentioned in the original repo, not sure where or why.  If you do flash woith hoverboard powered, you need to hold the power button else it powers off!.

AntumArk, 2019-06-08:
You should not connect 3.3 when the board is powered up with battery. My colleagues PC fired because of that. 
You CAN connect 3.3 if you are uploading just using ST-Link.

AntumArk, 2019-06-26:
@sjamthe Are you done with this issue? Can we close it?

sjamthe, 2019-06-30:
I was able to finally connect to the board with the pin config I mentioned (from bottom Ground, SDIO, SWCLK, 3.3) by providing power from st-link.

openocd is able to connect with the device and halt it if I use following (note stm32f1 didn't work f2 worked)

openocd -f interface/stlink-v2.cfg -f target/stm32f2x.cfg

But cannot proceed further. Get the following
invalid command name "stm32f1x"
Info : device id = 0x70050242
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed

sjamthe, 2019-07-01:
I think we can close this issue as this is resolved with lot more details [here](https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/97)

closed 2019-07-01 by sjamthe