bipropellant/bipropellant-hoverboard-firmware #38: WARNING! hoverboard fire hazard!

btsimonh, 2019-06-09:
it's no wonder these are renowned for starting fires:
[image]
On my second hoverboard, the inside edge of the joining tube is RAZOR sharp!
No wonder this came to me in a non working state....
Will dismantle and relieve with Dremmel, and re-insulate.

btsimonh, 2019-06-09:
[image]
now like:
[image]
and cables protected further with this:
[image]

atc1441, 2019-06-10:
That is some nasty failure.

just checked on three boards i have and there every cable i shielded via a tube and the connector tube is not charp at all.
these are the ones i ordered, the cheapest to found in germany for 100€ http://rover.ebay.com/rover/1/707-53477-19255-0/1?icep_ff3=2&pub=5575152713&toolid=10001&campid=5337805879&customid=&icep_item=223515232365&ipn=psmain&icep_vectorid=229487&kwid=902099&mtid=824&kw=lg

i dont have a picture of it because all are disassambled.

btsimonh, 2019-06-10:
I knew they were made cheap, but was very surprised that they had cut & turned the joining tube and left the inside edge with a bunch of swarf on it!.

p-h-a-i-l, 2019-06-15:
Wow, that looks dangerous.. I think the first generations of these boards had a habit of catching fire.