bipropellant/bipropellant-hoverboard-firmware #40: STM32 Utility ST link v2 Dongle not work

uhi2000, 2019-06-13:
Hello,
I have a problem with flashing the firmware with Windows 10. I have connected the USB -ST link v2 dongle as described without 3.3v. the STM32 ST Utility 004 . I always get the error writes "Can not connect to target". I really do not use the 3.3volts?  can not connect to target "why"???
please help....  I really not use the 3.3volts?

AntumArk, 2019-06-14:
Have you powered your board with battery?

uhi2000, 2019-06-14:
I have an external power supply connected.
I have an external 36v power supply connected.
Can I now connect the 3.3 Volts on hoverboard mainboard ????
without that which goes crash....

AntumArk, 2019-06-14:
you need to hold power button when flashing. Otherwise, you need to disconnect power supply, and connect stlink with 3.3v. It should flash.

uhi2000, 2019-06-14:
thanks my friend, I will try it tonight......I hope it works

uhi2000, 2019-06-14:
if I put the USB-Link dongle in the USB port of my PC, then I press the connected button in the STM32 Link Utility. but it always comes the error message "no target connected" of the stick but recognized in the STM32 link. What do I'm wrong ???
as long as there is no connection with the USB stick, I can not flash!!!please help me...

AntumArk, 2019-06-14:
Can you send me a photo of your setup? Have you unlocked your hoverboard?

uhi2000, 2019-06-14:
i am beginner, please tell me where can i upload pictures here?

AntumArk, 2019-06-14:
https://help.github.com/en/articles/file-attachments-on-issues-and-pull-requests

uhi2000, 2019-06-14:
[image]

uhi2000, 2019-06-14:
[image]

uhi2000, 2019-06-14:
[image]

uhi2000, 2019-06-14:
the Hoverboard Mainboard is new..

uhi2000, 2019-06-14:
I wanted to flash it with the Hverboard Hack master

AntumArk, 2019-06-14:
Connection mode - connect under reset (instead of normal).

Also attach how is your system connected(electronics).

Do you have linux system?

uhi2000, 2019-06-14:
[image]

uhi2000, 2019-06-14:
no windows 10

uhi2000, 2019-06-14:
and so I connected it. as described

AntumArk, 2019-06-14:
Connect 3.3v from Stlink to hoverboard and try again. Be SURE that your battery is disconnected!

uhi2000, 2019-06-14:
and put the dongle in  my usb port .This is right..power on and hold button...no akku no power supply on? ???

uhi2000, 2019-06-14:
no power on???

uhi2000, 2019-06-14:
ok, I'll try it right away

AntumArk, 2019-06-14:
Do not connect the battery.

uhi2000, 2019-06-14:
I just do not understand why Niklas Fauth shows the picture with 3 cables instead of 4 connections. that is not correct

AntumArk, 2019-06-14:
Because there are two ways of flashing. Flashing with and without battery. 
Flashing without battery is safer. 

Without battery you also connect stlink 3.3v.
With battery you do shouldn't connect 3.3v from stlink.

uhi2000, 2019-06-14:
I thank you first for your help and will try it out. I hope I get a connection and then can flash

AntumArk, 2019-06-14:
@btsimonh Do you know how to unlock hoverboard on Windows? I only know how to do it in linux.

uhi2000, 2019-06-14:
i dont no

btsimonh, 2019-06-14:
I used OCD on windows too... I think.  Was not easy :(

uhi2000, 2019-06-14:
I think I have the new board--AT32F403RCT6
[image]

uhi2000, 2019-06-14:
I think that does not work with the Hover.Master Hack.-Hex file..or yet?

AntumArk, 2019-06-14:
It is different series controller than original hoverboard. This code is for 103 series. Unfortunately you will need to look for another repository for that black board.

And for your green one, you need linux operating system computer (raspberry pi, or dual boot linux) to unlock your board before you can upload code.

uhi2000, 2019-06-14:
sorry i have the old board stm32f103 .soory but the old mainboard but still no connection

uhi2000, 2019-06-14:
shit

AntumArk, 2019-06-14:
Try using openocd on windows, or get linux

btsimonh, 2019-06-15:
Seems at one time I wrote it up :)
try
https://github.com/btsimonh/hoverboard-firmware-hack/blob/newprotocol/otherboards/YST-6.5-10-PCBA-10/programming.md

btsimonh, 2019-06-15:
@uhi2000 - I can't quite read the CPU identifier - but it's not an ST chip. no?
Maybe it has different programming needs?
A 'true' ST utility will most likely not connect on commercial grounds :).
But OpenOCD should (and should help you identify the chip, I guess).

when you have it running, pls buzz these:
[image]
to the CPU to see what extra lines we have :).  Then document in the wiki the board identification and the likely available pins, if you have the time.
Also this:
[image]

btsimonh, 2019-06-15:
found this:
https://www.eevblog.com/forum/microcontrollers/problems-compiling-code-for-at32f403/
does not help a lot, but worth following.

uhi2000, 2019-06-15:
thanks for help, i have a MainboardSTM32F103RCT6, maybe it's locked

uhi2000, 2019-06-15:
if I put the USB-Link dongle in the USB port of my PC, then I press the connected button in the STM32 Link Utility. but it always comes the error message "no target connected" of the stick but recognized in the STM32 link. What do I'm wrong ???
as long as there is no connection with the USB stick, I can not flash!!!please help me...i have the 3,3volts connected and the mainboard is blinking and piep, and i hold the power-Button!!!

uhi2000, 2019-06-15:
[image]
[image]

uhi2000, 2019-06-15:
[image]

uhi2000, 2019-06-15:
[image]
i have order  now!!!

AntumArk, 2019-06-15:
your wiring is incorrect. 3.3v is correct but others not.

uhi2000, 2019-06-15:
yes is correct

uhi2000, 2019-06-15:
the mainboard is blinking

uhi2000, 2019-06-15:
and piep

uhi2000, 2019-06-15:
but no connected with STM32 or STM8

uhi2000, 2019-06-15:
shit

uhi2000, 2019-06-15:
mybe the USB Dongle is defect??

AntumArk, 2019-06-15:
Your red(swclk) wire goes to last pin in your photo. Last wire should be SWDIO. 

Your brown wire(swdio) goes to third pin in your photo. It should be  GND.

uhi2000, 2019-06-15:
[image]

uhi2000, 2019-06-15:
white03,3v---GND=Black

AntumArk, 2019-06-15:
switch black with brown, then red with brown. And then send me a photo.

uhi2000, 2019-06-15:
i have seen the mistake there are different sicks soory

uhi2000, 2019-06-15:
sorry

uhi2000, 2019-06-15:
[image]
[image]

uhi2000, 2019-06-15:
Great that was the mistake, but I have a new one
[image]

uhi2000, 2019-06-15:
ok its work thanks for help

btsimonh, 2019-06-15:
Yeah! let us know how you get on with the firmware :) - and what your project is going to be....

uhi2000, 2019-06-15:
i use the firmware  hoverboard master hack.

[28 more replies]

closed 2019-06-16 by uhi2000