bipropellant/bipropellant-hoverboard-firmware #42: electrical noise on serial lines?

btsimonh, 2019-06-16:
Would be interested to hear other's experience with serial control and noise on the lines....

I've done quite a bit of (software) investigation, and am coming to the conclusion that my wiring is not thought through enough (laugh at my setup [here](https://github.com/bipropellant/bipropellant-hoverboard-firmware/wiki/btsimonh).

I use software serial, and when the board is under load (e.g. with current limit set at 1A and a motor held still with drive - on the sinsquash PR, so current limiting should be graceful), when I send characters to the board, a LOT are wrong :(.

I also see quite a few sensor misreads at times.

From a software perspective, I'm now running GPIO_SPEED_FREQ_LOW on all FET drive lines, and on the serial RX GPIO, and i think it reduced the noise, but it's still unacceptable....

My next step will be to re-work the physical installation, maybe shield the sensor serial and the software serial lines, and put in thicker ground lines between my pi and the main board.
@EmerickH - did you solve all your comms issues in the end?

EmerickH, 2019-06-16:
(Sorry if I don't reply very quickly: I'm actually preparing an exam so Bipropellant is not my priority! But from ~ the 28 of June I'll be in vacation for 2 month so I'm going to have more time to work on this)

All the problems I had with serial had disappeared after I changed the mainboard to a new one so I think the GD32 was defective... (I think the new one has a STM32 but I'm not sure at all). Now my serial is working great with some NAK sometimes but they are well handled by your NodeRed script so they don't cause any problem!

btsimonh, 2019-06-16:
bought some snap-on ferrite cores on amazon, will try them next weekend :).

btsimonh, 2019-06-25:
Applied ferrite cores to all major wires; the interference is reduced but not eliminated - serial is still not good enough for me on my YST-6.5-10-PCBA-10 board.  I'm still getting sensor mis-reads and software serial mis-reads (nacks on machine protocol).
Next step has to be more detail on the low level PWM drive; maybe scoping the PWM of the original firmware and comparing with our firmware in terms of dead time, drive style, etc.
Another 'next step' may be an RC snubbing network on the software serial lines; at the moment they run direct to the CPU GPIOs, which may be a little naive.
@EmanuelFeru - did you fire up your new fresh board yet?  May be worth doing some analysis of it's output before flashing it?

EmanuelFeru, 2019-06-26:
Yes, I did spin it a little. I did not have time to do intensive measurements. I just checked the motor control behavior factory firmware vs our sinusoidal control. What I can say is that theirs is a bit quieter, not much but a bit quieter. I am quite sure they do FOC, this is the reason I want to look a bit into it.
At the moment we do not use phase current measurements and I think we should.
Would you like to check other things on the factory firmware?

btsimonh, 2019-06-26:
I'm looking to a mod to the firmware which will dump messages on the sensor ports - wanted to try to capture calibration, startup, etc. sent to the sensor boards by connecting two boards together - but if you don't have serial, it's difficult to get the data out :).  (P.S. I think there was an issue with the USART2 serial switching - fix in new branch today, but the new branch still has some discussion to be had).
You could compare the PWM - to see if they drive all three phases or two at a time, and also to check the dead-time.  I got my new 'scope last week, not as special as I hoped, but functional enough (dual channel 20Mhz for £50 - Hantek6066BE).
I've also got a branch with capture of our U/V/W values and the three current values as a diagnostic aid to 'see' what the driver does.  You really need serial :).

p-h-a-i-l, 2019-06-26:
@btsimonh I've been struggling with serial a long time now, still haven't fixed it completely. Too many distractions ;)

Aside from keeping the serial lines short, picking a quiet GND, adding ferrites, shielding wires, keeping wires away from noise and low pass filters,how about using the protocol?
How bad is your package loss? Do you see bursts or single flipped bits?

We could easily add a command cosoleLog to the machine protocol. Retries and ACKs should take care of the rest. Decrease timeout, increase retries, blow up buffer. Maybe bandwith could be a problem at 9600 baud..

btsimonh, 2019-06-26:
I did some diagnostics on the software serial, actually measuring and dumping the bit timing; trying to see if I could digitally filter; but no firm conclusion apart from that the software serial is actually very good (timing wise), given a good signal.  The fact that the 'official' serial to the sensor boards (which should be well designed at both ends in theory) is still screwy leads me to believe that we are making the board noisy in the way we are driving the motors.  Hence I tried with fast and slow GPOs on the FETs etc., but again no banana.  The next thing is to examine the PWM from the original firmware, and compare to ours - beyond that we're on to extra soldered filtering (modding the board).
Are you GD32 or STM32?  What board ID?
Ref the protocol, I got > 10% nacks on messages every 500ms in 'normal hoverboard action' in my livingroom - i.e. not too much current at any time. (about 450messages, only about 1/2 of which occurred whilst 'riding').

EmanuelFeru, 2019-06-26:
The new board that I have is GD32 with board ID: **JKL-PHC02-A-V02 2016-07-02** 1628
Regarding the PWM, I am also interested to check out the wave form. For me it would be a bit tricky because I have a 1 channel cheap oscilloscope... I should upgrade at some point.

By the way, I confirm the non-zero torque at 0 angle. So when motors are enabled with input 0, also in factory firmware there is some holding torque and wheels are not freewheeling.

For the noise we see on serial, maybe you can check with your oscilloscope. I am not an expert, it could be that the Software serial lines are close or next to PWM driving lines from the micro controller. And there is some antenna effect above a certain frequency. Maybe trying different baud rates helps.