bipropellant/bipropellant-hoverboard-firmware #44: Turn off Timeout fires at half the time.

atc1441, 2019-06-17:
I just made some tests and deactivated the super loud beeping when you do nothing for a few minutes,

what i found out after deactivating,even when in Platformio.ini the 

> -D INACTIVITY_TIMEOUT=8

 is set the main board turns of at exactly 4 minutes and not 8.

is that correct or is there something else wrong?

i know added a timeout function for the machine protocol, when longer than 200ms not command is received it will turn the motors of and will beeo for 100ms. this does only work on the setpwm cmd but is also easy to add to other cmds.

AntumArk, 2019-06-25:
@atc1441 Where have you implemented machine protocol timeout? Main repo?

btsimonh, 2019-06-25:
I was thinking a dedicated command to set the timeout - e.g. set a timeout variable to a number of ms, decrement every 16 cycles of DMA interrupt if > 0, when it hits zero, disable.
Initialise to -1.
This would then mean that you could set the timeout to what you want (and if you want), and disable at any time.
Also @p-h-a-i-l has a watchdog implementation, I believe similar, but not dependent on the DMA interrupt.  Was waiting to see his implementation :).

AntumArk, 2019-06-25:
@btsimonh So when a command is received, it just resets? I am guessing timeout should be over 500ms, because timeout1 is 500ms?

atc1441, 2019-06-25:
@AntumArk this evening I could Upload here the changes I made to at the time out.  if you want.

It was in the Maine Repo but very simple changes

btsimonh, 2019-06-25:
I was thinking a dedicated command 'reset timeout' which contained the desired timeout value in ms as a short, so allowing for -1 (disabled) and ~0-32000, so up to 32s possible.

atc1441, 2019-06-25:
@btsimonh it would be very nice to have a dedicated timeout command maybe an extra command for which command the Timeout to reset.
 so that's not every command that is received resets the timeout

AntumArk, 2019-06-25:
in https://github.com/bipropellant/bipropellant-protocol/blob/14a52e2d3c10cea96944cf04c9baf723f4f9a7de/protocol.c#L57 
`extern volatile uint32_t timeout; // global variable for timeout
,` we have global **timeout.** which seems to reset at some write functions(fn_SpeedData,fn_PositionIncr,fn_PWMData and after attempt to read other control types(does not matter if it is correct or not) ). I can't seem to find what increments it. Could it be that it is from original repo?

Could we call fn_enable with disable, when timeout occurs? And to increment when protocol state is idle?

p-h-a-i-l, 2019-06-25:
Timeout was a bit confusing, but it got fixed recently in the hoverboard-firmware-hack repo. Back then `timeout` was only incremented in the Nunchuk code. 

Now `timeout` is incremented in the main loop and can be used for all inputs
Additionally `ppm_timeout` (incremented in systick) is used to process PPM input.

Quick fix would be to adopt @larsmm changes from the hoverboard-firmware-hack repo, but I'd like to go further as described above. Also we shouldn't call the variable "timeout", very worn out name, used everywhere.

AntumArk, 2019-06-25:
How about we call it `input_timeout`?

atc1441, 2019-06-25:
Here are in short the changes i made to get a Timeout when more than 100ms no PWMdata command is received: 

In File protocol.c in the hbprotocol:

> At line 336 insert:  extern volatile uint32_t timeoutPWM;
> At line 348 instert: timeoutPWM = 0;

In File main.c:

> At line 43 insert: volatile uint32_t timeoutPWM;
> At line 475 insert: timeoutPWM++;
> At line 732 insert: if (timeoutPWM> (200) / DELAY_IN_MAIN_LOOP){pwms[0] = pwms[1] = 0;}

as this issue says on my board everything seems to be at double the speed so here 200ms is in reality only 100.

i removed the code for the short beep from this example it will work this way, this is just a very quick solution.

AntumArk, 2019-06-25:
How do you fork branch with reference repos? It is kind of messed up. I keep on trying to fork it, but repos are getting out of sync.

p-h-a-i-l, 2019-06-25:
Use your normal git folder where you are working all the time
if the branch you are trying to use as a base is in a different fork, add the remote (git remote add ..)
check out the branch you want to use as base (git checkout ..)
create new branch based on the current workng copy (git checkout -b yournewbranchname)

AntumArk, 2019-06-25:
> Here are in short the changes i made to get a Timeout when more than 100ms no PWMdata command is received:
>
> In File protocol.c in the hbprotocol:
>
>
> In File main.c:
>
>
> as this issue says on my board [...]

Your edits works quite well. I used it for speed control. I haven't tested timing, but I made that wheels would slow down gradually instead of sudden halt. Plugging off USB works quite well. And it gets back on after 2 seconds. 
`       case FN_TYPE_PRE_WRITE:
            control_type = CONTROL_TYPE_SPEED;
            timeout = 0;
            timeoutSpeed = 0;
            break;    ` - at protocol.c
`case CONTROL_TYPE_SPEED:
              for (int i = 0; i < 2; i++){
 if (timeoutSpeed> (200) / DELAY_IN_MAIN_LOOP){ 
   if(SpeedData.wanted_speed_mm_per_sec[i]>0)
                SpeedData.wanted_speed_mm_per_sec[i]--;
                 if(SpeedData.wanted_speed_mm_per_sec[i]<0)
                SpeedData.wanted_speed_mm_per_sec[i]++;} ` - at main.c

atc1441, 2019-06-25:
@AntumArk that is a nice idea to have it slowly slow down. 

But i think i will leave it for me as an abrupt stop because it in that stage where for example the esp32 is not powered anymore, it is important to have it stop imidieatly.

p-h-a-i-l, 2019-06-25:
The default is not a abrupt stop, when you set `enable = 0`, the motors start coasting (freewheeling, no load. no braking, no acceleration). Its just friction which stops them.
This is also intended behaviour, sometimes a sudden braking is unsafe. Also when communication recovers, you can continue where you left (small pause vs brake bursts)

atc1441, 2019-06-25:
@p-h-a-i-l yes that is what i am doing.

AntumArk, 2019-06-26:
I have made time-out function based on @atc1441 changes. 
Here is the demonstration https://photos.app.goo.gl/V6zawA2GK3L7XRNs5

p-h-a-i-l, 2019-06-26:
Could you make a PR or provide a link to your code?
This behaviour looks for me like a good emergency stop. But when I have a timeout, I expect communication to recover and I want no active braking. Maybe this can be activated when timeout persists for >1s?

AntumArk, 2019-06-26:
@p-h-a-i-l  you can check it out. Be warned that i am quite raw with PR :D

atc1441, 2019-06-26:
I just also made a video to show the Timeout function in "real life" it is online here: https://youtu.be/vWcUI8V5Epk

btsimonh, 2019-06-26:
:) looking good.

verial, 2019-08-13:
can you publish a sketch using the LCD screen?

verial, 2019-08-13:
can post the code for the arduino using the LCD screen?

atc1441, 2019-08-13:
> can post the code for the arduino using the LCD screen?

Hi, unfortunately the code is super unreadable and only works with  a library is changed a lot, i will publish the code with the display running when it nicer :)