bipropellant/bipropellant-hoverboard-firmware #47: Update protocol submodule to master branch from #p27

AntumArk, 2019-06-26:
@btsimonh Are you working to update protocol submodule? I am just trying to remind you in case you may forget.

btsimonh, 2019-06-26:
I think it's done (it was a few mins after the main repo....)?  pls confirm it all builds once both masters are pulled....

AntumArk, 2019-06-26:
[image]
It is still detached head. 

Building things now :D

btsimonh, 2019-06-26:
subrepos - bahh...  trying to fix it now!

btsimonh, 2019-06-26:
pls check again.  all should be good now.

AntumArk, 2019-06-26:
@btsimonh Everything built successfully.

AntumArk, 2019-06-26:
Ok, now it is latest commit, but still detached head. 
[image]
You can't make PR to detached heads.

btsimonh, 2019-06-26:
hmm...   not seeing that through github or sourcetree....
Are you working from btsimonh or bipropellant?  - they should be the same, but the btsimonh one is a little of a plaything.

AntumArk, 2019-06-26:
bipropellant 

Cloning with reccursive submodules using gitbash, then adding all repos to github desktop.

btsimonh, 2019-06-26:
ok, I'll try a small mod on each and commit from github, change to readme maybe.  one moment.

AntumArk, 2019-06-26:
In the meantime I am trying to clone on Sourcetree

AntumArk, 2019-06-26:
Same thing, it checks out on that specific commit. 
[image]
Although it can handle submodules, where github desktop - not.

p-h-a-i-l, 2019-06-26:
Well, detached head is "ok" when working with submodule. A submodule does not reference a branch or head, it references a single commit. This way you can make sure that the "main" repo decides which commit is ok.

You can open git in your submodule folder and checkout the correct branch. This way the head shouldnt be detached anymore.

AntumArk, 2019-06-26:
@btsimonh It is now synced. Head is now synced upon cloning. 
[image]

EmanuelFeru, 2019-06-26:
@btsimonh, maybe you can update the Travis link in Readme to point to the bipropellant build check. At the moment it links to my fork Travis build.

p-h-a-i-l, 2019-06-26:
> @btsimonh, maybe you can update the Travis link in Readme to point to the bipropellant build check. At the moment it links to my fork Travis build.

done

btsimonh, 2019-06-26:
> In the meantime I am trying to clone on Sourcetree

I use sourcetree for all my windows git operations; it's not perfect for submodules, but I'm kind of getting used to it :).

p-h-a-i-l, 2019-06-26:
I gave up on using a fancy GUI for git. Just using the built in functions from vscode and git bash.

closed 2019-06-26 by AntumArk