bipropellant/bipropellant-hoverboard-firmware #60: Take off problems

p-h-a-i-l, 2019-07-03:
I noticed some starting problems, sometimes I wasn't able to pull of without pushing forward with my foot or moving the vehicle a little bit and trying again. 

Wheels: 10" Air
Load: ~100 kg (vehicle and driver)
Road conditions: bumpy

I was using my watchdog branch, Analog input (=pwm directly).
Default config, but phase advance disabled.

AntumArk, 2019-07-05:
I am experiencing similar problem. My left wheel is barely holding any load. It often needs some push to start spinning, and it takes almost no force at all to stop it.

btsimonh, 2019-07-05:
> I am experiencing similar problem. My left wheel is barely holding any load. It often needs some push to start spinning, and it takes almost no force at all to stop it.

check your phase wires, maybe you are simply missing one?  not sure if it would run on two phases with a push?

AntumArk, 2019-07-05:
Checked the wheel on a different board, had no problems. And used this to check phase wires with continuity in case i have messed up the order. Order was correct and wires were not broken.

AntumArk, 2019-07-05:
Using two phase wires on a stand (6.5inch motors), works fine, and has quite a lot of torque. So missing phase is not a problem.

AntumArk, 2019-07-05:
I am suspecting, that 3 FETs have burned.

AntumArk, 2019-07-09:
Fixed my problem. It's bad wire connection of one phase wire.

jpiat, 2019-10-15:
I have a hoverboard with 10" Air wheels and i encounter the same issue. When driving forward, the left wheel is weak and can easily be stopped while teh right wheel works fine. When driving backward, the right wheel is weak but the left wheel is fine ... Could those 10" air wheel be a lot different and require different timing?