bipropellant/bipropellant-hoverboard-firmware #62: No more being able to control HB

EmerickH, 2019-07-04:
Hello everyone,
I'm back and I see you made a lot of improvements!
I updated today the firmware of my HB but I have a problem:
Using `USART2_CONTROLLED` (or your predefined usart2 PIO env) I'm no more able to control speed or posn inc of the robot...
The serial is replying correctly to test buffer, hall position request, ... but when I send a speed command nothing happens...
Have you got any idea of what's not working?
Also, the board is sending some [119,13,1] buffers ("Sensor (Hoverboard mode)" ?), maybe something it is going into hb mode even if `USART2_CONTROLLED` is used?
Thank you!

p-h-a-i-l, 2019-07-06:
wlecome back @EmerickH !
Areyou using machine protocol or ASCII mode?

EmerickH, 2019-07-06:
Machine protocol, I also tested with the new NodeRed example, same problem...

p-h-a-i-l, 2019-07-06:
looks like input_timeout_counter isn't reset for speed

p-h-a-i-l, 2019-07-06:
What debugging capabilities do you have right now? 

You can first try if setting PWM works for you. Maybe manually enforce enable=1 and input_timeout_counter=0 in the main?
Also see if you can get ASCII commands to turn the wheels.

EmerickH, 2019-07-19:
Sorry for the late reply...
I started the robot today, and it's working fine! I've not changed anything, I've not changed anything!
Weird but nice!

EmerickH, 2019-07-19:
OK, after playing a little bit with it the problem come back sometimes, I need to restart the hb or wait a little bit  for it to listen to my commands.
It seems like when the wheels are not turning for ~30s it goes in a "sleep" mode and I need to send it a lot a commands before it "wakes up"

EmerickH, 2019-07-23:
Ok I found something: this is happening when I write the max current limit flash value: even if I send the same value as it is by default, it's crashing...
Also speed PID seems to do nothing, even with P=0 the hoveboard is moving!

p-h-a-i-l, 2019-07-23:
Can you check if writing the set point by ASCII protocol has the same problems as with machine protocol?

w4ff3l, 2019-08-20:
Heyo, I seem to have the same problem. After updating the firmware, communication over machine protocol with usart2 is no longer working, although i receive an answer after using send or requestread from HoverboardAPI. I tried using the ascii protocol but all i get back is strange chars.

p-h-a-i-l, 2019-08-27:
Are you using the same baud rate on both sides?

w4ff3l, 2019-08-27:
I actually found the problem... One of my jumper wire cables was broken. Btw thanks for the awesome HoverboardAPI :)

p-h-a-i-l, 2019-08-30:
Thanks, you're welcome!