bipropellant/bipropellant-hoverboard-firmware #63: Compilation issues when using ADC functionality

barnyrelph, 2019-07-06:
Firstly, thanks for creating and updating this project!

I've been looking at the ADC functionality added by @p-h-a-i-l  in PR #55 , as it's exactly what I'm after (previously I was looking at the BBCar repo). 

I've followed the Wiki page on ADC and Defined CONTROL_ADC, along with setting the ADC values for single-pot mode, but I get the following compilation error:
```In file included from src/setup.c:39:0:
inc/config.h:489:6: error: #error SERIAL_USART2_IT not allowed, cable already in use.
     #error SERIAL_USART2_IT not allowed, cable already in use.
      ^~~~~
make: *** [build/setup.o] Error 1
```

I've attempted to #undef DEBUG_SERIAL_USART2 and SERIAL_USART2_IT (both together and individually) but that results in multiple compilation errors. 

Is there something I'm missing in order to compile in ADC Mode?

barnyrelph, 2019-07-06:
[image]
This is the sum total of my changes to config.h if that helps

p-h-a-i-l, 2019-07-06:
Hi @barnyrelph,

there are some presets for configurations on the top of config.h. By default https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/67c9c47059b740a1ddebdc64bb99ee32f099cfa8/inc/config.h#L19 is used.

If you want to deviate from that (as you have to for analog input), you have to create a new set of presets or disable presets completely ( `#define CONTROL_TYPE = 0` for example)

If you like, you can make a PR with a new Analog input preset.

Hope to hear from your experience, most of the Analog examples are not tested yet.

barnyrelph, 2019-07-06:
Thanks! I'll have a go at defining my own control type and see how I get on.

barnyrelph, 2019-07-13:
I've just spent an evening trying to understand what's happening with setting up CONTROL_TYPE 0. 

It appears to be leaving main.c with an extra closing `}`, somewhere inside `main()` but for the life of me, I can't spot how that's happening. 

I've got a fork with my config.h up at https://github.com/barnyrelph/bipropellant-hoverboard-firmware

I suspect my C is more rusty than I'd like :)

p-h-a-i-l, 2019-07-15:
found the culprit: https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/5d296378f30891fdbd40b45df61b52666f4f4cdc/src/main.c#L702

There are waaay to many #ifdefs in the code...
I made a PR to your repo: https://github.com/barnyrelph/bipropellant-hoverboard-firmware/pull/1

It's probably been a very long time since someone compiled the project without protocol :) Glad if you can do some testing.

Next steps: 
- create sample config for analog input in config.h
- create pio environment without protocol so our CI can complain about compilation errors
- get rid of the #ifdef jungle..

p-h-a-i-l, 2019-07-15:
@btsimonh do we need HALL_INTERRUPTS and FLASH_STORAGE optional? I'd quickly remove all #ifdefs and make it permanent..
And since I'm there, remove old debug stuff: LOG_PWM, MINIMUMCODE and DOTX

barnyrelph, 2019-07-15:
Boom! The project now builds successfully! Thanks very much for looking at that for me, I just wasn't seeing where the error came from.
I'll get the code onto my board in the next day or two and run some tests to see if I can get the wheels spinning. 

Thanks again

Barny