bipropellant/bipropellant-hoverboard-firmware #64: Unwanted PowerOn problem

AntumArk, 2019-07-08:
I have this odd problem, when board powers on as soon as the battery is connected. You could expect this if the power button is faulty, but no! It reacts to button presses and turns off to endless bleep. Tried with original firmware and got same result. 
Do you have any idea what went wrong?

p-h-a-i-l, 2019-07-08:
one of the TIP127 is damaged

AntumArk, 2019-07-08:
> one of the TIP127 is damaged

Ok, I will check it out.

p-h-a-i-l, 2019-07-08:
[image]
This one probably.

AntumArk, 2019-07-08:
To what level resistance should vary compared to healthy one?

AntumArk, 2019-07-08:
I got 450 and 770 on outer legs. While new ones has 956.

p-h-a-i-l, 2019-07-08:
I'd check with diode mode, not resistance

AntumArk, 2019-07-08:
They seem to be fine. No shorts.

p-h-a-i-l, 2019-07-08:
Weird. Thats's exactly the behaviour I see when that part is damaged. Replaced quiete a few already..
Did you check the middle leg in diode mode?

AntumArk, 2019-07-08:
Yes, both are fine. Replacing them to check if they were bad quality

AntumArk, 2019-07-08:
Yep. Replaced both of them and now it works. So changed resistance is also a sign of burned or damaged tip127. I suggest to add this to documentation.

closed 2019-07-08 by AntumArk

p-h-a-i-l, 2019-07-08:
> Yep. Replaced both of them and now it works. So changed resistance is also a sign of burned or damaged tip127. I suggest to add this to documentation.

Great, please do!

btsimonh, 2019-07-08:
I didn't think there was a restriction :). - i'm fine with every contribution...

AntumArk, 2019-07-08:
In progress.

AntumArk, 2019-07-08:
How do you add photo from google photos? It cant seem to load

btsimonh, 2019-07-08:
what id do is:

AntumArk, 2019-07-08:
https://github.com/bipropellant/bipropellant-hoverboard-firmware/wiki/Hardware-troubleshooting Check it out

p-h-a-i-l, 2019-07-08:
Looking good, thanks!

sjamthe, 2019-07-08:
This is not directly related but may be you can help.
I am using usart2 to drive my hoverboard
As soon as usart us connected it powers on without me pressing power button.
When the timeout kickin it powers off and on again.

How do I change this behavior? I can't see in code what signals power on on
usart ?

AntumArk, 2019-07-09:
What do you mean by connected? physically or opened a port?

btsimonh, 2019-07-09:
Maybe you are (by mistake) supplying power down the red wire?  I'd imagine if you feed 5v down the red wire, the board MCU would power up (because you're feeding enough to the input of the 12v->3.3v).  Once the board powers up, it would then feed 12v back at you, possibly smoking the device feeding the power!.  (but an interesting idea for anyone who wants 'wake on lan' functions! - feed it from a diode & control the voltage fed....).

sjamthe, 2019-07-09:
Just checked my connections.
1. I am not supplying power from UART. Have only connected GND, TX, RX
2. I AM connected to the battery power,
3. As soon as I connect the USB-Serial cable to my raspberry pi and if
raspberry pi is powered on the hoverboard wakes up with a chime as if I
have pressed the power on button.

So obviously some signal is coming over over RX to signal the
micro-controller to come out of power off mode. I have not tried debugger
so not sure how. Also, I am using code from Tom's branch
<https://github.com/sjamthe/hoverboard-firmware-hack> on AT32 which is
behind all the changes here (like no PID etc.)

p-h-a-i-l, 2019-07-09:
As this is a separate Issue, I just creatde a new one :)
For future googlers!

sjamthe, 2019-07-09:
Thanks. I guess probably I don't really understand how the power on / off
works.
I noticed I copied timeout=0 in my settings just as it was in ADC control.

When the power off is called does the mcu go to sleep state ? Then does the
power button pin wakes it up when pressed ?

Is it possible that uart being connected wakes the mcu ? Just a guess.