bipropellant/bipropellant-hoverboard-firmware #65: power on over uart

p-h-a-i-l, 2019-07-09:
This is not directly related but may be you can help.
I am using usart2 to drive my hoverboard
As soon as usart us connected it powers on without me pressing power button.
When the timeout kickin it powers off and on again.

How do I change this behavior? I can't see in code what signals power on on
usart ?

On Mon, Jul 8, 2019, 6:23 AM phail <notifications@github.com> wrote:

> Looking good, thanks!
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/64?email_source=notifications&email_token=AAXUAJYKNXPC5BMESTBS7D3P6M5VHA5CNFSM4H62QU32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZNCCXI#issuecomment-509223261>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AAXUAJYIJFCCAYCZ3SEYIIDP6M5VHANCNFSM4H62QU3Q>
> .
>

_Originally posted by @sjamthe in https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/64#issuecomment-509430905_

p-h-a-i-l, 2019-07-09:
What do you mean by connected? physically or opened a port?

_Originally posted by @AntumArk in https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/64#issuecomment-509487972_

p-h-a-i-l, 2019-07-09:
Maybe you are (by mistake) supplying power down the red wire?  I'd imagine if you feed 5v down the red wire, the board MCU would power up (because you're feeding enough to the input of the 12v->3.3v).  Once the board powers up, it would then feed 12v back at you, possibly smoking the device feeding the power!.  (but an interesting idea for anyone who wants 'wake on lan' functions! - feed it from a diode & control the voltage fed....).

_Originally posted by @btsimonh in https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/64#issuecomment-509491753_

p-h-a-i-l, 2019-07-09:
Just checked my connections.
1. I am not supplying power from UART. Have only connected GND, TX, RX
2. I AM connected to the battery power,
3. As soon as I connect the USB-Serial cable to my raspberry pi and if
raspberry pi is powered on the hoverboard wakes up with a chime as if I
have pressed the power on button.

So obviously some signal is coming over over RX to signal the
micro-controller to come out of power off mode. I have not tried debugger
so not sure how. Also, I am using code from Tom's branch
<https://github.com/sjamthe/hoverboard-firmware-hack> on AT32 which is
behind all the changes here (like no PID etc.)

_Originally posted by @sjamthe in https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/64#issuecomment-509521708_

p-h-a-i-l, 2019-07-09:
Looks like you are somehow providing enough energy to wake up the microcontroller.

Here is my guess:
There are port diodes on each I/O. Just little diodes going to GND and VCC of the micro. These should protect the inputs from negative voltages and spikes.
When the board is off, VCC is also 0V. So every voltage fed into an I/O goes directly to VCC. 
If there is enough power, VCC might rise to a level where the microconbtroller can wake up.
As soon as the microcontroller wakes up, it switches on self-hold.

Sooo. We could change the code to validate if the power button is pressed on bootup? (Maybe it alread does?) If not, wait for it to be pressed. Or reset.
Or you could try to limit the power transmitted. Maybe with a series resistor in line with TX (RX when seen from the board)

btsimonh, 2019-07-09:
& double check the line voltages - they should be 3.3v, not 5v (or god forbid +-12v RS232!).

p-h-a-i-l, 2019-07-09:
> & double check the line voltages - they should be 3.3v, not 5v (or god forbid +-12v RS232!).

Yeah for sure! (If it really is a raspberry which connects, it schould be 3.3V)

btsimonh, 2019-07-09:
@sjamthe - it's a physical power circuit; there is a schematic somewhere...  so the MCU is definitely completely powered off normally.
p.s. watch out for the power button connector - it carries 36v - i.e. beware of connecting anything else to it or trying to control it remotely without investigation (I torched a CPU like that...).
You MUST be suffering some form of parasitic power.  Sure you don't have power attached to the programming connector?  I DO, but my pi is powered from the controlled 12v, so when the board powers down, so does my pi (and I don't use usart2... but phail does).

p-h-a-i-l, 2019-07-09:
I think I've seen something similar..
Need to check again, but I with a Maple Mini board connected to UART the board powered up, with ESP32 not. Might be different output pin drivers in the micro?

p-h-a-i-l, 2019-07-09:
> Thanks. I guess probably I don't really understand how the power on / off works. I noticed I copied timeout=0 in my settings just as it was in ADC control. When the power off is called does the mcu go to sleep state ? Then does the power button pin wakes it up when pressed ? Is it possible that uart being connected wakes the mcu ? Just a guess.

Timeout is not related directly to poweroff.
On shutdown the microcontroller disables the self-hold transistor (switching off the supply) and enters a infinite loop, waiting for power to disappear.

sjamthe, 2019-07-10:
Thanks for all the responses. 
I am using a USB 2.0 to TTL UART Module which send 3.3v on Tx as soon as it is powered on.

Earlier I was using a Teensy instead and had no problems. We can close this issue.

closed 2019-07-10 by p-h-a-i-l

DaviVolcan, 2019-07-28:
Does the mcu have clamping diodes ? I have already seen some mcu's being powerd even with vcc not connected... https://encrypted-tbn0.gstatic.com/images?q=tbn%3AANd9GcQHVE5K7hTe2BZX4bqGEIkXV_uljXVMfPTku8guPVVbiRHNe9ev

p-h-a-i-l, 2019-07-28:
I think they have, but I didn't look into the datasheet to verify.
When I wrote "port diodes" above, I was talking about the clamping diodes you drew. Thanks for clarifying.