bipropellant/bipropellant-hoverboard-firmware #67: sudden crash, battery drain with ascii protocol Ip

sjamthe, 2019-07-27:
I was testing ASCII Immediate protocol Is worked well, but when I tried Iw the motors started at 100, 200 and at 400 suddenly the motors stopped abruptly.

I noticed system was off so I tried to restart it but no luck. Checked voltage and it had suddenly dropped to 20V and dropping very fast.

No here are some observations
1. Even when power is turned off my right wheel is somewhat rigid but if I remove the yellow power cable to the wheel it turns ok when turned by hand.
2. When charger is connected battery doesn't charge (charger light stays green) 
3. I am able to power of the microprocessor and use ascii protocol to to run some commands like ?, G, C etc. But as soon as I send any Immediate commands to turn motors the microprocessor powers off without power-off tune.

Questions?
1. Is my board fried? Why is it not charging or turning?
2. How can I test?
The battery is getting 41V but still not charging. I will leave it on overnight and see.

thanks

sjamthe, 2019-07-27:
The battery is getting 41V but charging slowly. I measured current going to the battery and it was 3mA only. 
I replaced that board with my AT32 board and battery is charging.

What is limiting current on GB32 board ? Looking at schematic I see 6A10 diode but that looks ok.
The resistance between two charging terminals is about 33K which doesn't look bad.
Any ideas how to debug?

thanks

sjamthe, 2019-07-27:
may be by battery is discharged 100% so not charging again. Not sure how that happened though.

btsimonh, 2019-07-27:
it does sound like two of your FETs went short circuit - so shorting the battery .  I've not had any kind of failure like this, so can't advise more :(.

sjamthe, 2019-07-27:
Thanks for the pointers. Let me poke there and see which ones.

p-h-a-i-l, 2019-07-27:
The protection circuit of the batter probably kicked in at one point, that's why you see the voltage dropping quickly. I think the battery protection resets by unplugging the load.

sjamthe, 2019-07-29:
you are correct about the battery. Battery protection didn't reset but I played around with the BMS by removing it and putting back and that worked.

The battery is back up and I am able to go back to my AT32 board.
I wish I had code on AT32 that supports bipropellent-protocol (it doesn't for now)

closed 2019-07-29 by sjamthe

p-h-a-i-l, 2019-07-29:
Which repo are you using for AT32? Maybe you can merge both projects to get the best from both worlds :)

sjamthe, 2019-07-29:
I forked off of TomTinkering code , Here is my git https://github.com/sjamthe/hoverboard-firmware-hack

Last I checked he was the only one with AT32 changes, Is there a more later repo I can start with?

I tried to merge but there are a lot of diffs and I was not successful in 1st attempt. I think UART is DMA in mine and uses IT in final repo, may be some mistake there.

juodumas, 2019-07-30:
@sjamthe could you please share how you flashed your AT32 board? I wasn't able to and have an open issue here: https://github.com/cloidnerux/hoverboard-firmware-hack/issues/3

sjamthe, 2019-07-31:
@juodumas I modified openocd code to detect AT32 device id (code below).
Then take a look at my [Makefile](https://github.com/sjamthe/hoverboard-firmware-hack/blob/at32_master/Makefile) that I use to unlock and flash code.
make load #load's the flash
make unlock #unlocks it (You may have to try other commands to unlock if this doesn't work).

--- changes to openocd ----
diff --git a/src/flash/nor/stm32f1x.c b/src/flash/nor/stm32f1x.c
index cf10e374..41b41bda 100644
--- a/src/flash/nor/stm32f1x.c
+++ b/src/flash/nor/stm32f1x.c

        /* set page size, protection granularity and max flash size depending on family */
        switch (device_id & 0xfff) {
@@ -762,6 +763,12 @@ static int stm32x_probe(struct flash_bank *bank)
                stm32x_info->default_rdp = 0xAA;
                stm32x_info->can_load_options = true;
                break;
+       case 0x242: /* AT32F403 added by sjamthe */
+               LOG_WARNING("Found 0x700 for ATF32F403");
+               page_size = 2048;
+               stm32x_info->ppage_size = 2;
+               max_flash_size_in_kb = 512;
+               break;
        case 0x446: /* stm32f303xD/E */
                page_size = 2048;
                stm32x_info->ppage_size = 2;

juodumas, 2019-07-31:
Thank you @sjamthe! Confirming that it works. Can you contribute this back to openocd?

After patching `opencd` master, this worked for me:
```
diff --git a/Makefile b/Makefile
index bb062f4..76a1a77 100644
--- a/Makefile
+++ b/Makefile
@@ -244,11 +244,15 @@ clean:
 flash:
        st-flash --reset write $(BUILD_DIR)/$(TARGET).bin 0x8000000

+flash-at32:
+       openocd -f interface/stlink-v2.cfg -f target/stm32f3x.cfg -c init -c "reset halt" -c "stm32f1x unlock 0" -c "flash write_image erase $(BUILD_DIR)/$(TARGET).hex 0 ihex" -c "shutdown"
 flash-jlink:
         JLink.exe -if swd -device Cortex-M4 -speed 4000 -SettingsFile .\JLinkSettings.ini -CommanderScript jlink-command.jlink

 unlock:
        openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "stm32f1x unlock 0"
+unlock-at32:
+       openocd -f interface/stlink-v2.cfg -f target/stm32f3x.cfg -c init -c "reset halt" -c "stm32f1x unlock 0"

 #empty target for MCU selection
 AT32: all
```

Brnpimenta, 2021-06-04:
@sjamthe 
could you describe the steps you take to flash and reprogram the AT32 + st link?
as begginer with this type of MU, I didn't understand pretty much all of the it, what software you used?
Really appreciate it!

sjamthe, 2021-06-05:
Sorry, it has been more than an year since I played all this.
I think I used stlink utilities running on my raspberry pi as well as
macbook (when necessary) to load the MU.
You need a stlink loader hardware. They are very cheap on Amazon/ebay but
also break very easily. I must have lost at least 3-4

Brnpimenta, 2021-06-06:
@sjamthe , thank you so much for your response!
I do have a st-link v2 to recognize/reprogram the hoverboard board with AT32 chip.
but I saw your comment that you changed the openocd in order to recognize the chip. Would be great if you describe the steps you took in order to detect the chip.

sjamthe, 2021-06-07:
Ah, I remember now.
Yes, I submitted a patch for openocd. I don't think it was merged but you
can apply the patch.
Here it is
http://openocd.zylin.com/#/c/5269/

Amrfathy1, 2024-02-17:
where should I apply these changes because i dont understand @sjamthe @juodumas

suzy311278, 2025-11-22:
suzy@suzy:~/openocd_build/openocd$ openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "stm32f1x unlock 0"
Open On-Chip Debugger 0.12.0+dev-02293-g3e7a7092d-dirty (2025-11-22-19:39)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : DEPRECATED: auto-selecting transport "swd (dapdirect)". Use 'transport select swd' to suppress this message.
Info : STLINK V2J28S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.204197
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : SWD DPIDR 0x2ba01477
Info : [stm32f1x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : [stm32f1x.cpu] Examination succeed
Info : [stm32f1x.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Info : device id = 0x70030240
Warn : Found AT32F413 device id
Warn : STM32 flash size failed, probe inaccurate - assuming 256k flash
Info : flash size = 256 KiB
Error: Fail reading CTRL/STAT register. Force reconnect
Error: Fail reading CTRL/STAT register. Force reconnect
Error: Failed to read memory and, additionally, failed to find out where
Error: failed to get read pointer
Error: Fail reading CTRL/STAT register. Force reconnect
Error: Fail reading CTRL/STAT register. Force reconnect
Error: Failed to write memory and, additionally, failed to find out where
Error: Fail reading CTRL/STAT register. Force reconnect
Warn : [stm32f1x.cpu] target was in unknown state when halt was requested
Error: Fail reading CTRL/STAT register. Force reconnect
- anyone please help