bipropellant/bipropellant-hoverboard-firmware #68: Compiling API on X86 (or non Arduino)

ducktrA, 2019-07-28:
Has somebody compiled the API on X86? It's a bit a puzzle still. Has anybody gotten beyond that point already? I get the millis() are to be replaced, but where is setup_protocol(). Cannot find it.

Tx

-T430s:~/bipropellant-hoverboard-api/my⟫ g++ read.cpp ../src/HoverboardAPI.cpp ../src/hbprotocol/protocol.c ../src/hbprotocol/machine_protocol.c ../src/protocolFunctions.c ../src/hbprotocol/ascii_protocol.c -I ../src -I ../src/hbprotocol -lserial -std=c++11 -Wwrite-strings -Wnarrowing -fpermissive

...............

../src/hbprotocol/protocol.c:322:1: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
../src/hbprotocol/protocol.c:322:1: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
../src/hbprotocol/protocol.c: In function ‘void protocol_process_message(PROTOCOL_STAT*, PROTOCOL_MSG2*)’:
../src/hbprotocol/protocol.c:433:67: warning: invalid conversion from ‘void*’ to ‘unsigned char*’ [-fpermissive]
                     unsigned char *src = params[writevals->code]->ptr;
                                                                   ^
../src/hbprotocol/protocol.c:458:68: warning: invalid conversion from ‘void*’ to ‘unsigned char*’ [-fpermissive]
                     unsigned char *dest = params[writevals->code]->ptr;
                                                                    ^
../src/hbprotocol/protocol.c:498:68: warning: invalid conversion from ‘void*’ to ‘unsigned char*’ [-fpermissive]
                     unsigned char *dest = params[writevals->code]->ptr;
                                                                    ^
../src/hbprotocol/machine_protocol.c: In function ‘void protocol_send_nack(int (*)(unsigned char*, int), unsigned char, unsigned char)’:
../src/hbprotocol/machine_protocol.c:262:53: warning: narrowing conversion of ‘som’ from ‘unsigned char’ to ‘char’ inside { } [-Wnarrowing]
     char tmp[] = { som, CI, 1, PROTOCOL_CMD_NACK, 0 };
                                                     ^
../src/hbprotocol/machine_protocol.c:262:53: warning: narrowing conversion of ‘CI’ from ‘unsigned char’ to ‘char’ inside { } [-Wnarrowing]
../src/hbprotocol/machine_protocol.c: In function ‘void protocol_send_ack(int (*)(unsigned char*, int), unsigned char)’:
../src/hbprotocol/machine_protocol.c:268:65: warning: narrowing conversion of ‘CI’ from ‘unsigned char’ to ‘char’ inside { } [-Wnarrowing]
     char tmp[] = { PROTOCOL_SOM_ACK, CI, 1, PROTOCOL_CMD_ACK, 0 };
                                                                 ^
/tmp/ccihzymg.o: In function `tickWrapper()':
HoverboardAPI.cpp:(.text+0x5): undefined reference to `millis'
/tmp/ccihzymg.o: In function `HoverboardAPI::HoverboardAPI(int (*)(unsigned char*, int))':
HoverboardAPI.cpp:(.text+0x34): undefined reference to `setup_protocol'
HoverboardAPI.cpp:(.text+0x6f): undefined reference to `delay'
/tmp/ccx2ZUJZ.o: In function `protocol_init':
protocol.c:(.text+0xb0f): undefined reference to `ascii_init'
/tmp/ccsJ09fJ.o: In function `protocol_byte':
machine_protocol.c:(.text+0x89): undefined reference to `ascii_byte'
collect2: error: ld returned 1 exit status

p-h-a-i-l, 2019-07-28:
never have, but please share your experience if you get it working! Maybe we can make it compatible.

for sure you need to update the functions protocol_GetTick and protocol_Delay.
But there might be some other dependencies covered by the arduino framework.

setup_protocol() can be found here: https://github.com/bipropellant/bipropellant-hoverboard-api/blob/38f9384b7f0eb0079ff13d76856c4772fd20fa6d/src/protocolFunctions.c#L19

ducktrA, 2019-07-28:
Somehow i miss something..... Feels like protocolFunctions.c not included. But its getting much worse if not.
 
g++ read.cpp ../src/HoverboardAPI.cpp ../src/hbprotocol/protocol.c ../src/protocolFunctions.c ../src/hbprotocol/machine_protocol.c ../src/hbprotocol/ascii_protocol.c  -I ../src -I ../src/hbprotocol -lserial -std=c++11 -Wwrite-strings -Wnarrowing -fpermissive

.......
                                                                                 
../src/hbprotocol/ascii_protocol.c:208:58: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
         ascii_add_line_fn( '?', line_help, "display help");
                                                          ^
/tmp/ccKc0nG4.o: In function `HoverboardAPI::HoverboardAPI(int (*)(unsigned char*, int))':
HoverboardAPI.cpp:(.text+0xda): undefined reference to **`setup_protocol'**
/tmp/ccSsmIWa.o: In function `protocol_byte':
machine_protocol.c:(.text+0x89): undefined reference to **`ascii_byte'**
collect2: error: ld returned 1 exit status

ducktrA, 2019-08-05:
Part success. Got rid of some more "extern ..." which prevents the compiler from searching locally. put port communication to 38400. the serial library didnt like the 26315. Now one wheel is turning when sending the pwm(pwm, steer, som) commands. I cannot read from the board and cannot the the second wheel going.

p-h-a-i-l, 2019-08-05:
Does the wheel stop turning as soon as you stop sending?
I'd use a "normal" baud rate like 115200 or something like that too...

If you can get one wheel going, sending is working. Maybe something wrong with the hoverboard?

Do you have your code somewhere to look at?

ducktrA, 2019-08-05:
the wheel stop if command is not repeated. Hoverboard checked with Ascii commands. Fully healthy. I would assume a misaligned buffer. I find a place to put the code to.

ducktrA, 2019-08-05:
#include <HoverboardAPI.h>
//#include <libserial/SerialPort.h>
//#include <libserial/SerialStream.h>

#include "serial/serial.h"

#include <iostream>
#include <unistd.h>

#include <cstring>
#include <vector>
#include <chrono>
#include <thread>

using namespace std;

void print_usage()
{
	cerr << "Usage: test_serial {<serial port address>} ";
    cerr << "<baudrate> [test string]" << endl;
}

serial::Serial *my_serial;

int serialWrapper(unsigned char *data, int len) {
    return my_serial->write(data, len);
}

int main(int argc, char **argv) {
    if(argc < 2) {
        print_usage();
        return 0;
    }

    string port(argv[1]);
    unsigned long baud = 26315;

    sscanf(argv[2], "%lu", &baud);
    std::cout << port << std::endl;

    my_serial = new serial::Serial(port, baud, serial::Timeout::simpleTimeout(1000));

    HoverboardAPI hoverboard = HoverboardAPI(serialWrapper);

    int pwm = 0;
    int direction = 1;

    for(int i = 0; i < 10000; i++) {
        
        if(pwm > 60){
            direction = -1;
        }

        if(pwm < -60) {
            direction = 1;
        }

        hoverboard.sendPWM(pwm, 0, PROTOCOL_SOM_NOACK);
        

        pwm += direction;
        std::cout << pwm << " " << hoverboard.getBatteryVoltage() << std::endl;

        std::this_thread::sleep_for(std::chrono::milliseconds(50));
    }

ducktrA, 2019-08-06:
I was wondering how the protocol reads from serial. i got that from one of the samples. For the single wheel turn there is no other way than inspecting the fields send across  i guess.

p-h-a-i-l, 2019-08-06:
You need to hand incoming data to protocolPush, one byte at a time. See https://github.com/bipropellant/bipropellant-hoverboard-api/blob/38f9384b7f0eb0079ff13d76856c4772fd20fa6d/examples/readSpeed/readSpeed.ino#L50-L52

You can add some debug output to dump the outgoing message somewhere or connect yourself to the serial interface to see what your program is sending. I can help to decode the captured data.

What values are you sending? Try to keep one value (speed or steer) at zero, so they don't cancel each other out.

Both wheels running forward at 10% duty cycle: `hoverboard.sendPWM(100, 0, PROTOCOL_SOM_NOACK);` 
One wheel forward, one backward at 10% duty cycle: `hoverboard.sendPWM(0, 100, PROTOCOL_SOM_NOACK);` 
One wheel forward at 20% duty cycle, the other wheel stopped: `hoverboard.sendPWM(100, 100, PROTOCOL_SOM_NOACK);`

ducktrA, 2019-08-06:
Sending repeatingly  hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK); as below leads to the following log further down below.

serial::Serial *my_serial;

int serialWrapper(unsigned char *data, int len) {
    
    printf("to board: ");
    for(int i = 0; i < len; i++) {
        printf("%x", data[i]);
    }
    printf("\n");

    return my_serial->write(data, len);
}

int main(int argc, char **argv) {
    if(argc < 2) {
        print_usage();
        return 0;
    }

    string port(argv[1]);
    unsigned long baud = 26315;

    sscanf(argv[2], "%lu", &baud);
    std::cout << port << std::endl;

    my_serial = new serial::Serial(port, baud, serial::Timeout::simpleTimeout(1000));

    HoverboardAPI hoverboard = HoverboardAPI(serialWrapper);

    int pwm = 0;
    int direction = 1;

    for(int i = 0; i < 10000; i++) {
        
        if(pwm > 60){
            direction = -1;
        }

        if(pwm < -60) {
            direction = 1;
        }

        hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK);
        hoverboard.requestRead(HoverboardAPI::Codes::sensHall, PROTOCOL_SOM_NOACK);

        //std::cout << pwm << " " << hoverboard.getBatteryVoltage() << std::endl;

        std::this_thread::sleep_for(std::chrono::milliseconds(30));
        
        int j = 0;
        string r = "";

        printf("      from board: ");

        while(my_serial->available() && j++ < 1024) { // read maximum 1024 bytes at once.
            string rc = my_serial->read();
            const char *cstr = rc.c_str();
            if (rc.length() == 1) {
                printf("%x", cstr[0]);
                hoverboard.protocolPush(cstr[0]);
            }
            
            r.append(rc);
        }

        printf("\n\n");

        hoverboard.protocolTick();

        pwm += direction;
        std::this_thread::sleep_for(std::chrono::milliseconds(30));
    }
}

log in hex:
/dev/ttyUSB0
to board: 411257e32000000032000000024
to board: 422522a8
      from board: 4ffffffb1377e1ffffffc64ffffffb24a722ffffff8a0000000216bffffffb84000001b300000000000000000000000000216bffffffb840000000000000000000000000ffffffe0

to board: 431257e32000000032000000022
to board: 442522a6
      from board: 4ffffffb3377e1ffffffc44ffffffb44a722ffffff8b0000000216bffffffb8400000213000000000031ffffff98471000000000000216bffffffb840000000000000000000000000ffffffc6

to board: 451257e32000000032000000020
to board: 462522a4
      from board: 4ffffffb5377e1ffffffc24ffffffb64a722ffffff8d00020000216bffffffb84000002c30000004a0005f1e00000000000000216bffffffb840000000000000000000000000ffffffe1

to board: 471257e3200000003200000001e
to board: 482522a2
      from board: 4ffffffb7377e1ffffffc04ffffffb84a722ffffff8e0001f000216bffffffb8400000323000000470004d1f00000000000000216bffffffb840000000000000000000000000ffffffed

to board: 491257e3200000003200000001c
to board: 4a2522a0
      from board: 4ffffffb9377e1ffffffbe4ffffffba4a722ffffff900001f000216bffffffb84000003d300000047000501f00000000000000216bffffffb840000000000000000000000000ffffffdb

to board: 4b1257e3200000003200000001a
to board: 4c25229e
      from board: 4ffffffbb377e1ffffffbc4ffffffbc4a722ffffff9100022000216bffffffb84000004330000004e000ffffffd91c00000000000000216bffffffb84000000000000000000000000042

to board: 4d1257e32000000032000000018
to board: 4e25229c
      from board: 4ffffffbd377e1ffffffba4ffffffbe4a722ffffff9300020000216bffffffb84000004f300000048000ffffffdd1e00000000000000216bffffffb84000000000000000000000000034

to board: 4f1257e32000000032000000016
to board: 41025229a
      from board: 4ffffffbf377e1ffffffb84ffffffc04a722ffffff9400020000216bffffffb84000005430000004a000631e00000000000000216bffffffb840000000000000000000000000ffffffa4

to board: 4111257e32000000032000000014
to board: 412252298
      from board: 4ffffffc1377e1ffffffb64ffffffc24a722ffffff9600020000216bffffffb840000060300000049000ffffffc61e00000000000000216bffffffb84000000000000000000000000032

[log.txt](https://github.com/bipropellant/bipropellant-hoverboard-firmware/files/3473584/log.txt)
[Uploading log.txt…]()

ducktrA, 2019-08-06:
Sending repeatingly  hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK); as below leads to the following log further down below.

serial::Serial *my_serial;

int serialWrapper(unsigned char *data, int len) {
    
    printf("to board: ");
    for(int i = 0; i < len; i++) {
        printf("%x", data[i]);
    }
    printf("\n");

    return my_serial->write(data, len);
}

int main(int argc, char **argv) {
    if(argc < 2) {
        print_usage();
        return 0;
    }

    string port(argv[1]);
    unsigned long baud = 26315;

    sscanf(argv[2], "%lu", &baud);
    std::cout << port << std::endl;

    my_serial = new serial::Serial(port, baud, serial::Timeout::simpleTimeout(1000));

    HoverboardAPI hoverboard = HoverboardAPI(serialWrapper);

    int pwm = 0;
    int direction = 1;

    for(int i = 0; i < 10000; i++) {
        
        if(pwm > 60){
            direction = -1;
        }

        if(pwm < -60) {
            direction = 1;
        }

        hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK);
        hoverboard.requestRead(HoverboardAPI::Codes::sensHall, PROTOCOL_SOM_NOACK);

        //std::cout << pwm << " " << hoverboard.getBatteryVoltage() << std::endl;

        std::this_thread::sleep_for(std::chrono::milliseconds(30));
        
        int j = 0;
        string r = "";

        printf("      from board: ");

        while(my_serial->available() && j++ < 1024) { // read maximum 1024 bytes at once.
            string rc = my_serial->read();
            const char *cstr = rc.c_str();
            if (rc.length() == 1) {
                printf("%x", cstr[0]);
                hoverboard.protocolPush(cstr[0]);
            }
            
            r.append(rc);
        }

        printf("\n\n");

        hoverboard.protocolTick();

        pwm += direction;
        std::this_thread::sleep_for(std::chrono::milliseconds(30));
    }
}

log in hex:
/dev/ttyUSB0
to board: 411257e32000000032000000024
to board: 422522a8
      from board: 4ffffffb1377e1ffffffc64ffffffb24a722ffffff8a0000000216bffffffb84000001b300000000000000000000000000216bffffffb840000000000000000000000000ffffffe0

to board: 431257e32000000032000000022
to board: 442522a6
      from board: 4ffffffb3377e1ffffffc44ffffffb44a722ffffff8b0000000216bffffffb8400000213000000000031ffffff98471000000000000216bffffffb840000000000000000000000000ffffffc6

to board: 451257e32000000032000000020
to board: 462522a4
      from board: 4ffffffb5377e1ffffffc24ffffffb64a722ffffff8d00020000216bffffffb84000002c30000004a0005f1e00000000000000216bffffffb840000000000000000000000000ffffffe1

to board: 471257e3200000003200000001e
to board: 482522a2
      from board: 4ffffffb7377e1ffffffc04ffffffb84a722ffffff8e0001f000216bffffffb8400000323000000470004d1f00000000000000216bffffffb840000000000000000000000000ffffffed

to board: 491257e3200000003200000001c
to board: 4a2522a0
      from board: 4ffffffb9377e1ffffffbe4ffffffba4a722ffffff900001f000216bffffffb84000003d300000047000501f00000000000000216bffffffb840000000000000000000000000ffffffdb

to board: 4b1257e3200000003200000001a
to board: 4c25229e
      from board: 4ffffffbb377e1ffffffbc4ffffffbc4a722ffffff9100022000216bffffffb84000004330000004e000ffffffd91c00000000000000216bffffffb84000000000000000000000000042

to board: 4d1257e32000000032000000018
to board: 4e25229c
      from board: 4ffffffbd377e1ffffffba4ffffffbe4a722ffffff9300020000216bffffffb84000004f300000048000ffffffdd1e00000000000000216bffffffb84000000000000000000000000034

to board: 4f1257e32000000032000000016
to board: 41025229a
      from board: 4ffffffbf377e1ffffffb84ffffffc04a722ffffff9400020000216bffffffb84000005430000004a000631e00000000000000216bffffffb840000000000000000000000000ffffffa4

to board: 4111257e32000000032000000014
to board: 412252298
      from board: 4ffffffc1377e1ffffffb64ffffffc24a722ffffff9600020000216bffffffb840000060300000049000ffffffc61e00000000000000216bffffffb84000000000000000000000000032

[log.txt](https://github.com/bipropellant/bipropellant-hoverboard-firmware/files/3473584/log.txt)

ducktrA, 2019-08-06:
Sending repeatingly  hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK); as below leads to the following log further down below.

serial::Serial *my_serial;

int serialWrapper(unsigned char *data, int len) {
    
    printf("to board: ");
    for(int i = 0; i < len; i++) {
        printf("%x", data[i]);
    }
    printf("\n");

    return my_serial->write(data, len);
}

int main(int argc, char **argv) {
    if(argc < 2) {
        print_usage();
        return 0;
    }

    string port(argv[1]);
    unsigned long baud = 26315;

    sscanf(argv[2], "%lu", &baud);
    std::cout << port << std::endl;

    my_serial = new serial::Serial(port, baud, serial::Timeout::simpleTimeout(1000));

    HoverboardAPI hoverboard = HoverboardAPI(serialWrapper);

    int pwm = 0;
    int direction = 1;

    for(int i = 0; i < 10000; i++) {
        
        if(pwm > 60){
            direction = -1;
        }

        if(pwm < -60) {
            direction = 1;
        }

        hoverboard.sendPWM(50, 0, PROTOCOL_SOM_NOACK);
        hoverboard.requestRead(HoverboardAPI::Codes::sensHall, PROTOCOL_SOM_NOACK);

        //std::cout << pwm << " " << hoverboard.getBatteryVoltage() << std::endl;

        std::this_thread::sleep_for(std::chrono::milliseconds(30));
        
        int j = 0;
        string r = "";

        printf("      from board: ");

        while(my_serial->available() && j++ < 1024) { // read maximum 1024 bytes at once.
            string rc = my_serial->read();
            const char *cstr = rc.c_str();
            if (rc.length() == 1) {
                printf("%x", cstr[0]);
                hoverboard.protocolPush(cstr[0]);
            }
            
            r.append(rc);
        }

        printf("\n\n");

        hoverboard.protocolTick();

        pwm += direction;
        std::this_thread::sleep_for(std::chrono::milliseconds(30));
    }
}

log in hex:
/dev/ttyUSB0
to board: 411257e32000000032000000024
to board: 422522a8
      from board: 4ffffffb1377e1ffffffc64ffffffb24a722ffffff8a0000000216bffffffb84000001b300000000000000000000000000216bffffffb840000000000000000000000000ffffffe0

to board: 431257e32000000032000000022
to board: 442522a6
      from board: 4ffffffb3377e1ffffffc44ffffffb44a722ffffff8b0000000216bffffffb8400000213000000000031ffffff98471000000000000216bffffffb840000000000000000000000000ffffffc6

to board: 451257e32000000032000000020
to board: 462522a4
      from board: 4ffffffb5377e1ffffffc24ffffffb64a722ffffff8d00020000216bffffffb84000002c30000004a0005f1e00000000000000216bffffffb840000000000000000000000000ffffffe1

to board: 471257e3200000003200000001e
to board: 482522a2
      from board: 4ffffffb7377e1ffffffc04ffffffb84a722ffffff8e0001f000216bffffffb8400000323000000470004d1f00000000000000216bffffffb840000000000000000000000000ffffffed

to board: 491257e3200000003200000001c
to board: 4a2522a0
      from board: 4ffffffb9377e1ffffffbe4ffffffba4a722ffffff900001f000216bffffffb84000003d300000047000501f00000000000000216bffffffb840000000000000000000000000ffffffdb

to board: 4b1257e3200000003200000001a
to board: 4c25229e
      from board: 4ffffffbb377e1ffffffbc4ffffffbc4a722ffffff9100022000216bffffffb84000004330000004e000ffffffd91c00000000000000216bffffffb84000000000000000000000000042

to board: 4d1257e32000000032000000018
to board: 4e25229c
      from board: 4ffffffbd377e1ffffffba4ffffffbe4a722ffffff9300020000216bffffffb84000004f300000048000ffffffdd1e00000000000000216bffffffb84000000000000000000000000034

to board: 4f1257e32000000032000000016
to board: 41025229a
      from board: 4ffffffbf377e1ffffffb84ffffffc04a722ffffff9400020000216bffffffb84000005430000004a000631e00000000000000216bffffffb840000000000000000000000000ffffffa4

to board: 4111257e32000000032000000014
to board: 412252298
      from board: 4ffffffc1377e1ffffffb64ffffffc24a722ffffff9600020000216bffffffb840000060300000049000ffffffc61e00000000000000216bffffffb84000000000000000000000000032

ducktrA, 2019-08-06:
let me know if this is helpful. thanks for your support.

p-h-a-i-l, 2019-08-06:
can you format the data with 
`printf("%02X ", data[i]);` ? Otherwise it is hard to decipher.

Also would be great if you'd upload you code to a repository, easier to read (and for you to track changes :) )

ducktrA, 2019-08-06:
Here we go. Note that i am sending pwm = 60, steer= 10 now and i have one wheel turning. same with high pwms and steer = 0

to board: 04 0112570E 46000000 00000000 32000000 00000000 10
to board: 04 02025202 A8
      from board: 04010377 0E017604 024A7202 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 38

to board: 04 0312570E 46000000 00000000 32000000 00000000 0E
to board: 04 04025202 A6
      from board: 04030377 0E017404 044A7202 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 36

to board: 04 0512570E 46000000 00000000 32000000 00000000 0C
to board: 04 06025202 A4
      from board: 04050377 0E017204 064A7202 03000000 3B000000 216BFFFFFFB840 00000000 11000000 00000000 FFFFFF85000000 FFFFFFD8100000 00000000 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 78

to board: 04 0712570E 46000000 00000000 32000000 00000000 0A
to board: 04 08025202 A2
      from board: 04070377 0E017004 084A7202 06000000 39000000 216BFFFFFFB840 00000000 22000000 00000000 FFFFFF81000000 54110000 00000000 00000000 00000000 216BFFFFFFB840 00000000 00000000 00000000 00000000 00000000 00000000 FFFFFFEB

to board: 04 0912570E 46000000 00000000 32000000 00000000 08

[log.txt](https://github.com/bipropellant/bipropellant-hoverboard-firmware/files/3474059/log.txt)

p-h-a-i-l, 2019-08-07:
Ok, found the problem:

to board: 04 0112570E 46000000 00000000 32000000 00000000 10

decodes to
SOM:NoACK | CI:01 | len:12 | CMD:Write | setPointPWM | 46 | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 32 | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 10

So we are talking about this command: https://github.com/bipropellant/bipropellant-protocol/blob/fe96935cb2f550110ffaaa6d74852bedd79e25c7/protocol.c#L291

with this data: https://github.com/bipropellant/bipropellant-protocol/blob/fe96935cb2f550110ffaaa6d74852bedd79e25c7/protocol.h#L64

Problem is: datatype is long. On 32 bit systems usually 4 bytes, on 64 bit systems 8.
So a change in protocol.h from long to int32_t should solve your problem!

(And what a coincidence, there is also a PR from yesterday regarding the same topic: https://github.com/bipropellant/bipropellant-protocol/pull/31)

btsimonh, 2019-08-07:
@p-h-a-i-l - good spot.
I've not build enough on 64 bit in c/c++ to realise this!.
We should do a brief double check that any 'long', 'int', 'short' and 'char' are made more specific.  I don't think short or char ever change, but best to be sure.

p-h-a-i-l, 2019-08-07:
Didn't find any good sources, but I'd say we're good if we replace long with int or int32_t. 
I'd even replace all int just to avoid confusion.

ducktrA, 2019-08-07:
spot on diagnose. my first query replace all hammer was a bit too exhaustive as the UI_LONG constants were affected too. 

pwm and steer commands reflect in proper wheel turns. Jolly good and big thank you. I close this issue

I compile it using below. read.cpp is the humble program. There were a couple of redundant "extern" in the code which prevented from compiling. Next is shut up the buzzer, absolute hall-ticks, electric brake around pwm close to zero.

g++ read.cpp ../src/HoverboardAPI.cpp ../src/hbprotocol/protocol.c ../src/protocolFunctions.c ../src/hbprotocol/machine_protocol.c ../src/hbprotocol/ascii_protocol.c -I ../src -I ../src/hbprotocol -lserial -std=c++11 -Wwrite-strings -Wnarrowing -fpermissive

closed 2019-08-07 by ducktrA

p-h-a-i-l, 2019-08-07:
could you provide me with a copy of your complete source?

ducktrA, 2019-08-07:
sure. was looking for your email already. best from stuttgart

ducktrA, 2019-08-07:
give me a hint how to share it. github doesnt like tarballs.

p-h-a-i-l, 2019-08-08:
You could add it as a gist. Or even better, create a repository and use git to manage the code