bipropellant/bipropellant-hoverboard-firmware #69: Configuration for GD32F103 based boards

fender90, 2019-07-30:
Hi guys. I've got a board that is based on GD chip
Has anyone managed to run the firmware on these boards?
Could you advice on how to configure it properly? 
I've got several theories but it would be great is someone could confirm.
As mentioned in config file the baud rate for GD chips is 52177
Below are a couple of configuration schemes

`// thoery says this is the only thing you need to change....
#define CONTROL_TYPE SOFTWARE_SERIAL_A2_A3
//////////////////////////////////////////////////////////`

So i need to change the baud rate to 52177 in

`// ############################### SOFTWARE SERIAL ###############################
//
// there should now be a free choice of serial GPIO pins
#define SOFTWARE_SERIAL_BAUD 52177`

The second option is to select 
`// thoery says this is the only thing you need to change....
#define USART3_CONTROLLED
//////////////////////////////////////////////////////////`

and to change the baud rate in 
`// #define SERIAL_USART3_IT
  #define USART3_BAUD       52177                  // UART baud rate`

Do I get it right?
Sorry if my questions are a bit dumb :-)

btsimonh, 2019-07-30:
if using software serial, leave the software serial baud rate at 9600.
the 52177 is for the sensor boards, but it should auto sense this if using the 'hoverboard' env in platform.io.

p-h-a-i-l, 2019-07-30:
> Hi guys. I've got a board that is based on GD chip
> Has anyone managed to run the firmware on these boards?

Yes, I got both ST and GD boards in use.

> Could you advice on how to configure it properly?

How do you want to control the baord? I guess you want to use the sensor boards (hoverboard mode) since you are talking about the baud rate below.)

> I've got several theories but it would be great is someone could confirm.
> As mentioned in config file the baud rate for GD chips is 52177

The baud rate applies to the chips on the sensor boards.

> `// thoery says this is the only thing you need to change.... #define CONTROL_TYPE SOFTWARE_SERIAL_A2_A3 //////////////////////////////////////////////////////////`
>
> So i need to change the baud [...]

Software serial can be used as an additional serial connection to communicate to a PC or microcontroller when bot UART cables are in use. So the baud rate there is not the baud rate needed to communicate to the sensor boards

> The second option is to select
> `// thoery says this is the only thing you need to change.... #define USART3_CONTROLLED //////////////////////////////////////////////////////////`
> 
> and to change the baud rate in
> `// #define SERIAL_USART3_IT #define USART3_BAUD 52177 // UART baud rate`

yes, when connecting to the sensor boards

fender90, 2019-07-30:
> Yes, I got both ST and GD boards in use.
>
>
> How do you want to control the baord? I guess you want to use the sensor boards (hoverboard mode) since you are talking about the baud rate below.)
>
>
> The [...]

Thanks for your reply
Oh, I missed that the baud rate was 52177 for sensor boards only
I want to control the board via software serial with Arduino and move to Blue Pill in future.
Is the configuration for GD boards the same as for STM?

p-h-a-i-l, 2019-07-30:
yes, the configuration is the same. 
If you are using arduino or bluepill I'd recommend using UART2 or UART3, baud rate on software serial is limited

fender90, 2019-07-30:
> yes, the configuration is the same.
> If you are using arduino or bluepill I'd recommend using UART2 or UART3, baud rate on software serial is limited

So just leaving the baud rate at 9600 and setting the #define CONTROL_TYPE USART3_CONTROLLED

Thanks will try and write the results :-)

fender90, 2019-07-30:
Sorry for another dumb question. What baud rate should I use in Arduino sketch?
115200?
there is a baud rate set in config.h
`#ifndef USART3_BAUD
  #define USART3_BAUD 26315`

this baud rate is not supported by arduino, I can use it with BluePill
What would be the best for Arduino?

p-h-a-i-l, 2019-07-30:
115200 is good, the default baud rate is for communication to the sensor boards.

fender90, 2019-07-30:
> 115200 is good, the default baud rate is for communication to the sensor boards.

So I should change the baud rate in
`#ifndef USART3_BAUD
#define USART3_BAUD 115200` ?

Me and my friend tried using configuration with
#define CONTROL_TYPE USART3_CONTROLLED and baud rate 9600 and 115200 but didn't get any result

we changed the control type to CONTROL_TYPE SOFTWARE_SERIAL_A2_A3 and were able to control the wheels on some GD based boards.
Actually we are getting some very strange results with this option
We have boards with GD32F103 chips. We are using the same firmware builds on both of them and the same Arduino sketch from HoverboardAPI examples
https://github.com/bipropellant/bipropellant-hoverboard-api/blob/master/examples/readSpeed/readSpeed.ino

The wheel on the first board runs OK and we get data in Terminal
https://www.youtube.com/watch?v=HgDN6uisttc

But the second wheels runs with interruptions
https://www.youtube.com/watch?feature=youtu.be&v=L-sIF51fuGM

we even tried to reboot the board as shown in the video but the motor still spins with interruptions

p-h-a-i-l, 2019-07-30:
> So I should change the baud rate in
> `#ifndef USART3_BAUD #define USART3_BAUD 115200` ?

yes

> Me and my friend tried using configuration with
> #define CONTROL_TYPE USART3_CONTROLLED and baud rate 9600 and 115200 but didn't get any result

I just noticed that other defaults do not work with USART3_CONTROLLED. (I'm always using the environments from platformio.ini so I haven't noticed problems with config.h.

you also need to set

```
    #define SERIAL_USART3_IT
    #define USART3_BAUD 115200 
    #define USART3_WORDLENGTH UART_WORDLENGTH_8B   
    #define  SERIAL_USART_IT_BUFFERTYPE unsigned char
```

> we changed the control type to CONTROL_TYPE SOFTWARE_SERIAL_A2_A3 and were able to control the wheels on some GD based boards.
> Actually we are getting some very strange results with this option
> We [...]

Might be EMI related. Short wires, electrical filters, shielding..

p-h-a-i-l, 2019-07-30:
https://github.com/bipropellant/bipropellant-hoverboard-firmware/pull/70 should also fix the config problems..

closed 2019-08-12 by fender90