bipropellant/bipropellant-hoverboard-firmware #72: Acceleration determination from HB angle

desert0014, 2019-08-05:
Hey, I hope I haven't overlooked this issue in the open once if it is already known...

I use the board only in hoverboard mode (that's why I chose this repo ;-)), flashing and starting works fine for me. But when standing on the board and leaning forward, the hoverboard accelerates much slower than before (with the original fw), causing the driver to almost fall off forward (if used to the "normal" hoverboard speed). I tested the speed with wheels in the air and noticed, that holding one angle does not accelerate as it does with the original fw, but keeping the same speed. Top speed is okay, but it's not reachable in hoverboard mode when standing on it. With the original fw, the angle determines the acceleration, with this fw, it determines the speed. Is there a way to change this?

Thank you in advance and thank you also for your great work with this repo so far!

EmanuelFeru, 2019-08-06:
Yes, this is a valid remark. I also mentioned it to @btsimonh 
To reach higher speeds the control strategy needs to be updated to accelleration control. This will allow also driving costantly with flat/leveled hover board. At the moment is not possible since the angle needs to be tilted all the time in order to move.
There are 2 ways to fix this:
* integrate the angle reading with a tunable gain (this is not more than an I controller) and feed that to the speed motor controller
* or the low level motor controller to be update to TORQUE mode instead of speed mode. This is only possible using FOC motor control. I am working in making this mode availble. For now, my suggestion is to use the integration approach.

Chaos99, 2019-12-14:
I am just about to flash my hardware with this firmware fork, but this issue stopped me for a moment. My conversion currently runs with normal Hoverboard firmware. I don't want to jeopardize this with the firmware that doesn't behave the same.

That said, currently, with original Firmware,  I also have constant speed with constant angle. How else would self-balancing work?   Returning to a leveled board needs the bring you back to a standstill. Does you board need to be tilted backwards to decelarate? Does this mean if you keep it tilted backwards until you reach a standstill, you fall off backwards?
Is this a special mode you activated via the phone app?

bionic24, 2020-02-08:
@Chaos99 Did you ever resolve this issue. From reading your posts on Github I am guessing that you have build a Onewheel from your Hoverboard? 

I naively flashed Bipropellant firmware to my board thinking it had improved Hoverboard functionality. Mostly to adjust the voltage settings. But right now my Onewheel is unusable. Response is much to slow and if you try to speed up you kick the nose on the ground. The FOC firmware from @EmanuelFeru sounds interesting but doesn't support the original sensorboard (6word).

Is there a way to increase I term dramatically. Or perhaps limit the max angle of the gyro sensorboard so you get 100% throttle at 25% angle in Bipropellant or am likely better of buying another mainboard?

Chaos99, 2020-02-10:
@bionic24 Hi. I'm currently trying to adapt the bipropellant firmware to my hardware. I have a so far unsupported hardware version. I did get the motor-test mode of the original Niklas Fauth firmware to run, but this one does not support the sensors boards as inputs.
The bipropellant firmware, even after adapting to my hardware, did not work out of the box. I suspected that it was more strict than the official hoverboard firmwar in that it did not work after disconnecting one of the sensor boards. So I modified it now to run a debug UART conection on the left interface (so I get output on my PC) and receive sensor readings from the senor board on the right interface. 
I know the communcation works, as I get debug output about the foot switches being pressed or not, but the motors do not turn yet.
I still have to figure out some kinks in my debug connection until I can verify the actual angle readings. From then I will continue to debug why the motors won't turn.

To your original question: modifying the max lean angle was also my motivation for flashing a custom firmware. I'm sure modifing the formula will be the easy part once I'll get it to work at all.

I'll keep you posted here once I'm done and I will upload a pull request for the new mode then.

bionic24, 2020-02-10:
@Chaos99 , awesome I'll keep an eye out for it. Right now I will resort to using a Naze32(rev5) in combination with Emanuel his FOC version of firmware.
https://github.com/blezalex/WheelBoardMain
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

The idea is to have the PPM output of the Naze32 flightcontroller connected to the PA03 pin of the hoverboard and run the FOC firmware with the remote control version.

Naze32(rev5) is currently selling for 5$ on Hobbyking as the hardware is outdated for use on drones.

EmanuelFeru, 2020-02-10:
@bionic24 @Chaos99 I just released 1st hacked firmware version for the sideboards:
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM

It is not yet linked to my FOC branch.. still to do... Feel free to contribute if you want.

bionic24, 2020-02-10:
@EmanuelFeru Wow, thanks I wasn't expecting code this soon. For my Noob understandings:
1. flash sideboard firmware
2. flash FOC firmware mainboard and use the USART variant.
3. connect sideboard with original cables

And that is it? Photo interuptor is active? Sounds almost too easy :).

Testing and further communication on the FOC fork?

EmanuelFeru, 2020-02-10:
Indeed, USART variant should work. We still need to match thee Tx Rx structures. Basically to add a struct in main board  called Sideboard. 
In the future I prefer creating a VARIANT_HOVERBOARD in main board too and build on that.

All the sideboard functionality can be used, it might require a bit of work, because I just created the framework to work with.