bipropellant/bipropellant-hoverboard-firmware #73: Read absolute hall ticks

ducktrA, 2019-08-07:
I need the absolute Hallticks. Thus added an enum for it in Codes ({ 0x07, "hall position steps",     NULL,  UI_NONE,  &contentbuf, sizeof(PROTOCOL_POSN),  ...      )
in alignment of corresponding function and code in protocol.c/h. 

Via API i can now do this.

hoverboard.requestRead(HoverboardAPI::Codes::sensHallPosSteps, PROTOCOL_SOM_NOACK);

Question: Where do i find the result once the request is completed?

Thank you.

btsimonh, 2019-08-08:
hi ducktrA,
I've not used the protocol implementation externally from C as you have, but I think it should populate the variable you mention - i.e. contentbuf.
You should also be able to add a function in your defn line to be called with the data....
s
@p-h-a-i-l will be able to help more :).

p-h-a-i-l, 2019-08-08:
There are 2 possibilities: You can register a storage location to this code, so each time data is received the variable gets updated: setParamVariable
Or you can register a callback which gets called each time data is received: setParamHandler
In this callback you can process or copy the data which is temporary stored in contentbuf.

Of course you can do btoh ;)

Have a look at https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/master/src/protocolfunctions.c to get an idea what we are doing.