bipropellant/bipropellant-hoverboard-firmware #75: Motor not starting (USART2 Communication OK)

pasoftdev, 2019-08-29:
Hello,

I'm using the USART2 control type (Using a Raspberry pi 3 with a usb-to-serial),
so far I can communicate with the board but the motors don't spin, they just make an high frequency noise. I see in the troubleshooting that it is suggested to check the phase mapping,
can someone help me on this?
The board is a STM32, the hoverboard is a new one and was working,
So far I just soldered the SWD header, unlocked the STM and flashed it using st-flash.
I can provide photos and logs, I can use an oscilloscope and a multimeter if it can help
see the attached photo for the board and the brief log:

---- Serial log ---
cf_speedCoef 10690, n_commDeacvHi 60, n_commAcvLo 30
?- ?
Locked. Enter unlockASCII to enable ASCII input mode.
>Locked. Enter unlockASCII to enable ASCII input mode.
>unlockASCII
- unlockASCII
ASCII input active. Type ? for help
>?
- ?
? - display help
A - set alarm
B - toggle sensor control
C - show electrical measurements
E - dEbug control, E->off, Ec->console on, Es->console+scope
F - print/set a flash constant (Fa to print all, Fi to default all):
  Fss - print, Fss<n> - set
  Fm<n> - flash magic
  Fpkp<n> - posn kp x 100
  Fpki<n> - posn ki x 100
  Fpkd<n> - posn kd x 100
  Fpl<n> - posn pwm lim
  Fskp<n> - speed kp x 100
  Fski<n> - speed ki x 100
  Fskd<n> - speed kd x 100
  Fsl<n> - speed pwm incr lim
  Fcl<n> - max current limit x 100
  Fhe<n> - hoverboard enable

G - display stm32 specific
I - enable immediate commands - Ip/Is/Iw - direct to posn/speed/pwm control

L - Lock ascii protocol
M -  M - dump memory
   Mf - dump flash
   M<hexaddr>,<hexlen> - dump mem

N - display sensor data
P -  P -power control
  P -disablepoweroff
  PE enable poweroff
  Pn power off in n seconds
  Pr software reset

R -  - R! -> Reset Firmware
S - show main loop timing stats
T - tt - send a test protocol message 
u - unlockASCII - unlock ASCII protocol
>C
- C
Bat: 39484mV(0) Temp:181C(1684)
L: Current:200mA Avg:62mA r1:263 r2:301
R: Current:79mA Avg:47mA r1:287 r2:290
>P
- P
disablepoweroff now 1
powerofftimer now 0
>Iw
- Iw
Immediate commands enabled - WASDXHCGQ - Power (pWm) control
>>speed now 10, steer now 0, pwm 5, pwm 5
>speed now 20, steer now 0, pwm 10, pwm 10
>speed now 30, steer now 0, pwm 15, pwm 15
>speed now 40, steer now 0, pwm 20, pwm 20
>speed now 50, steer now 0, pwm 25, pwm 25
>speed now 60, steer now 0, pwm 30, pwm 30
>speed now 70, steer now 0, pwm 35, pwm 35
>speed now 80, steer now 0, pwm 40, pwm 40
>speed now 70, steer now 0, pwm 35, pwm 35
>speed now 60, steer now 0, pwm 30, pwm 30
>speed now 50, steer now 0, pwm 25, pwm 25
>speed now 40, steer now 0, pwm 20, pwm 20
>speed now 30, steer now 0, pwm 15, pwm 15
>speed now 20, steer now 0, pwm 10, pwm 10
>speed now 10, steer now 0, pwm 5, pwm 5
>speed now 0, steer now 0, pwm 0, pwm 0
>speed now 0, steer now -10, pwm 0->-5, pwm 0->5
>speed now 0, steer now -20, pwm 0->-10, pwm 0->10
>speed now 0, steer now -30, pwm 0->-15, pwm 0->15
>speed now 0, steer now -20, pwm 0->-10, pwm 0->10
>speed now 0, steer now -10, pwm 0->-5, pwm 0->5
>speed now 0, steer now 0, pwm 0->0, pwm 0->0
>Stop set
>?
- ?
? - display help
A - set alarm
B - toggle sensor control
C - show electrical measurements
E - dEbug control, E->off, Ec->console on, Es->console+scope
F - print/set a flash constant (Fa to print all, Fi to default all):
  Fss - print, Fss<n> - set
  Fm<n> - flash magic
  Fpkp<n> - posn kp x 100
  Fpki<n> - posn ki x 100
  Fpkd<n> - posn kd x 100
  Fpl<n> - posn pwm lim
  Fskp<n> - speed kp x 100
  Fski<n> - speed ki x 100
  Fskd<n> - speed kd x 100
  Fsl<n> - speed pwm incr lim
  Fcl<n> - max current limit x 100
  Fhe<n> - hoverboard enable

G - display stm32 specific
I - enable immediate commands - Ip/Is/Iw - direct to posn/speed/pwm control

L - Lock ascii protocol
M -  M - dump memory
   Mf - dump flash
   M<hexaddr>,<hexlen> - dump mem

N - display sensor data
P -  P -power control
  P -disablepoweroff
  PE enable poweroff
  Pn power off in n seconds
  Pr software reset

R -  - R! -> Reset Firmware
S - show main loop timing stats
T - tt - send a test protocol message 
u - unlockASCII - unlock ASCII protocol
>speed now -10, steer now 0, pwm -5, pwm -5
>>Stop set
>Ip
- Ip
Immediate commands enabled - WASDXHCGQ - Position control
>>wanted_posn now 100mm 100mm
>wanted_posn now 200mm 200mm
>wanted_posn now 300mm 300mm
>wanted_posn now 200mm 200mm
>wanted_posn now 100mm 100mm
>wanted_posn now 0mm 200mm
>wanted_posn now 100mm 100mm
>wanted_posn now 0mm 0mm
>Stop set
>>?
- ?
? - display help
A - set alarm
B - toggle sensor control
C - show electrical measurements
E - dEbug control, E->off, Ec->console on, Es->console+scope
F - print/set a flash constant (Fa to print all, Fi to default all):
  Fss - print, Fss<n> - set
  Fm<n> - flash magic
  Fpkp<n> - posn kp x 100
  Fpki<n> - posn ki x 100
  Fpkd<n> - posn kd x 100
  Fpl<n> - posn pwm lim
  Fskp<n> - speed kp x 100
  Fski<n> - speed ki x 100
  Fskd<n> - speed kd x 100
  Fsl<n> - speed pwm incr lim
  Fcl<n> - max current limit x 100
  Fhe<n> - hoverboard enable

G - display stm32 specific
I - enable immediate commands - Ip/Is/Iw - direct to posn/speed/pwm control

L - Lock ascii protocol
M -  M - dump memory
   Mf - dump flash
   M<hexaddr>,<hexlen> - dump mem

N - display sensor data
P -  P -power control
  P -disablepoweroff
  PE enable poweroff
  Pn power off in n seconds
  Pr software reset

R -  - R! -> Reset Firmware
S - show main loop timing stats
T - tt - send a test protocol message 
u - unlockASCII - unlock ASCII protocol
>Is
- Is
Immediate commands enabled - WASDXHCGQ - Speed control
>>speed now 10, steer now 0, speedL 5, speedR 5
>speed now 20, steer now 0, speedL 10, speedR 10
>speed now 30, steer now 0, speedL 15, speedR 15
>speed now 20, steer now 0, speedL 10, speedR 10
>speed now 10, steer now 0, speedL 5, speedR 5
>speed now 0, steer now 0, speedL 0, speedR 0
>speed now 0, steer now -10, speedL -5, speedR 5
>speed now 0, steer now 0, speedL 0, speedR 0
>speed now 0, steer now 10, speedL 5, speedR -5
>Stop set
>>power off by low voltage
----

[image]
[image]

btsimonh, 2019-08-30:
May sound silly, but did you try pwm with >40? - because of dead zone in pwm, it needs >~40 before the motors start to turn!  (if they are not turning by 100, then there is a problem).
But, I see you tried posn - this should have worked, unless the PID values are not yet configured/really screwy.

pasoftdev, 2019-08-30:
Hi, thx for your reply,
I just tried with PWM up to 100 but the motors don't start
what should I check next?

> - Iw
> Immediate commands enabled - WASDXHCGQ - Power (pWm) control
> -
> No cmd se ? for help
> >speed now 10, steer now 0, pwm 5, pwm 5
> >speed now 20, steer now 0, pwm 10, pwm 10
> >speed now 30, steer now 0, pwm 15, pwm 15
> >speed now 40, steer now 0, pwm 20, pwm 20
> >speed now 50, steer now 0, pwm 25, pwm 25
> >speed now 60, steer now 0, pwm 30, pwm 30
> >speed now 70, steer now 0, pwm 35, pwm 35
> >speed now 80, steer now 0, pwm 40, pwm 40
> >speed now 90, steer now 0, pwm 45, pwm 45
> >speed now 100, steer now 0, pwm 50, pwm 50
> >speed now 110, steer now 0, pwm 55, pwm 55
> >speed now 120, steer now 0, pwm 60, pwm 60
> >speed now 130, steer now 0, pwm 65, pwm 65
> >speed now 140, steer now 0, pwm 70, pwm 70
> >speed now 150, steer now 0, pwm 75, pwm 75
> >speed now 160, steer now 0, pwm 80, pwm 80
> >speed now 170, steer now 0, pwm 85, pwm 85
> >speed now 180, steer now 0, pwm 90, pwm 90
> >speed now 190, steer now 0, pwm 95, pwm 95
> >speed now 200, steer now 0, pwm 100, pwm 100

pasoftdev, 2019-08-30:
I'll try disabling phase advance and different motor control type..

pasoftdev, 2019-08-30:
- Result with PHASE_ADV_ENA           0 the same as PHASE_ADV_ENA           1
motor not spinning, high-pitch noise

pasoftdev, 2019-08-30:
Result withPHASE_ADV_ENA set to 1 and
 CTRL_TYP_SEL  set to 3,2,1,0 the same as default config,
motor not spinning, high-pitch noise ...
:((((

EmanuelFeru, 2019-08-30:
Hi, some debugging that you can try:
- use CTRL_TYP_SEL = 0 until you find the problem then you can go back to CTRL_TYP_SEL = 3, sinusoidal method
- try overwriting pwml=pwmr=100 in bldc.c just before they are assigned to the rtU structure. For safety, make sure the wheels are free and off the ground.
- if you still hear the high pitch noise without moving, I would double check the motor phases and Hall sensors if they are correctly connected.
- are the wheels stiff/blocked when you hear the high pitch noise? Can you move them by hand?

pasoftdev, 2019-08-30:
hi @EmanuelFeru ,

Ok i'll keep CTRL_TYP_SEL = 0  for the following tests.
The wheels can rotate freely when not powered on, they offer a bit more resistance when powered and when I set the PWM signal (>speed now 210, steer now 0, pwm 105, pwm 105)
about the motor phases and the Hall sensors, since I haven't disconnected them when I opened the hoverboard and since the hoverboard was working fine I think they are correct. Do you think it is worth a shot shuffling the Phase cables?
I'm going to try your suggestion (pwml=pwmr=100 in bldc.c )
thank you

pasoftdev, 2019-08-30:
hi @EmanuelFeru,
I just tried using pwml=pwmr=100 in bldc.c  but I've got the same behaviour, high pitch noise, motor not spinning, a slight resistance to turn by hand when using Iw

EmanuelFeru, 2019-08-30:
If you say the cables connections are the original ones, then I expect to be OK. Can you give a try with the same settings as I suggested above with the code in my repository? 
My repository is a bit more light because it does not contain the hoverboard mode. I try to see if we can isolate the problem.

pasoftdev, 2019-08-30:
Sure! I'll clone your repo, thank you a lot for your help

pasoftdev, 2019-08-30:
Hi,

I just tried with the code from your repo but I'm unable to see anything in the serial output
the motors don't spin, same behaviour
what I made:

Modified lines 48,49 of bldc.c
```
volatile int pwml = 100;
volatile int pwmr = 100;
```

Modified config.h, line 56
`#define CONTROL_SERIAL_USART2`
Modified config.h, line 67
`//define CONTROL_ADC`

To open the serial port i'm using
`screen /dev/ttyUSB0 19200`

(it works for the other firmware of course by setting the baudrate to 115200)

pasoftdev, 2019-08-30:
same behaviour with the above plus #define CTRL_TYP_SEL            0

EmanuelFeru, 2019-08-30:
I think the motors do not get enabled. I added a protection so that the motor don't jump at the start. To fix temporarily, go to main.c and comment the following if condition
```
// ####### MOTOR ENABLING: Only if the initial input is very small (SAFETY) #######

if (enable == 0 && (cmd1 > -50 && cmd1 < 50) && (cmd2 > -50 && cmd2 < 50)){

enable = 1; // enable motors

}
```
Leave uncomented only
```
enable = 1;
```
To see data on serial with my repositiry you need a serial converter connected to one of the sensor wire. But we don't really need serial for this exercise.
If the motors are enabled you will feel some resistance from the motors and you are still able to spin them by hand.

pasoftdev, 2019-08-30:
I just tried with enable = 1; in main.c, the motors offer no resistance just like when the power is off

EmanuelFeru, 2019-08-30:
And another thing. Put pwml=100; pwmr=100 just before line 122 in bldc.c
```
// ############################### MOTOR CONTROL ###############################
```
Because where you put it is just initialisation and will be overwritten immediately by the incoming data (to probably 0)

pasoftdev, 2019-08-30:
Ok,
I've put
```
  pwml = 100;
  pwmr = 100;
```
before 
`  // ############################### MOTOR CONTROL ###############################
`
and
`enable = 1`
in main.c
reflashed, powered on, the motors offer no resistance

EmanuelFeru, 2019-08-30:
If it still doesn't work, let's check again tomorrow because now I have to leave. First step, is to get motors enabled.

pasoftdev, 2019-08-30:
ok, thank you for your help :)

pasoftdev, 2019-08-31:
I'm going to try with a different control (poti) to see whether it makes any difference

pasoftdev, 2019-08-31:
Success! The poti control works! maybe now I can just proxy the control via an arduino, but it would be super cool to use the serial protocol if anyone can help me to debug the issue

pasoftdev, 2019-08-31:
I just tried the ADC with tankmode and two pots, it is working really well using the pots, still it is not working via the serial protocol (the human-readable one)
I'm using this config:
```
  // thoery says this is the only thing you need to change....
  #define CONTROL_TYPE USART3_CONTROLLED
  #define ADC_TANKMODE 1
  #define CONTROL_ADC               // use ADC as input. disable DEBUG_SERIAL_USART2!
  #define ADC1_MIN         0        // min ADC1-value while poti at minimum-position (0 - 4095)
  #define ADC1_ZERO     2047        // ADC1-value while poti at zero-position (0 - 4095)
  #define ADC1_MAX      4095        // max ADC1-value while poti at maximum-position (0 - 4095)
  #define ADC1_MULT_NEG 1000.0f     // Use 1000.0f to calibrate from MIN to MAX
  #define ADC1_MULT_POS 1000.0f     // Use 1000.0f to calibrate from MIN to MAX
  #define ADC2_MIN         0        // min ADC2-value while poti at minimum-position (0 - 4095)
  #define ADC2_ZERO     2047        // ADC2-value while poti at zero-position (0 - 4095)
  #define ADC2_MAX      4095        // max ADC2-value while poti at maximum-position (0 - 4095)
  #define ADC2_MULT_NEG 1000.0f     // Use 1000.0f to calibrate from MIN to MAX
  #define ADC2_MULT_POS 1000.0f     // Use 1000.0f to calibrate from MIN to MAX
  #define ADC_OFF_START    0          // Start Value of Area at which other inputs can be active (0 - 4095) Applies to Speed ADC
  #define ADC_OFF_END      0          // End Value of Area at which other inputs can be active (0 - 4095) Applies to Speed ADC
  #define ADC_OFF_FILTER 1.0          // Additional low pass Filter applied only to ADC Off functionality. 1.0=No Filter, 0.1 lots of Filtering
  //////////////////////////////////////////////////////////
```
I'll try too look into the code now that I'm sure it works via the potis

EmanuelFeru, 2019-08-31:
Nice! :wink: So, we know that all your connections are ok and the firmware compiles properly. Now indeed as you said is to dig a little to see why the protocol doesn't work.

pasoftdev, 2019-09-01:
yep!
by the way I see that some people have the same problem but they had it work with previous versions, is there a specific change that is worth to look at?

btsimonh, 2019-09-01:
Double check the flash config with the read all command.  Try resetting flash to defaults, re-read and compare.  If SHOULD have initialised the flash the first time you ran the firmware, but of course this functionality is not so easy to test....  Maybe you have a very small current limit set (thinking about it, that would make the noise - if the motors were trying but getting cut off in the DMA interrupt).  Read the current values (C?).  Maybe your current sensors are calibrated very differently, or even not working?

pasoftdev, 2019-09-01:
I just tried, the motor don't spin
if I get it right, the serial control and the adc one can coexist right? I've set the pot to middle values in order to not interfere, but I can reflash without adc to test.
here's the log with adc tankmode, usart3, working with the potis, not working via serial ascii
```
----
cf_speedCoef 10710, n_commDeacvHi 60, n_commAcvLo 30
unlockASCII
- unlockASCII
ASCII input active. Type ? for help
>?
- ?
? - display help
A - set alarm
B - toggle sensor control
C - show electrical measurements
E - dEbug control, E->off, Ec->console on, Es->console+scope
F - print/set a flash constant (Fa to print all, Fi to default all):
  Fss - print, Fss<n> - set
  Fm<n> - flash magic
  Fpkp<n> - posn kp x 100
  Fpki<n> - posn ki x 100
  Fpkd<n> - posn kd x 100
  Fpl<n> - posn pwm lim
  Fskp<n> - speed kp x 100
  Fski<n> - speed ki x 100
  Fskd<n> - speed kd x 100
  Fsl<n> - speed pwm incr lim
  Fcl<n> - max current limit x 100
  Fhe<n> - hoverboard enable

G - display stm32 specific
I - enable immediate commands - Ip/Is/Iw - direct to posn/speed/pwm control

L - Lock ascii protocol
M -  M - dump memory
   Mf - dump flash
   M<hexaddr>,<hexlen> - dump mem

N - display sensor data
P -  P -power control
  P -disablepoweroff
  PE enable poweroff
  Pn power off in n seconds
  Pr software reset

R -  - R! -> Reset Firmware
S - show main loop timing stats
T - tt - send a test protocol message 
u - unlockASCII - unlock ASCII protocol
>C
- C
Bat: 39545mV(0) Temp:201C(1690)
L: Current:59mA Avg:138mA r1:-1 r2:3
R: Current:59mA Avg:108mA r1:0 r2:1
>Fss
- Fss
unknown flash data Fss
>Fa
- Fa
flash magic(m): 1234
posn kp x 100(pkp): 50
posn ki x 100(pki): 50
posn kd x 100(pkd): 0
posn pwm lim(pl): 1000
speed kp x 100(skp): 20
speed ki x 100(ski): 10
speed kd x 100(skd): 0
speed pwm incr lim(sl): 20
max current limit x 100(cl): 1
hoverboard enable(he): 0
>?
- ?
? - display help
A - set alarm
B - toggle sensor control
C - show electrical measurements
E - dEbug control, E->off, Ec->console on, Es->console+scope
F - print/set a flash constant (Fa to print all, Fi to default all):
  Fss - print, Fss<n> - set
  Fm<n> - flash magic
  Fpkp<n> - posn kp x 100
  Fpki<n> - posn ki x 100
  Fpkd<n> - posn kd x 100
  Fpl<n> - posn pwm lim
  Fskp<n> - speed kp x 100
  Fski<n> - speed ki x 100
  Fskd<n> - speed kd x 100
  Fsl<n> - speed pwm incr lim
  Fcl<n> - max current limit x 100
  Fhe<n> - hoverboard enable

G - display stm32 specific
I - enable immediate commands - Ip/Is/Iw - direct to posn/speed/pwm control

L - Lock ascii protocol
M -  M - dump memory
   Mf - dumo flash
   M<hexaddr>,<hexlen> - dump mem

N - display sensor data
P -  P -power control
  P -disablepoweroff
  PE enable poweroff
  Pn power off in n seconds
  Pr software reset

R -  - R! -> Reset Firmware
S - show main loop timing stats
T - tt - send a test protocol message 
u - unlockASCII - unlock ASCII protocol
>Fi
- Fi

flash root 803f800
pe 15
wrote flash at 803f81e len 28Flash initialised
>Fa
- Fa
flash magic(m): 1234
posn kp x 100(pkp): 50
posn ki x 100(pki): 50
posn kd x 100(pkd): 0
posn pwm lim(pl): 1000
speed kp x 100(skp): 20
speed ki x 100(ski): 10
speed kd x 100(skd): 0
speed pwm incr lim(sl): 20
max current limit x 100(cl): 1
hoverboard enable(he): 0
>Iw
- Iw
Immediate commands enabled - WASDXHCGQ - Power (pWm) control
>>speed now 10, steer now 0, pwm 5, pwm 5
>speed now 20, steer now 0, pwm 10, pwm 10
>speed now 30, steer now 0, pwm 15, pwm 15
>speed now 40, steer now 0, pwm 20, pwm 20
>speed now 50, steer now 0, pwm 25, pwm 25
>speed now 60, steer now 0, pwm 30, pwm 30
>speed now 70, steer now 0, pwm 35, pwm 35
>speed now 80, steer now 0, pwm 40, pwm 40
>speed now 90, steer now 0, pwm 45, pwm 45
>speed now 100, steer now 0, pwm 50, pwm 50
>speed now 110, steer now 0, pwm 55, pwm 55
>speed now 120, steer now 0, pwm 60, pwm 60
>speed now 130, steer now 0, pwm 65, pwm 65
>speed now 140, steer now 0, pwm 70, pwm 70
>speed now 150, steer now 0, pwm 75, pwm 75
>speed now 160, steer now 0, pwm 80, pwm 80
>speed now 170, steer now 0, pwm 85, pwm 85
>speed now 180, steer now 0, pwm 90, pwm 90
>speed now 190, steer now 0, pwm 95, pwm 95
>speed now 200, steer now 0, pwm 100, pwm 100
>speed now 210, steer now 0, pwm 105, pwm 105
>speed now 220, steer now 0, pwm 110, pwm 110
>speed now 230, steer now 0, pwm 115, pwm 115
>speed now 240, steer now 0, pwm 120, pwm 120
>speed now 250, steer now 0, pwm 125, pwm 125
>speed now 260, steer now 0, pwm 130, pwm 130
>speed now 270, steer now 0, pwm 135, pwm 135
>speed now 280, steer now 0, pwm 140, pwm 140
>speed now 290, steer now 0, pwm 145, pwm 145
>speed now 300, steer now 0, pwm 150, pwm 150
>speed now 310, steer now 0, pwm 155, pwm 155
>speed now 320, steer now 0, pwm 160, pwm 160
>speed now 330, steer now 0, pwm 165, pwm 165
>speed now 340, steer now 0, pwm 170, pwm 170
>speed now 350, steer now 0, pwm 175, pwm 175
>speed now 360, steer now 0, pwm 180, pwm 180
>speed now 370, steer now 0, pwm 185, pwm 185
>speed now 380, steer now 0, pwm 190, pwm 190
>speed now 390, steer now 0, pwm 195, pwm 195
>speed now 400, steer now 0, pwm 200, pwm 200
>speed now 410, steer now 0, pwm 205, pwm 205
>speed now 420, steer now 0, pwm 210, pwm 210
>speed now 430, steer now 0, pwm 215, pwm 215
>speed now 440, steer now 0, pwm 220, pwm 220
>speed now 450, steer now 0, pwm 225, pwm 225
>speed now 460, steer now 0, pwm 230, pwm 230
>speed now 470, steer now 0, pwm 235, pwm 235
>speed now 480, steer now 0, pwm 240, pwm 240
>speed now 490, steer now 0, pwm 245, pwm 245
>speed now 500, steer now 0, pwm 250, pwm 250
>speed now 510, steer now 0, pwm 255, pwm 255
>speed now 520, steer now 0, pwm 260, pwm 260
>Stop set
>Stop set
>Stop set
>Stop set
>Stop set
>Stop set
>Stop set
>Stop set
>power off by button
---
```

pasoftdev, 2019-09-01:
.. and, I can't thank you enough!
I just power cycled after the previous log and it now works like a charm!
thank you very much @btsimonh 
however I don't see differences in the Fa output. is that right?

btsimonh, 2019-09-01:
current limit is defaulting to a very low value.  I don't know how it would work with pots, but not with protocol - both should account for the current limit...
try Fcl150, which from reading the comments on the command would be 15A.
double check with fa.
not sure on the co-existence; never use anything but serial and hoverboard :(.  
And of course my flash has always been initialised (flash has been there from day 1 ish...).

There must be some strange fault in the use of flash vars if you see no difference but now it's working....  but good news! - but do increase the limit unless you want to specifically stop the HB from driving through walls (or legs).

pasoftdev, 2019-09-01:
I actually would enjoy driving through walls since I'm making a rugged tank ;)
Seems like that by entering the Iw mode the pots stop responding. 
I'll now just recompile without adc control and continue with the building, once the drivetrain is ready I'll tune the current limit. again, thank you very much, I was stuck with this issue!

pasoftdev, 2019-09-01:
btw seems like it is not working without potis and buck converter on the left sensor cable,
even when reflashed with USART3 and without adc control. as a temporary workaround I can just keep a steady 1,67v to the adc lines but it feels like a waste
I'll try disabling flash storage..

sderr, 2019-09-30:
Hello guys,
For anyone experiencing that same behaviour -- slight resistance in the wheels when increasing the PWM, but no motor spinning, maybe you made the same mistake as I did :

As suggested in the README, I replaced the power button with a jumper to flash the firmware. Well, with that jumper still in place, everything seems to work, happy beeps, able to talk on the serial line, but nothing more ! Well, there's gotta be some kind of safety feature that prevents the wheel from spinning when the power button is pressed. Once I removed the jumper, ta-da ! The motors started.

pasoftdev, 2019-10-11:
thanks to everyone!
I just recompiled the firmware and triple checked every config parameters,
everything seems to work fine now!
I made a build log of my tank, if you're interested you can red it on my site at [pasoftdev.net](https://www.pasoftdev.net/)

closed 2019-10-11 by pasoftdev

btsimonh, 2019-10-11:
I want one!

EmanuelFeru, 2019-10-11:
@pasoftdev amazing build! I read full article.
The part "Since I never rode an hoverboard, I gave myself an hour or so of plain fun trying not to smash my head against walls." Hahaha, this is really funny. I recognize myself in this too.
Again, awesome project!

pasoftdev, 2019-10-12:
Thank you guys,
you helped a lot :)
I'll keep adding info on the hardware/software used