bipropellant/bipropellant-hoverboard-firmware #76: Poweroff not correctly working when using a buck converter to step down 15v to 3,3v - not a bug, probably hardware related

pasoftdev, 2019-08-31:
Hi there,

I'm using the ADC with tankmode and two pots,
stepping the +15V from the left sensor cable down to +3,3V going to the pots,
the control is very responsive and works fine so far (still I'm unable to use the USART2, see the other issue, comms are ok but motors just make an high-pitch noise),
I noticed that when I power-off the hoverboard using the original power button, the red led next to the STM doesn't turn off and it still emits the noise as it does when turned on.
I thought that the culprit could be some leftover current in the capacitor of the buck converter but when I disconnect the cables from the buck converter the led is still on, to turn completely off the hoverboard I have to disconnect the battery.
If I turn on the hoverboard without connecting the buck converter and then press the power button it shuts down gracefully and makes the bleeping sound before shutting completely down.
any suggestion / did it happen yo you?
thanks

pasoftdev, 2019-08-31:
I'll try the suggestion in the "Unwanted powerOn and bleep at poweroff" of the hardware troubleshooting, still it seems ok when not using the buck converter

btsimonh, 2019-09-01:
As you investigate, add to that bit of the guide.  We've had this kind of thing with serial apparently supplying enough power to keep the CPU alive, and also watch out for supplying power over programming connector.  I've never seen the issue, because everything is powered from the switched HB power (RPiZeroW inside, powered off DC-DC from a sensor 12v.).

sjamthe, 2019-09-01:
Yes, I reported this issue earlier. One of my USB to serial gives enough
power to keep the processor on. Other one works ok. Definitely a problem
with my adapter.

pasoftdev, 2019-10-13:
Turns out it was the buck converter I used to step the 15v down to 3,3v in order to use that as a base for the potis control scheme (took the 15v from the left sensor board cable), the control scheme with the potis worked very well but prevented a proper shutdown. Since I was using this control scheme only to advance my build and for debug purposes, I have removed it in the final build and how the shutdown works as intented even with the usb-to-serial connected and powered