bipropellant/bipropellant-hoverboard-firmware #77: Control hoverboard via Arduino

stefanfobos, 2019-09-08:
Hi there, 
I was trying   to control the board with 3.3 v arudino, but it didnt work at all. 
These are steps , that i've done:
*Downloaded the firmware and combined it with bipropellant-hoverboard-api ( just added files from api to the folder with the firmware)
*configured the firmware in config.h ( changed control method to  USART2_CONTROLLED) 
* flashed it sucessfully ( i hadn't been  able to flash the firmware until i added api files to the firmware folder )
* Took a sketch for arduino from the main page 
*connected Arduino to Left uart 
When i want to change the control method , shoud i change it here?
#ifndef CONTROL_TYPE
  #define CONTROL_TYPE HOVERBOARD_WITH_SOFTWARE_SERIAL_B2_C9
#endif
 
or comment everything in the top and only leave neccesary ?
P.S sorry for  my eng.
```
typedef struct MsgToHoverboard_t{
  unsigned char SOM;  // Start of Message
  unsigned char CI;   // continuity counter
  unsigned char len;  // len is len of bytes to follow, NOT including CS
  unsigned char cmd;  // read or write
  unsigned char code; // code of value to write
  int32_t pwm1;           // absolute value ranging from -1000 to 1000 .. Duty Cycle *10 for first wheel
  int32_t pwm2;           // absolute value ranging from -1000 to 1000 .. Duty Cycle *10 for second wheel
  unsigned char CS;   // checksumm
};

typedef union UART_Packet_t{
  MsgToHoverboard_t msgToHover;
  byte UART_Packet[sizeof(MsgToHoverboard_t)];
};

char hoverboardCI = 0;  // Global variable which tracks CI

void setHoverboardPWM( int32_t pwm1, int32_t pwm2 )
{
  UART_Packet_t ups;

  ups.msgToHover.SOM = 4 ;    // Start of Message, 4 for No ACKs;
  ups.msgToHover.CI = ++hoverboardCI; // Message Continuity Indicator. Subsequent Messages with the same CI are discarded, need to be incremented.
  ups.msgToHover.len = 1 + 1 + 4 + 4 ; // cmd(1), code(1), pwm1(4) and pwm2(4)
  ups.msgToHover.cmd  = 'r';  // Pretend to send answer to read request. This way HB will not reply. Change to 'W' to get confirmation from board
  ups.msgToHover.code = 0x0E; // "simpler PWM"
  ups.msgToHover.pwm1 = pwm1;
  ups.msgToHover.pwm2 = pwm2;
  ups.msgToHover.CS = 0;

  for (int i = 0; i < (2 + ups.msgToHover.len); i++){  // Calculate checksum. 2 more for CI and len.
    ups.msgToHover.CS -= ups.UART_Packet[i+1];
  }

  Serial.write(ups.UART_Packet,sizeof(UART_Packet_t));
}

void setup()
{
Serial.begin(9600);
//Serial.begin(115200);
}
void loop()
{
setHoverboardPWM(300,-300);
}
```

p-h-a-i-l, 2019-09-08:
Mixing the files from the API with the normal code is not a good idea. They share some files, but might rely on different versions.

Make sure to also clone the subrepositories when you clone into the hoverboard firmware. (see README)

Setting the control Method to USART2 is correct, it should then work with the Arduino Samples.

p-h-a-i-l, 2019-09-14:
Haven't used PPM in a along time together with this fork.

Might be neccessary to fix some things. 
Make sure to define a new CONTROL_TYPE or set it to 0

  `#define CONTROL_TYPE 0`