bipropellant/bipropellant-hoverboard-firmware #78: Can't use ascii protocol in Raspberry PI

rizals, 2019-09-16:
Sorry I'm beginner can you share app using in raspberry pi to send command ASCII protocol? 
and please you share sample code if use python for using command ASCII protocol?

AntumArk, 2019-09-17:
Have you tired googling "Send strings through UART with RPI"? And have you read protocol wiki?

rizals, 2019-09-25:
@AntumArk Thanks you, now I can use ASCII protocol but how control speed & use PID control using ASCII protocol?

w4ff3l, 2019-09-25:
Since im using a Raspberry Pi too, heres how I use the Protocol with Phails API. Although I have no clue if what I am doing is good practice. I am using wiringPi.
Open the UART Filedescriptor with
`fd = serialOpen("/dev/ttyUSB0", 115200)`
now Setup the Function Wrapper to send
```
int serialWrapper(unsigned char *data, int len) {
    for (int i = 0; i < len; i++) {
        serialPutchar(fd, data[i]);
    }
    return 1;
}
```
for reading just take what you get and let the protocol decipher it
```
char inc = serialGetchar(fd);
hoverboard.protocolPush(inc);
```
Now you can create the Hoverboard object with the SerialWrapper
```
HoverboardAPI hoverboard = HoverboardAPI(serialWrapper);
```
and use all of its functionality sending pwm and speed
```
void HoverboardAPI::sendPWM(int16_t pwm, int16_t steer, char som)
```
If you want to use the PID speed control you have to add functionality. Look for what you need in PARAMSTAT add the Hex to the enum in HoverboardAPI and add the variable to protocolFunctions. Looking at the already available functions in HoverboardAPI you will quickly see how to add your own.
[HoverboardAPI](https://github.com/p-h-a-i-l/HoverboardAPI)
[Examples Arduino](https://github.com/p-h-a-i-l/bipropellant-hoverboard-api/tree/master/examples)

p-h-a-i-l, 2019-09-25:
I wouldn't use ASCII protocol to control something, the machine protocol is more robust and designed just for that.

@w4ff3l If you extend the functionality or add rpi examples I'm happy to merge Pull requests ;)

rizals, 2019-09-25:
hi @w4ff3l thank you for your input and suggestions this is very helpful to me.

rizals, 2019-09-25:
hi @w4ff3l can you share your sample script? I have other problem after compile you can see this "HoverboardAPI.cpp:94:32: error: ‘Stream’ was not declared in this scope"

w4ff3l, 2019-09-26:
@p-h-a-i-l I have to embarrassingly admit that I never worked with GitHub. But I will try doing that after my school project...

@rizals Stream is a function from Arduino. You either reimplement the fuction without Arduino libraries or just comment it out if you don't need statistics from the serial right now.

rizals, 2019-09-26:
Hi @w4ff3l but if use #include <HoverboardAPI.h> and I'm compile the script I found error #include "Stream.h" not found ??, can you help for this and maybe you can send me sample code sorry I'm newbie. thank you

w4ff3l, 2019-09-26:
Yes, because you have to adapt the API from Arduino to Raspberry Pi. In simple words, you cannot just include the HoverboardAPI in your Raspberry Pi project before changing the Arduino functions to the Raspberry functions. I will try to give rpi examples when I have some spare time as I was saying...

rizals, 2019-09-26:
Hi @w4ff3l thank you I'm waiting your update examples.

alex-makarov, 2019-09-26:
My fork should be R-Pi friendly -- https://github.com/alex-makarov/bipropellant-hoverboard-api, I've removed Arduino dependencies.

p-h-a-i-l, 2019-09-26:
> My fork should be R-Pi friendly -- https://github.com/alex-makarov/bipropellant-hoverboard-api, I've removed Arduino dependencies.

@alex-makarov: Care to make a PR?

alex-makarov, 2019-09-26:
@p-h-a-i-l  My pleasure. Give me some time.

alex-makarov, 2019-09-26:
https://github.com/bipropellant/bipropellant-hoverboard-api/pull/8

closed 2019-12-08 by rizals