bipropellant/bipropellant-hoverboard-firmware #83: Help. Can't get it compiled with CONTROL_PPM set.

developmore, 2019-10-18:
Did anyone get it compiled and running with RC remote (PPM)? 
I set CONTROL_PPM to 1 and since then could not compile this firmware. 
Any help is greatly appreciated.

Ofekgg, 2019-10-28:
Happens to me too when I try to compile with ADC and serial off.

p-h-a-i-l, 2019-10-31:
do you get some kind of error message?

developmore, 2019-11-06:
Thank you for your response! Yes i got the following set of errors: 

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing genericSTM32F103RC (platform: ststm32; framework: stm32cube; board: genericSTM32F103RC)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-stm32cube 2.0.181130
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio\build\genericSTM32F103RC\src\ascii_proto_funcs.o
Compiling .pio\build\genericSTM32F103RC\src\bldc.o
Compiling .pio\build\genericSTM32F103RC\src\comms.o
Compiling .pio\build\genericSTM32F103RC\src\control.o
Compiling .pio\build\genericSTM32F103RC\src\flashaccess.o
Compiling .pio\build\genericSTM32F103RC\src\flasharea.o
Compiling .pio\build\genericSTM32F103RC\src\hallinterrupts.o
Compiling .pio\build\genericSTM32F103RC\src\hbprotocol\machine_protocol.o
Compiling .pio\build\genericSTM32F103RC\src\main.o
Compiling .pio\build\genericSTM32F103RC\src\protocolfunctions.o
Compiling .pio\build\genericSTM32F103RC\src\sensorcoms.o
Compiling .pio\build\genericSTM32F103RC\src\setup.o
src\main.c: In function 'main':
src\main.c:344:5: warning: implicit declaration of function 'PPM_Init'; did you mean 'HAL_Init'? [-Wimplicit-function-declaration]
     PPM_Init();
     ^~~~~~~~
     HAL_Init
src\main.c:479:13: warning: unused variable 'scale' [-Wunused-variable]
       float scale = ppm_captured_value[2] / 1000.0f;
             ^~~~~
src\main.c:1158:1: error: expected declaration or statement at end of input
 }
 ^
src\main.c:357:7: warning: unused variable 'last_control_type' [-Wunused-variable]
   int last_control_type = CONTROL_TYPE_NONE;
       ^~~~~~~~~~~~~~~~~
At top level:
src\main.c:1113:6: warning: 'SystemClock_Config' defined but not used [-Wunused-function]
 void SystemClock_Config(void) {
      ^~~~~~~~~~~~~~~~~~
src\main.c:940:6: warning: 'check_power_button' defined but not used [-Wunused-function]
 void check_power_button(){
      ^~~~~~~~~~~~~~~~~~
*** [.pio\build\genericSTM32F103RC\src\main.o] Error 1
======================================================================================= [FAILED] Took 12.74 seconds =======================================================================================

Environment             Status    Duration
----------------------  --------  ------------
genericSTM32F103RC      FAILED    00:00:12.740
hoverboard              IGNORED
hoverboard_6word        IGNORED
control_softwareserial  IGNORED
control_usart2          IGNORED
control_usart3          IGNORED
control_usart3_ADC      IGNORED
================================================================================== 1 failed, 0 succeeded in 00:00:12.740 ==================================================================================
The terminal process terminated with exit code: 1

I was able to build and deploy this project straight out of git and also when i changed CONTROL_TYPE to one of predefined values (e.g. USART2_CONTROLLED).

Then i changed these settings  in config.h
- set CONTROL_TYPE to  SOFTWARE_SERIAL_A2_A3, 
- uncommented these lines:
      #define CONTROL_PPM 
      #define PPM_NUM_CHANNELS 6  
- set not to include protocol because of some validation errors in config.h
      #define INCLUDE_PROTOCOL NO_PROTOCOL

[config.zip](https://github.com/bipropellant/bipropellant-hoverboard-firmware/files/3811932/config.zip)

I have a feeling i misconfigured  config.h
Your help is greatly appreciated!

p-h-a-i-l, 2019-11-06:
Hmm, I'm prettx sure I've already seen and fixed your problem ;)

Have a look at https://github.com/p-h-a-i-l/bipropellant-hoverboard-firmware/tree/adcSteerNonlinear
This is my "working copy", should be fixed there. _Someone_ should probably merge back some bugfixes..

p-h-a-i-l, 2019-11-06:
Also be warned, PPM is pretty alpha. Someone made a fork to niklasfauths firmware with improvementsif I remember it right..

p-h-a-i-l, 2019-11-06:
ok, fixed on master. From now on travis also checks if PPM can be compiled. See https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/5dcbbaa3fdbd990e167c8e7943c5f5944cd0b45d/platformio.ini#L207-L227

closed 2019-11-06 by p-h-a-i-l

developmore, 2019-11-08:
Thank  you @p-h-a-i-l , it compiles for me now! Awesome! Time to put it to a test.