bipropellant/bipropellant-hoverboard-firmware #84: Info: FOC for hoverboard-firmware-hack

gnbl, 2019-10-24:
@EmanuelFeru has implemented FOC based on https://github.com/NiklasFauth/hoverboard-firmware-hack:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

Maybe this could be integrated into "bipropellant". Then all that's missing are the modifications / driving modes for cars from https://github.com/larsmm/hoverboard-firmware-hack-bbcar ;-)

p-h-a-i-l, 2019-10-24:
Hello @gnbl,
I also follow Emanuels development, unfortunately I don't have time right now to work on new features. (And there are also other PRs and features in the pipeline which are more important to me right now)
I'd really like to see FOC in biporpellant, but on the other hand Sinus is already quite nice..

Maybe you have some time to merge FOC or the driving modes ;)

gnbl, 2019-10-24:
I'm just getting started, and have yet to get my bearings, but should I happen to come up with anything interesting, like merging the two projects, I'll let you know of course.