bipropellant/bipropellant-hoverboard-firmware #85: Where is steer and filter funtions?

papaxiclas, 2019-11-11:
Hello,
I am moving from the original repo to bipropellant but before I dig any deeper in the code can someone confirm that using the serial protocol I have to send the order for the 2 wheels independently and not the power+steering values?
Also, I am trying to fine tune the FILTER and SPEED_COEFFICIENT on config.h but they seem to have no effect. They are no used anymore? how can I achieve the same result?

Thanks

p-h-a-i-l, 2019-11-11:
When the values are send via protocol, FILTER and SPEED_COEFFICIENT have no effect. You can easily implement them on the sending side. (for example: SPEED_COEFFICIENT is a simple multiplication)
The values for both wheels have to be sent independently.

papaxiclas, 2019-11-12:
Ok, understood!:) thanks
By the way of these command modes (described on protocol.c) which ones are available?
0x02, "hall data"
0x03, "speed control mm/s"
0x04, "hall position mm", 
0x05,"position control increment mm"
0x06,"position control mm"
0x07,"hall position steps"
0x08, "electrical measurements"
0x09, "enable motors"
0x0A, "disable poweroff timer"
0x0B, "enable console logs"
0x0C, "read/clear xyt position"
0x0D, "PWM control"
0x0E, "simpler PWM"

Thanks for your great job with this fork! :)

p-h-a-i-l, 2019-11-13:
All of them are implemented. Also have a look at https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/master/src/protocolfunctions.c to understand which global variables and functions are used for these values