bipropellant/bipropellant-hoverboard-firmware #86: Possible to drive other motors using this board/firmware?

shane0926, 2019-11-14:
I have some 13" wheels that have a different motor configuration than that of the hoverboard motors.  My motors have less stator arms/windings. Is it possible to alter the firmware to account for this? Currently, when I drive the motor, it stalls easily and runs jerky with high amps.
[image]

shane0926, 2019-11-14:
I'm not sure if the positioning of the hall sensors is causing the issue? Here is a photo of my motor and an image of the hoverboard motor.
[image]
[image]

EmanuelFeru, 2019-12-13:
Hi, could be that indeed you have a different Hall configuration. Possibly you have a 60 deg configuration and not a 120 deg configuration (120 deg is what the firmware expects).
According to this article, it should work by inverting Hall 3 sensor in the code bldc.c
https://community.parker.com/technologies/electromechanical-group/w/electromechanical-knowledge-base/174/hall-sensors-60-vs-120degrees

Give it a try, I don't guarantee anything but it worth trying.