bipropellant/bipropellant-hoverboard-firmware #91: Please create a releases page

sevsev9, 2020-01-26:
Today i tried to build the project "as is" both locally and on the [pioneerland](https://pionierland.de/hoverhack/) site. Both failed and I am stuck as i am not a native in coding in C/++.

Compiler states the following:
Locally build it says, that in 'control_structures.h' protocol.c cannot be found.

And on the page it states the following:
```
src/protocolfunctions.c: In function 'setup_protocol':
src/protocolfunctions.c:356:5: error: 'protocol_GetTick' undeclared (first use in this function); did you mean 'protocol_tick'?
     protocol_GetTick = HAL_GetTick;
     ^~~~~~~~~~~~~~~~
     protocol_tick
src/protocolfunctions.c:356:5: note: each undeclared identifier is reported only once for each function it appears in
src/protocolfunctions.c:357:5: error: 'protocol_Delay' undeclared (first use in this function); did you mean 'protocol_enable'?
     protocol_Delay = HAL_Delay;
     ^~~~~~~~~~~~~~
     protocol_enable
src/protocolfunctions.c:358:5: error: 'protocol_SystemReset' undeclared (first use in this function); did you mean 'protocol_byte'?
     protocol_SystemReset =HAL_NVIC_SystemReset;
     ^~~~~~~~~~~~~~~~~~~~
     protocol_byte
make: *** [build/protocolfunctions.o] Error 1
```

Would be glad if help arrives fast.
So i can get my 500W shopping cart working soon :)

NickLD, 2020-01-28:
Hi @sevsev9 
It sounds like you are missing the "hbprotocol" directory and files.
This defines the Machine and ASCII protocol for the communication between your HB Mainboard and your main controller.

If you cloned the repo with git, it should've automatically pulled down the [hbprotocol repo](https://github.com/bipropellant/bipropellant-protocol) with it.
If you instead download via "Download .ZIP" option, it will NOT download the hbprotocol repo necessary to build the firmware. It will instead just create an empty /hbprotocol/ directory in it's place.

Easiest way to fix this is to simply do the 'proper' git clone of the repo, but you can also simply download the hbprotocol repo and drop its directory into /src/ .

sevsev9, 2020-01-28:
As you stated I did download the .zip.
Did the git clone. Worked like a charm.
Thanks for the fast reply!

closed 2020-01-28 by sevsev9