bipropellant/bipropellant-hoverboard-firmware #92: Speed reported with wrong sign

jpiat, 2020-01-28:
Hi,

i can control my hoverboard using the PWM commands but i fail to get the speed reported correctly. When using a positive PWM value, i get a negative speed reported and when i use a negative PWM value i get a positive speed. This also affect the speed control commands since the PID error keeps growing and the wheel speed gets out of control.

regards,

Jonathan Piat

p-h-a-i-l, 2020-01-28:
Hello jpiat,
That's a known issue, and I already fixed it ;)
Unfortunately this is part of a bigger protocol rewrite, and I didn't have time to clean up and merge yet.
If you like, you can use my repo and tell me if you notice some problems.

Please note: The whole protocol got changed, it should be way more reliable now, but unfortunately it is not compatible. 
If you were using the hoverboard API, there is also an update version in my repositories:

https://github.com/p-h-a-i-l/bipropellant-hoverboard-firmware
https://github.com/p-h-a-i-l/bipropellant-hoverboard-api

jpiat, 2020-01-28:
https://github.com/p-h-a-i-l/bipropellant-hoverboard-firmware refers to the bipropellant repository for the protocol, so what part of the protocol was changed? I'm working on a 64bits platform, so did you merge alex-makarov pull request also in your api repository ?

p-h-a-i-l, 2020-01-28:
yeah, that's merged. Really appreciate if you test with 64 bit, some things might need adjustment.

I forgot to mention the correct branches.. There the protocol from my repo should be referenced. Sorry for the confusion

https://github.com/p-h-a-i-l/bipropellant-hoverboard-firmware/tree/protocolCOBSR
https://github.com/p-h-a-i-l/bipropellant-hoverboard-api/tree/protocolCOBSR

jpiat, 2020-01-28:
Makefile of the COBSR branch does not have cobsr.c listed in the sources.

p-h-a-i-l, 2020-01-28:
That happens when you ignore your CI :) https://github.com/p-h-a-i-l/bipropellant-hoverboard-firmware/runs/361356094
I've been using only platformio during development. Will fix soon. (or you can make a PR)

jpiat, 2020-02-21:
Hi, i still cannot get the speed to work. When i set the speed to 0.5 m/s for one wheel and the other to 0, both wheels start to spin and the wheel with 0.5 m/s speed starts going very fast until the controller shuts down (low battery). 

PWM control works fine, and reading speed also works fine and reported speed looks good. 

I tried changing  this line :

`sendRawData(
    PROTOCOL_CMD_WRITEVAL,
    Codes::setSpeed,
    (unsigned char *) &writespeed,
    sizeof(writespeed.wanted_speed_mm_per_sec) + sizeof(writespeed.speed_max_power) + sizeof(writespeed.speed_min_power) + sizeof(writespeed.speed_minimum_speed),
    som
  );`

to 

`sendRawData(
    PROTOCOL_CMD_WRITEVAL,
    Codes::setSpeed,
    (unsigned char *) &writespeed,
    sizeof(writespeed),
    som
  );`

and i get a very weird behavior with the wheel starting to spin and then stopping and starting again.

p-h-a-i-l, 2020-02-24:
You have to tune the PID parameters for the controller to work. I never actually used this feature myself. If I  remember correctly, the closed loop control works "ok" for position control but Is not very good for speed control. So some work neccessary.

jpiat, 2020-02-26:
i think there is an error in the API code to set speed. The API code does the following to set speed

`sendRawData( PROTOCOL_CMD_WRITEVAL, Codes::setSpeed, (unsigned char *) &writespeed, sizeof(writespeed.wanted_speed_mm_per_sec) + sizeof(writespeed.speed_max_power) + sizeof(writespeed.speed_min_power) + sizeof(writespeed.speed_minimum_speed), som );`

while in the protocol.c, the speed data size is sizeof(PROTOCOL_SPEED_DATA). I believe this results in the setSpeed method of the API to send too little data, resulting in weird behavior.

p-h-a-i-l, 2020-02-27:
You can send too little data, the protocol overwrites only the beginning then. It's only important to have the alignment of the data correct. This is very useful when you want to update the set point often, but do not want to sent the overhead all the time.
Didn't look specifically into that case now, can do if needed.

jpiat, 2020-02-27:
I get a different behavior using the two methods (sending the full struct
or just the first useful words) so there must be something wrong here.

jpiat, 2020-02-27:
When sending the full struct for speed, I can somewhat get the wheel
turning at the set speed but with high frequency oscillations (vibrations).