bipropellant/bipropellant-hoverboard-firmware #94: Doesn't Connect board to PMM.

ilkarataev, 2020-02-21:
Hi, i want to configure hoverboard working with PPM protocol(speed and steering).I build the project with platformio and only declare PPM section, loaded it ti board and when connect  r6 transmitter(i had one more) connected it to Green right cable UART,but nothing moving.I tried so many times and with another clone of project.I don't have signal analyzing to see what happen.
And i don't understand in README about "Sensor data can control the PWM demands (power to the wheels). Double tap on pads to enable."What mean Double tap on pads double tap on power button? I am not very good at English and sorry for this.

I set up platformio.ini only default_envs = control_PPM need i enable define control_PPM and PPM_NUM_CHANNELS 6?

I try many times but it doesn't work.
I will very appreciate you if you can give some advice.

ilkarataev, 2020-03-13:
Hi i found the answer, if someone have this issue you need add the same count of channels that your transmitter have.In my case i had RC6 with ten channels and when i add the PPM_NUM_CHANNELS 10 all start work, sorry for my bad english

closed 2020-03-13 by ilkarataev

Andrew-rw, 2021-07-12:
In my case for FS-IA6B receiver PPM_NUM_CHANNELS=8 worked. 
Helped this issue: https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/158