bipropellant/bipropellant-hoverboard-firmware #96: STM 32 ST link dongle problem

ACG-TR, 2020-03-25:
Hello I am new here, I try to flash the firmware i created from pioneerland to the new hoverboard mainboard i have. 

I tried powering the mainboard from 36V battery, from the ST32 dongle(while disconnecting the battery) the red light is blinking when powered but i always get the same error as "cannot connect to target" 

the cable connections are correct i checked 10 times already:) GND to GND CLK to CLK and DIO to DIO. 

I was not able to unlock the unit as there is no clear tutorial about it. I am using windows 10. is it the problem? 

Can anyone help please.

NickLD, 2020-03-31:
Hi ACG-TR.
It sounds to me that the board is resetting when the ST-LINK attempts to talk to it.
This is due to how the power circuitry works from what I understand. The board, when reset, powers off the CPU and will disconnect from ST-LINK.
It is CRITICAL that you hold the power button PRESSED (or jump it with a wire or link) the entire time you talk to it with the ST-LINK (aka, when unlocking and flashing)

If you are still having issues with this let me know and I'll try to send you a video depicting what exactly to do.

Hope that helps!

ACG-TR, 2020-04-02:
Hi, Thank you for your feedback. I tried 3 boards after creating that issue. I follow the jumping the button protocole seriously and that doesnt help in 1 board at all. The STM does not see it. I managed to flash 1 mainboard succesfully. 

the 3rd board when i click connect it connects to the board succesfully for a second and than the board resets it self. I also tried powering the board with the ST link dongle 3.3V other than powering it with a battery. 

I tried all the combinations of pressed button, released button etc. It just resets the board after communication is established. 

Pretty weird. I dont think that all the boards are the same. Especially newer generations are equipped with some kind of protection i think.

NickLD, 2020-04-04:
Hmm, That is really odd.
Are the main CPUs any different between the boards (specifically, the successfully flashed one and the 3rd board) ? Only boards with this design I've seen use either STM32 or GD32 processors.

Did you test if the boards work at all with the original firmware? Perhaps could be DOA?