bipropellant/bipropellant-hoverboard-firmware #97: bipropellant/bipropellant-hoverboard-firmware  compile error

pav18, 2020-03-27:
Hello.
Can't compile bipropellant/bipropellant-hoverboard-firmware - error. Nothing helps

NickLD, 2020-03-31:
Hi pav18
How are you attempting to compile the firmware online?
it is fairly easy to pull down the repo and build the firmware with VSCode, have you tried that?

pav18, 2020-04-03:
https://pionierland.de/hoverhack/

RoboDurden, 2020-04-04:
Yes, my online compiler can not clone the full code successfully. And i have forgotten almost all about that process.. That is what my php script execs:

```
exec git -C source/bipropellant_bipropellant-hoverboard-firmware fetch --all && git -C source/bipropellant_bipropellant-hoverboard-firmware reset --hard origin/master
stdout:
Fetching origin
HEAD is now at 5dcbbaa ease the pain of ifdef-hell

stderr:
```

I also tried to download the bipropellant-protocol-master.zip and unzip it into src/hbprotocol
I guess that helped last year.

That is my compiler error output: 

```
src/protocolfunctions.c: In function 'setup_protocol':
src/protocolfunctions.c:356:5: error: 'protocol_GetTick' undeclared (first use in this function); did you mean 'protocol_tick'?
     protocol_GetTick = HAL_GetTick;
     ^~~~~~~~~~~~~~~~
     protocol_tick
src/protocolfunctions.c:356:5: note: each undeclared identifier is reported only once for each function it appears in
src/protocolfunctions.c:357:5: error: 'protocol_Delay' undeclared (first use in this function); did you mean 'protocol_enable'?
     protocol_Delay = HAL_Delay;
     ^~~~~~~~~~~~~~
     protocol_enable
src/protocolfunctions.c:358:5: error: 'protocol_SystemReset' undeclared (first use in this function); did you mean 'protocol_byte'?
     protocol_SystemReset =HAL_NVIC_SystemReset;
     ^~~~~~~~~~~~~~~~~~~~
     protocol_byte
make: *** [build/protocolfunctions.o] Error 1

```

`protocol_GetTick` is defined in https://github.com/bipropellant/bipropellant-protocol/blob/master/protocol.c 

So i guess it is the same problem as last time.
Ideas welcome !

NickLD, 2020-04-04:
This appears to be the[ same issue I help another user with](https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/91) a while ago.
Judging by the errors, I believe that online compiler isn't correctly pulling down the HBProtocol sub-module as required to build the firmware.

Follow the steps listed in the other issue to pull down the repo with git and build with VSCode

RoboDurden, 2020-04-04:
Hi Nick, sorry to bother you with my online compiler.
Can you please tell what is wrong with my git call:

`git -C source/bipropellant_bipropellant-hoverboard-firmware fetch --all && git -C source/bipropellant_bipropellant-hoverboard-firmware reset --hard origin/master
`

NickLD, 2020-04-04:
@RoboDurden No worries.
The crucial part that is missing is the **--recurse-submodules** when you clone the repo.
This will pull down the required sub modules (dependencies) and place them in the appropriate location and with the proper version.

ex.
`git clone --recurse-submodules https://github.com/bipropellant/bipropellant-hoverboard-firmware.git`

will produce:
```
Cloning into 'bipropellant-hoverboard-firmware'...
remote: Enumerating objects: 1944, done.
remote: Total 1944 (delta 0), reused 0 (delta 0), pack-reused 1944
Receiving objects: 100% (1944/1944), 47.03 MiB | 11.11 MiB/s, done.
Resolving deltas: 100% (1216/1216), done.
Checking out files: 100% (602/602), done.
Submodule 'lib/hbprotocol' (https://github.com/bipropellant/hbprotocol.git) registered for path 'src/hbprotocol'
Cloning into 'C:/Users/nickL/source/testbipropellant/bipropellant-hoverboard-firmware/src/hbprotocol'...
remote: Enumerating objects: 354, done.
remote: Total 354 (delta 0), reused 0 (delta 0), pack-reused 354
Receiving objects: 100% (354/354), 118.34 KiB | 7.40 MiB/s, done.
Resolving deltas: 100% (211/211), done.
Submodule path 'src/hbprotocol': checked out '6f9d1bd6053a9e82bbf4e953ddf95e1274202e1a'
```
Note how it finds the hbprotocol submodule dependency and pulls it down, too.

RoboDurden, 2020-04-04:
thank you very much :-)
After `git clone` i need to use `git fetch` to update a repo. And i forgot to add the `--recurse-submodules` in that exec.
Unfortunately it still did not help to add `--recurse-submodules` to the `git fetch` command :-(
So i for now simply deleted the entire repo and let my online compile exec the `git clone` command.

Now it successfully compiles :-)

Maybe you know why adding `--recurse-submodules` to `git fetch` did not work.

NickLD, 2020-04-04:
Great! I'm glad it worked :)
I should have added a little more detail. --recurse-submodules only will work for the initial pull of the repo. This works fine if you 'blow out' the directory after you compile, but if you try to get the submodules after you have already cloned it will not work.
In that event, you should run something like:
`git submodule update --init --recursive`

That command will work after you have pulled down the repo.

RoboDurden, 2020-04-04:
Okay this is my new command line to update an existing repo:

`git -C source/bipropellant_bipropellant-hoverboard-firmware fetch --all && git -C source/bipropellant_bipropellant-hoverboard-firmware reset --hard origin/master && git -C source/bipropellant_bipropellant-hoverboard-firmware submodule update --init --recursive
`

Don't really know if i still need the git `reset --hard` command but when i now rename the `src/hbprotocol` into `src/hbprotocolOld` , a new `hbprotocol` folder gets created :-)

pav18, 2020-04-05:
I like you guys ;)))) The words "Hello" and " Bye" were the only word I was able to understand but everything works. Great job. Thank you very much.  ;)

NickLD, 2020-04-05:
@RoboDurden `reset --hard` isn't necessary if you are deleting the directory after the user compiles and then re-download it.
However, if you are keeping the directory around (say, for caching and performance reasons) then the `reset --hard` is a good idea to keep, as it will flush away any of the changes made to the code if they are not committed, and revert back to HEAD.
It really depends on how you have implemented your online compiler logic, but it certainly won't hurt anything to keep it :)

@pav18 I'm glad everything works now! Happy I could help out.

pav18, 2020-04-06:
@RoboDurden, @NickLD just suggested me new project on Github. https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

Is it possible to add it to your Online compilator?
Thank you in advance. ;)

RoboDurden, 2020-04-06:
done: www.pionierland.de/hoverhack

You have to activate a communication method in config.h like

```
#if !defined(PLATFORMIO)
  #define VARIANT_ADC         // Variant for control via ADC input
  //#define VARIANT_USART       // Variant for Serial control via USART3 input
  //#define VARIANT_NUNCHUK     // Variant for Nunchuk controlled vehicle build
  //#define VARIANT_PPM         // Variant for RC-Remote with PPM-Sum Signal
  //#define VARIANT_PWM         // Variant for RC-Remote with PWM Signal
  //#define VARIANT_IBUS        // Variant for RC-Remotes with FLYSKY IBUS
  //#define VARIANT_HOVERCAR    // Variant for HOVERCAR build
  //#define VARIANT_HOVERBOARD  // Variant for HOVERBOARD build
  //#define VARIANT_TRANSPOTTER // Variant for TRANSPOTTER build https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter https://hackaday.io/project/161891-transpotter-ng
#endif

```

**Now please** - don't be a stupid consumer but make a youtube review where you demonstrate the differences between this FOC-firmeware and the original(!) Niklas-firmware !!

pav18, 2020-04-06:
@RoboDurden WOW. You are the best!!!! ;) 
Thank you very much. 

Friendly speaking I am in shock about of all the information I had to dig last week ;)  And honestly till today I do not know what the difference between, for example, FOC and Niklas firmware - I am just following suggestions and accept any possible help. I am not programmer guy so all this is really complicated for me. 
I really apprciate your help. If I need to dig how to compile firmware by myself I would spend ate least another day ;)     
But I will try to combine and share all the information I have as a result of my project.

pav18, 2020-04-06:
@RoboDurden 
There are two versions of the firmware - for STM and GD
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM
https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD

If I understand right you made compiler for STM. May I ask you to add for GD too. 
I have two different boards - forward wheels and rear.
Thank you

pav18, 2020-04-06:
@RoboDurden  It compiles for STM but says "could not parse original config.h". Is it Ok?

NickLD, 2020-04-06:
@pav18
Those additional firmwares are not for mainboards.

They instead replace the firmware that runs on the small balance boards used in hoverboard. I dont expect you are trying to use them on your project?
Regardless, they likely don't belong on on his online compiler, or at least not on the same page, as it's meant for compilation of mainboard firmwares.

RoboDurden, 2020-04-06:
Oh nice someone made a firmeware for these balance boards - they come for free and might be useful.
All my compiler needs is a Makefile. And as these firmewares also come with a config.h and main.c 
i happly added them.

I am not so happy that this guy is not willing to add a Makefile: https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2/issues/10
And i am not experience enough to copy and paste a Makefile :-/

pav18, 2020-04-06:
@RoboDurden  I did nothing but site is down ;) Its not me ;)

RoboDurden, 2020-04-06:
Yes i was in the middle of adding info titles to the repo selecton.