bipropellant/bipropellant-hoverboard-firmware #100: Self balancing robot

Cra7y, 2020-04-23:
Can this firmware self balance and then just receive steering information from an Arduino?
I would want to use it for a humanoid kind of robot where i basically just strap it to top of a unmodified hoverboard.

N0TB0T, 2020-04-24:
Just want to secont that question, cause that's exactly what I was planning, too. There is a hoverboard mode, but does it need someone standing on the board or fake sensor data for steering? And I've read an unanswered question concerning the software serial pins, which I was not able to detect, either...

NickLD, 2020-04-24:
Hi @Cra7y and @N0TB0T 
While it's possible to implement self balancing robot with these mainboards ( it's clearly possible as that was their original purpose) this firmware isn't very well suited to do so.
Related to issue #72 

Typically, for self balancing applications, you would want torque control over motors rather than Voltage or Speed control. There is [another branch from one of the developers who has been working on this](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC). Torque control is implemented in addition with FOC. There is no balancing code implemented as of yet, but it is certainly a good base for you to help contribute. [He has also written replacement code for the balance boards](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-STM) for this purpose, with the Serial data already being received from them in the mainboard firmware.

Cra7y, 2020-04-30:
Thanks @NickLD 
That sounds promising, would love to implement it. But it seams that i miss the basic understanding of how the processor is programmed. I have done a fair bit of programming and Arduino projects, but it seams there is a long way from pressing that one-click upload button on the Arduino ide to compiling code for the mainboards.
And what is the difference between the inc and src folder?
If you have any idea on where i should start if i want to understand this it is greatly appreciated. I have a lot of general knowledge about electronics and microprocessors, so i guess i could figure it out if someone gave me the first clue to this puzzle ;)
If not then that's fair too, guess this isn't really the place for this type of help anyway.

NickLD, 2020-04-30:
Hi @Cra7y 
I understand looking at a project such as this can seem daunting, but It will become clear and easy to work with once you break it up in chunks to begin to understand it. You will see it's really quite easy to move to away from the Arduino IDE to a 'real' editor/IDE such as VSCode w/ PlatformIO.
Arduino IDE provides a great and super simple way to get started with programming embedded projects, but becomes quite a hindrance to develop larger programs due to how basic it is.
This project and many others are able to be built with the PlatformIO IDE, which provides many advantages such as Intellisense, real debugging, peaking at definitions, dark mode, and much more.
In-fact, development with PlatformIO is just as easy as the 'one-click' upload button, but comes with a host of features that easy and speed up development.
Developing Arduino sketches is also even easier than with the official Arduino IDE as it supports the Arduino framework on many platforms.
You should download it [here](https://platformio.org/platformio-ide)  and then look at the quickstart [here](https://docs.platformio.org/en/latest/integration/ide/vscode.html#quick-start) which will walk you through the classic 'blink sketch' in Arduino framework.

Now, this project isn't built with the Arduino framework, but instead uses [stm32cube](https://docs.platformio.org/en/latest/frameworks/stm32cube.html)

Also, the inc just contains all the [header files](https://www.tutorialspoint.com/cprogramming/c_header_files.htm) used by the rest of the code
Let me know if you have any more questions.

Cra7y, 2020-05-02:
Thank you @NickLD that's a really great introduction!
I am currently waiting for my STLink to arrive, then I plan on starting the project.
I will probably be back with more questions once it arrives and I try it myself.
Btw, do you know if a "blue pill" devboard could be used as a training platform so I don't risk breaking the main board immediately? I believe it uses an STM32F103C8T6.

buttim, 2023-09-20:
Any update on this issue? Is it still not possible to have the thing self balancing and at the same time some how pilot it remotely or by any other means not involving a human on board?