bipropellant/bipropellant-hoverboard-firmware #105: Error: jtag status contains invalid mode value - communication failure

gotros, 2020-07-28:
Hi,
I try to unlock the CPU with openocd commands in the tutorial. I cannot connect to the target device and get the following error. I triple checked the wirering.

```
openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "mww 0x40022004 0x45670123" -c "mww 0x40022004 0xCDEF89AB" -c "mww 0x40022008 0x45670123" -c "mww 0x40022008 0xCDEF89AB" -c targets -c "halt" -c "stm32f1x unlock 0"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 1.552756
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Error: mem2array: Read @ 0xe0042004, w=4, cnt=1, failed
/usr/bin/../share/openocd/scripts/mem_helper.tcl:6: Error: 
in procedure 'reset' 
in procedure 'ocd_bouncer' 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 248
in procedure 'stm32f1x.cpu' called at file "embedded:startup.tcl", line 299
in procedure 'ocd_bouncer' 
in procedure 'mmw' 
in procedure 'mrw' called at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 25
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 6

in procedure 'reset' 
in procedure 'ocd_bouncer'
```

Do I have to hold the power button even for the openocd commands?

I have a board with a STM32f103-RCT6 processor.

I can support with fotos.

Thx in advance,
gotros

gotros, 2020-07-30:
I also cannot connect with the st-link utility.

I fear to have broken something. Either the SWD USB Adapter or something on the board. If I power up the board over the USB Adapter I get a rhytmic beep (on,off,on.....). If I power up the board via the battery I get the regular turnon sound. After about 30 seconds the board starts beeping in no defined scheme. Maybe the differences are there, because the battery powers parts on the board which the USB adapter doesn't. Maybe somebody can explain that behavior.

I did power on the board with battery attached and power of the USB adapter connected. It was late^^ Maybe this difference in behavior mentioned before comes from that? Would this mistake render the device useless?

Thx for reading,
gotros

NickLD, 2020-07-30:
Hi gotros,
1. Do not power the board via the SWD programmer, you can kill the mainboard. Only connect the two programming wires & ground.
2. Ensure you are powering the board with a charged battery.
3. With the board off & SWD programmer connected, HOLD the power-button down. It will turn on, but do not release the button!
4. With the power button held, you can begin the unlock commands & programming.
5. Only once the CPU is unlocked & flashed with new firmware can you release the power-button.

Holding the power button down (or using a jumper on the button pins) is critical to having this work, as the CPU controls the power circuitry. When you enter programming mode, the CPU is no longer running the program, and thus it will inadvertently turn itself off while unlocking & programming.

btsimonh, 2020-07-31:
p.s. beware the power button connections - they carry the full 30v (I shorted one to a cpu pin by mistake and torched a board).

gotros, 2020-08-09:
At the end my original ST flasher was not working - maybe broken or not. A 7€ china rebuild worked like a charm.

closed 2020-08-09 by gotros