bipropellant/bipropellant-hoverboard-firmware #107: Get Functions is not Working

bekirbostanci, 2020-07-29:
Normally I can setPWM my hoverboard but i tried get speed data but is not working. 
All get function returns 0 or 0.00 data at the serial monitor. 
Is there any configuration about this or how can I check the program ? 

Note: 
I use USART2

bekirbostanci, 2020-07-31:
In the api I control the message and I am reading the board 
//Serial = ESP32 <=> My Computer 
//Serial2 = ESP32 <=> Hoverboard

Serial.println(Serial2.read());

This line print totaly "-1"  what does it mean I dont understand 
Can you help me @p-h-a-i-l

bekirbostanci, 2020-07-31:
15:00:52.059 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.093 -> 119414311474222152206513231121317113521522065135211313711217
15:00:52.093 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.127 -> 119414111474222152206513231121319113521522065135211313711202
15:00:52.127 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.127 -> 119314012341641531651311105111186115211123311211651311102111165154111-1
15:00:52.161 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.161 -> 1193140125416515616513111141111951185111236112216513111291111671441111
15:00:52.195 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.195 -> 119314012741671541651311132111187112711123911201651311155111159184111-1
15:00:52.228 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.228 -> 119314-111474222152206513231121317113521522065135211313811193
15:00:52.262 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.262 -> 119314013141691511651311149111179171111245111816513112091111521120111-1
15:00:52.296 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.296 -> 1193140133417115516513111671111921251111248111716513112351111501132111-1
15:00:52.296 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)
15:00:52.330 -> 119314-11147422215220651323112131811352152206513521131381153
15:00:52.330 -> Motor Speed: 0.00km/h (average from 0.00mm/s and 0.00mm/s)

Now I am getting some values but there are meaningness

boredomwontgetus, 2020-10-14:
Hey!

I am having the exact same problem with USART2. 
Did you manage to solve this by now? 

BR

bekirbostanci, 2020-10-19:
@boredomwontgetus I didnt solve the problem and changed firmware https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/

closed 2020-10-19 by bekirbostanci