bipropellant/bipropellant-hoverboard-firmware #110: Can not connect to target

triballo, 2020-12-03:
Hi,
I have a problem in my board mod, when I try to connect to the board ST-LINK tool it returns the error "Can not connect to target!".
As a programmer I am using a st-link v2 and the board is a new YST-DXT-J20 V 5.1.
I have checked the position of the PINs a thousand times, I have also tried both procedures both with 3.3v and without battery and without 3.3v but with battery. Some advice?
Please, I'm desperate

ashwin-ar, 2020-12-12:
hey @triballo !
I suppose you might have done these steps wrong while using the st link utility!
you need to perform erase chip after unlocking it and then select program and verify option

bekirbostanci, 2020-12-13:
If your card color is red dont try to work because I have that one. This brand of card didnt detect.

NickLD, 2020-12-14:
@triballo Are you holding the power button down the entire time? (or jumping the pins with a jumper)
That is required to keep the CPU alive for programming, as the way the power button works is it turns on the CPU, which then toggles a pin to keep itself on. When in programming mode, the code is of course not running, and cannot keep itself on.
Steps:
1. Connect programmer (not VCC/3.3V though)
2. Connect battery
3. Press and release power button (board should beep)
4. **Press AND HOLD** the power button now until firmware is finished flashing
5. Init flash procedure with PlatformIO Upload, or via ST-Link Utility.

triballo, 2020-12-15:
> @triballo Are you holding the power button down the entire time? (or jumping the pins with a jumper)
> That is required to keep the CPU alive for programming, as the way the power button works is it [...]

I performed exactly the steps you indicated but nothing has changed, calculate that my board has no LEDs.

I connected the ST-Link v2 to the board (without vcc / 3v).
I powered the board with the battery
I pressed the power button until I heard a beep
After a few seconds I kept the power button with a jumper
started ST-link utility .... same problem

triballo, 2020-12-15:
> If your board color is red dont try to work because I have that one. This brand of board is not detect.

No my board is black

triballo, 2020-12-15:
> hey @triballo !
> I suppose you might have done these steps wrong while using the st link utility!
> you need to perform erase chip after unlocking it and then select program and verify option

In my case, I never managed to connect to the board, neither with ST-Link Utility, nor with openocd.

chandru28k, 2021-02-22:
Any luck. I think I fried my board! I was able to connect and flash the image "controlsoftwareserial" , then I want to flash "control_uart2" but I could not connect to the board from STlink-v2 (firmware VJ37S7)  i have to upgrade to this. any help how i can connect back to motherBoard. -thx

aldock, 2024-12-03:
Hello, i had same problem. It happens when flash an wrong code and STM32 can not boot by flash memory any more. You need boot by system, connect Boot0 to Vcc and connect through an usart adapter, Tx to PA10, Rx to PA9, Vcc and Gnd, Just this. Use STM32 cube programmer, option usart, click connect(It works!) and at erase&programming menu , click start programming a correct code. Now You can again use stlink!! with Boot0 to Gnd.  That is it !!