bipropellant/bipropellant-hoverboard-firmware #113: Compile error

omkarg, 2021-03-20:
Hi,

I downloaded the code from Github and I am trying to compile it in PlatformIO but I am getting a compile error

inc/protocolfunctions.h:3:10: fatal error: protocol.h: No such file or directory

Should I be adding the protocol.h file from here https://github.com/bipropellant/bipropellant-hoverboard-api/blob/master/src/protocolFunctions.h

omkarg, 2021-03-20:
> Hi,
>
> I downloaded the code from Github and I am trying to compile it in PlatformIO but I am getting a compile error
>
> inc/protocolfunctions.h:3:10: fatal error: protocol.h: No such file or directory [...]

btsimonh, 2021-03-20:
you need this repo too:
https://github.com/bipropellant/bipropellant-protocol
- can't remember exactly how they are combined :(.