bipropellant/bipropellant-hoverboard-firmware #118: current sensing and measurement

flute2k3, 2021-08-12:
I go through the codes but seems there is no phase current measurement codes implemented. the hoverboard applies mosfet on resistance current sensing, the algorithm is kind of complicated to separate the phase current from bus current. without phase current sensing, how to ensure the typical foc control?

Candas1, 2021-08-12:
Why do you assume this repo does FOC control?

Candas1, 2021-08-16:
I am not familiar with this repo but I believe [this](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC) one is the only one with foc control

flute2k3, 2021-08-16:
thanks Candas1, the repo you mentioned is referring to Original firmware: https://github.com/NiklasFauth/hoverboard-firmware-hack, there is a video of original repo talking about FOC control. so the original repo does a simple FOC control considering only 6 rotation position, and without current loop - means the author treat DQ voltage as control target instead of current. by this way I assume there will be unavoidable torque ripple and less efficiency. a true FOC as the repo you mentioned is claimed to "reduced noise and vibrations smooth torque output and improved motor efficiency. Thus, lower energy consumption".
it will be very interesting to see the current sensing and calculation detail since there are 3 ops per motor on board for this purpose, two for 2 phase mosfet on-Rds sensing and one for bus current sensing.